esphome/issues #1077: dev branch - stack trace fails with AttributeError: 'str' object has no attribute 'decode'

geoffdavis, 2020-03-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Dev install of esphome on HassIO

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

core

**Description of problem:**
Running custom code that crashes fails to bring up the stack trace decoder.

```
 File "/usr/local/lib/python3.6/dist-packages/esphome/platformio_api.py", line 101, in run_idedata
    stdout = run_platformio_cli_run(config, False, *args, capture_stdout=True).decode()
AttributeError: 'str' object has no attribute 'decode'
```

**Problem-relevant YAML-configuration entries:**
```yaml
N/A

```

**Logs (if applicable):**

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[16:02:16][I][logger:156]: Log initialized
[16:02:16][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[16:02:16][I][app:028]: Running through setup()...
[16:02:16][D][text_sensor:015]: 'denheatpump ESPHome Version': Sending state '1.15.0-dev Mar  6 2020, 15:59:59'
[16:02:16][C][myheatpump:226]: Setting up UART bus...
[16:02:16][C][myheatpump:238]: hw_serial(0x3ffc1550) is &Serial(0x3ffc1580)? NO
[16:02:16][C][myheatpump:239]: Calling hp.connect(0x3ffc1550)
[16:02:16]Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
[16:02:16]Core 1 register dump:
[16:02:16]PC      : 0xfefefefe  PS      : 0x00060930  A0      : 0x800e7860  A1      : 0x3ffb1b50  
[16:02:16]A2      : 0x3ffb9308  A3      : 0x3ffc1550  A4      : 0x00000001  A5      : 0x3ffc1978  
[16:02:17]A6      : 0xfffdd3b1  A7      : 0x00000062  A8      : 0x800e77d0  A9      : 0x3ffb1b10  
[16:02:17]A10     : 0x3ffc1550  A11     : 0x00000960  A12     : 0x0800001e  A13     : 0x00000010  
[16:02:17]A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000014  
[16:02:17]EXCVADDR: 0xfefefefc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[16:02:17]
[16:02:17]Backtrace: 0x7efefefe:0x3ffb1b50 0x400e785d:0x3ffb1b90 0x400e2523:0x3ffb1bb0 0x400e1092:0x3ffb1be0 0x40168659:0x3ffb1c30 0x400e0e35:0x3ffb1c50 0x400e3723:0x3ffb1ca0 0x400f08bb:0x3ffb1fb0 0x40088cd1:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/denheatpump -t idedata
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome==1.15.0.dev0', 'console_scripts', 'esphome')()
  File "/usr/local/lib/python3.6/dist-packages/esphome/__main__.py", line 557, in main
    return run_esphome(sys.argv)
  File "/usr/local/lib/python3.6/dist-packages/esphome/__main__.py", line 544, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python3.6/dist-packages/esphome/__main__.py", line 329, in command_run
    return show_logs(config, args, port)
  File "/usr/local/lib/python3.6/dist-packages/esphome/__main__.py", line 207, in show_logs
    run_miniterm(config, port)
  File "/usr/local/lib/python3.6/dist-packages/esphome/__main__.py", line 114, in run_miniterm
    config, line, backtrace_state=backtrace_state)
  File "/usr/local/lib/python3.6/dist-packages/esphome/platformio_api.py", line 228, in process_stacktrace
    _decode_pc(config, addr.group())
  File "/usr/local/lib/python3.6/dist-packages/esphome/platformio_api.py", line 166, in _decode_pc
    idedata = get_idedata(config)
  File "/usr/local/lib/python3.6/dist-packages/esphome/platformio_api.py", line 121, in get_idedata
    IDE_DATA = run_idedata(config)
  File "/usr/local/lib/python3.6/dist-packages/esphome/platformio_api.py", line 101, in run_idedata
    stdout = run_platformio_cli_run(config, False, *args, capture_stdout=True).decode()
AttributeError: 'str' object has no attribute 'decode'
```

**Additional information and things you've tried:**

This seems like a bug in the Python 2 to 3 migration.

Using my amazing google skillz and recently (painfully) acquired python2 to 3 migration experience, my wild guess is that `run_platformio_cli_run` is already handling the conversion from bytes to str, and that just removing the call `.decode()` on line 101 should fix this issue.

geoffdavis, 2020-03-07:
Did a test in my docker instance, and indeed removing the the call to decode makes things work as expected.

```
[16:37:41][I][logger:156]: Log initialized
[16:37:41][C][ota:364]: There have been 9 suspected unsuccessful boot attempts.
[16:37:41][I][app:028]: Running through setup()...
[16:37:41][D][text_sensor:015]: 'denheatpump ESPHome Version': Sending state '1.15.0-dev Mar  6 2020, 16:36:03'
[16:37:41][C][myheatpump:226]: Setting up UART bus...
[16:37:41][C][myheatpump:238]: hw_serial(0x3ffc1550) is &Serial(0x3ffc1580)? NO
[16:37:41][C][myheatpump:239]: Calling hp.connect(0x3ffc1550)
[16:37:41]Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
[16:37:41]Core 1 register dump:
[16:37:41]PC      : 0xfefefefe  PS      : 0x00060b30  A0      : 0x800e7860  A1      : 0x3ffb1b50  
[16:37:41]A2      : 0x3ffb9278  A3      : 0x3ffc1550  A4      : 0x00000001  A5      : 0x3ffc1978  
[16:37:41]A6      : 0xfffb91ee  A7      : 0x00000030  A8      : 0x800e77d0  A9      : 0x3ffb1b10  
[16:37:41]A10     : 0x3ffc1550  A11     : 0x00000960  A12     : 0x0800001e  A13     : 0x00000010  
[16:37:41]A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000014  
[16:37:41]EXCVADDR: 0xfefefefc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[16:37:41]
[16:37:41]Backtrace: 0x7efefefe:0x3ffb1b50 0x400e785d:0x3ffb1b90 0x400e2523:0x3ffb1bb0 0x400e1092:0x3ffb1be0 0x40168659:0x3ffb1c30 0x400e0e35:0x3ffb1c50 0x400e3723:0x3ffb1ca0 0x400f08bb:0x3ffb1fb0 0x40088cd1:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e785d: HeatPump::connect(HardwareSerial*) at /data/denheatpump/.piolibdeps/denheatpump/HeatPump/src/HeatPump.cpp:187
WARNING Decoded 0x400e2523: MyHeatPump::setup() at /config/esphome/denheatpump/src/main.cpp:235
WARNING Decoded 0x400e1092: esphome::PollingComponent::call_setup() at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1754
WARNING Decoded 0x40168659: esphome::Component::call() at /config/esphome/denheatpump/src/esphome/core/component.cpp:82
WARNING Decoded 0x400e0e35: esphome::Application::setup() at /config/esphome/denheatpump/src/esphome/core/application.cpp:37
WARNING Decoded 0x400e3723: setup() at /config/esphome/denheatpump/src/main.cpp:230
WARNING Decoded 0x400f08bb: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x40088cd1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[16:37:42]
[16:37:42]Rebooting...
```

timsavage, 2020-03-07:
Hi Geoff have confirmed your findings and opened a PR with the fix.

closed 2020-03-12 by brandond