esphome/issues #1078: Web Server component crashing

abmantis, 2020-03-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Arch Linux, Pip installation.
**ESP (ESP32/ESP8266, Board/Sonoff):**

nodemcu-32s (clone probably).
**Affected component:**

https://esphome.io/components/web_server.html

**Description of problem:**
The web server component is calling abort(), causing the ESP to reboot.

**Problem-relevant YAML-configuration entries:**
```yaml
```

**Logs (if applicable):**

```
[23:34:48]abort() was called at PC 0x401cd027 on core 1
[23:34:48]
[23:34:48]Backtrace: 0x4009260c:0x3fffbcd0 0x4009283d:0x3fffbcf0 0x401cd027:0x3fffbd10 0x401cd06e:0x3fffbd30 0x401cc013:0x3fffbd50 0x401cc12e:0x3fffbd70 0x401cc0cd:0x3fffbd90 0x400ee1b1:0x3fffbdb0 0x400ee1f5:0x3fffbdd0 0x400ea210:0x3fffbdf0 0x400ea234:0x3fffbe10 0x400ed16e:0x3fffbe30 0x400ed195:0x3fffbe50 0x400daa11:0x3fffbe70 0x400dae52:0x3fffbe90 0x400db9fd:0x3fffbf90 0x400dbbac:0x3fffc0f0 0x400e97f5:0x3fffc110 0x400e98b9:0x3fffc150 0x400e9b21:0x3fffc190 0x400e4075:0x3fffc1b0 0x400e4109:0x3fffc1f0 0x400e47ae:0x3fffc210 0x4008ed55:0x3fffc240
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d esp32 -t idedata
WARNING Decoded 0x4009260c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x4009283d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x401cd027: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401cd06e: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401cc013: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
WARNING Decoded 0x401cc12e: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x401cc0cd: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
WARNING Decoded 0x400ee1b1: cbuf::resize(unsigned int) at /home/amfcosta/.platformio/packages/framework-arduinoespressif32/cores/esp32/cbuf.cpp:83
WARNING Decoded 0x400ee1f5: cbuf::resizeAdd(unsigned int) at /home/amfcosta/.platformio/packages/framework-arduinoespressif32/cores/esp32/cbuf.cpp:83
WARNING Decoded 0x400ea210: AsyncResponseStream::write(unsigned char const*, unsigned int) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/ESPAsyncWebServer-esphome_ID6758/src/WebResponses.cpp:594
WARNING Decoded 0x400ea234: non-virtual thunk to AsyncResponseStream::write(unsigned char const*, unsigned int)
WARNING Decoded 0x400ed16e: Print::write(char const*) at /home/amfcosta/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
WARNING Decoded 0x400ed195: Print::print(char const*) at /home/amfcosta/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
WARNING Decoded 0x400daa11: esphome::web_server::write_row(AsyncResponseStream*, esphome::Nameable*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/src/esphome/components/web_server/web_server.cpp:554
WARNING Decoded 0x400dae52: esphome::web_server::WebServer::handle_index_request(AsyncWebServerRequest*) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/src/esphome/components/web_server/web_server.cpp:554
WARNING Decoded 0x400db9fd: esphome::web_server::WebServer::handleRequest(AsyncWebServerRequest*) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/src/esphome/components/web_server/web_server.cpp:554
WARNING Decoded 0x400dbbac: non-virtual thunk to esphome::web_server::WebServer::handleRequest(AsyncWebServerRequest*)
WARNING Decoded 0x400e97f5: AsyncWebServerRequest::_parseLine() at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/ESPAsyncWebServer-esphome_ID6758/src/WebRequest.cpp:797
WARNING Decoded 0x400e98b9: AsyncWebServerRequest::_onData(void*, unsigned int) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/ESPAsyncWebServer-esphome_ID6758/src/WebRequest.cpp:797
WARNING Decoded 0x400e9b21: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/ESPAsyncWebServer-esphome_ID6758/src/WebRequest.cpp:797
 (inlined by) _M_invoke at /home/amfcosta/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400e4075: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/AsyncTCP-esphome_ID6798/src/AsyncTCP.cpp:1138
 (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/AsyncTCP-esphome_ID6798/src/AsyncTCP.cpp:935
WARNING Decoded 0x400e4109: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/AsyncTCP-esphome_ID6798/src/AsyncTCP.cpp:1138
WARNING Decoded 0x400e47ae: _async_service_task(void*) at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/AsyncTCP-esphome_ID6798/src/AsyncTCP.cpp:1138
 (inlined by) _async_service_task at /home/amfcosta/dev/arduino_projects/esphomeyaml/esp32/.piolibdeps/esp32/AsyncTCP-esphome_ID6798/src/AsyncTCP.cpp:194
WARNING Decoded 0x4008ed55: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[23:34:50]
[23:34:50]Rebooting...

```

**Additional information and things you've tried:**
This may be caused by long or fast log messages, since I had the RC receiver enabled, dumping ray and rc_switch messages, and it was receiving a lot of them.

brandond, 2020-03-08:
Looks to me like it's running out of memory. What were you doing at the time of the crash? Can it be reproduced on demand?

abmantis, 2020-03-08:
I had the remote_receiver dumping raw messages, and it was receiving a lot of garbage. Maybe having that with a floating pin would reproduce this too.

samnewman86, 2020-03-12:
as soon as i try to open the esp hosted web page the device crashes.

brandond, 2020-03-13:
@samnewman86 the web server component has been around for quite a while, and is in general pretty stable. Can you post more information on your configuration?

abmantis, 2020-03-13:
@brandond have you tried my suggestion?

netpok, 2020-03-31:
Can confirm the issue it crashes for me too, only when it's open in the browser. I can reproduce the problem by refreshing the page few times but it crashes on it's own too.

I don't have access to the stack trace, I guess it is printed out to the serial port and my device is not connected to my computer (if really needed probably I can connect an other esp to forward the serial logs).

My config:
```
esphome:
  name: watersensors
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "<REMOVED>"
  password: "<REMOVED>"
  domain: <REMOVED>

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "ws Fallback Hotspot"
#    password: "<REMOVED>"

#captive_portal:

#debug:

# Enable logging
logger:
#  level: debug
  baud_rate: 0

ota:

web_server:
  port: 80

time:
  - platform: sntp

mqtt:
  broker: <REMOVED>
  username: <REMOVED>
  password: <REMOVED>
  on_json_message:
    topic: tele/lights/SENSOR
    then:
      - sensor.template.publish:
          id: humidity
          state: !lambda 'return x["SHT3X-0x44"]["Humidity"];'

switch:
#  - platform: gpio
#    name: LED
#    id: blue_led
#    pin: 2
  - platform: gpio
    name: "Toilet Vent"
    id: toilet_vent
    pin: 13
  - platform: gpio
    name: "Mirror Light"
    id: mirror_light
    pin: 12
  - platform: gpio
    name: "Pantry Light"
    id: pantry_light
    pin: 14
  - platform: gpio
    name: "Mirror Heating"
    id: mirror_heating
    command_topic: cmnd/digoo/POWER1
    pin: 27
  - platform: gpio
    name: "Toilet Light"
    id: toilet_light
    pin: 15
  - platform: gpio
    name: "Bathroom Speaker"
    id: bathroom_speaker
    pin: 19
  - platform: gpio
    name: "Bathroom Light"
    id: bathroom_light
    pin: 21
    command_topic: cmnd/lights/POWER3
    on_turn_off:
      - switch.turn_off: mirror_light
      - switch.turn_off: bathroom_speaker
  - platform: restart
    name: "WaterSensors Restart"

dallas:
  - pin: 4
    update_interval: 10s

binary_sensor:
  - platform: gpio
    name: "Mirror switch"
    pin:
      number: 26
      mode: INPUT_PULLUP
      inverted: True
    on_click:
      - max_length: 1000ms
        then:
        - switch.toggle: mirror_light
      - min_length: 2000ms
        max_length: 5000ms
        then:
        - switch.toggle: bathroom_speaker
  - platform: gpio
    name: "Pantry switch"
    pin:
      number: 25
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.turn_on: pantry_light
    on_release:
      - switch.turn_off: pantry_light
  - platform: gpio
    name: "Toilet switch"
    pin:
      number: 23
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.turn_on: toilet_light
    on_release:
      - switch.turn_off: toilet_light

sensor:
  - platform: dallas
    address: 0xBD3C01A816748028
    name: "Mixed water temperature"

  - platform: dallas
    address: 0xB33C01A816709D28
    name: "Cold water temperature"

  - platform: dallas
    address: 0xF33C01A816A35728
    name: "Hot water temperature"

  - platform: pulse_counter
    pin:
      number: 34
      mode: INPUT
    unit_of_measurement: 'l/min'
    name: 'Toilet flow rate'
    id: fs1
    filters:
        - multiply: 0.0015
    update_interval: 10s
  - platform: integration
    name: "Water 1"
    sensor: fs1
    time_unit: min

  - platform: pulse_counter
    pin:
      number: 33
      mode: INPUT
    unit_of_measurement: 'l'
    name: 'Flow sensor 2'
    id: fs2
    filters:
        - multiply: 0.0015
    update_interval: 10s
  - platform: integration
    name: "Water 2"
    sensor: fs2
    time_unit: min

  - platform: pulse_counter
    pin:
      number: 36
      mode: INPUT
    unit_of_measurement: 'l'
    name: 'Incoming water flow rate'
    id: fs3
    filters:
        - multiply: 0.0018837
    update_interval: 10s
  - platform: integration
    name: "Water 3"
    sensor: fs3
    time_unit: min

  - platform: pulse_counter
    pin:
      number: 39
      mode: INPUT
    unit_of_measurement: 'l'
    name: 'Flow sensor 4'
    id: fs4
    filters:
        - multiply: 0.0015
    update_interval: 10s
  - platform: integration
    name: "Water 4"
    sensor: fs4
    time_unit: min
  - platform: adc
    pin: 32
    name: "Filtered Water Pressure"
    filters:
      - multiply: 1.68
      - calibrate_linear:
        - 0.5 -> 0.0
        - 4.5 -> 12.0
    attenuation: 11db
    unit_of_measurement: 'bar'
    update_interval: 10s

  - platform: adc
    pin: 35
    name: "Incoming Water Pressure"
    filters:
      - multiply: 1.67
      - calibrate_linear:
        - 0.5 -> 0.0
        - 4.5 -> 12.0
    attenuation: 11db
    unit_of_measurement: 'bar'
    update_interval: 10s

  - platform: template
    name: "Bathroom humidity"
    id: humidity

# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: Uptime

  - platform: wifi_signal
    name: "Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: version
    name: firmware_version
```

Extra lines in the main.cpp:
```
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  App.setup();
  gpio_set_drive_capability((gpio_num_t)13, GPIO_DRIVE_CAP_3);
  gpio_set_drive_capability((gpio_num_t)12, GPIO_DRIVE_CAP_3);
  gpio_set_drive_capability((gpio_num_t)14, GPIO_DRIVE_CAP_3);
  gpio_set_drive_capability((gpio_num_t)27, GPIO_DRIVE_CAP_3);
  gpio_set_drive_capability((gpio_num_t)15, GPIO_DRIVE_CAP_3);
  gpio_set_pull_mode(GPIO_NUM_33, GPIO_FLOATING);
```

Removing the extra lines, sensors and additional mqtt related commands does not seem to affect the crashes.

closed 2020-12-05 by stale[bot]