esphome/issues #1079: ESPHOME and HASSIO not integrating

Sander71, 2020-03-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHOME version 1.14.3  on HASSIO 0.106.2 running on a raspberry pi4.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff POW v2
Zemismart Curtain controller
Sonoff

**Affected component:**

**Description of problem:**

All esphome devices in hassio  show up as "not connected" in ESPHOME integration they show up as "offline" since yesterday. I am able to reflash the devices firmware. I used this on two devices to set a static ip adress.  If i use a static Ip adress i am able to use the "SHOW LOGS" option in esphome and it shows device statistics. Also sensor information is shown in the window.

To me it seems like esphome labels the devices offline incorectly (and therfore hassio too) . 

however , i think the Show logs windows proves that there is no issue with the devices or conection to the pi.

**Problem-relevant YAML-configuration entries:**
```
Which YAML would you like to see ?
```

**Logs (if applicable):**

```
what logs should i upload (sorry kinda green using github and related tools

```

**Additional information and things you've tried:**

I tried setting static ip adresses which solves the mDNS errors in esphome. Pinging the .local adress from my pc works perfectly though.
i tried removing the 5GHZ option in my WIFI
i tried disabling mesh WIFI
ESPHOME  v1.14.2, v1.13.6 (did not work), and v1.15_dev
Hassio 106.5
Tried restoring a snapshot from when it was working
rebooting and restarting between changes

h0jeZvgoxFepBQ2C, 2020-07-19:
I have the same, but not with hassio, but a native linux install. Tried everything (remove devices completely from HA, remove json entries). 

ESPHome logs say everything is ok, but HA show all sensors as offline / not available.

glmnet, 2020-07-19:
You'll need to provide hass logs I guess 

And will need to increase log level. Check hass documentation on how to do that

h0jeZvgoxFepBQ2C, 2020-07-25:
See also here: https://community.home-assistant.io/t/esphome-device-says-connected-successfully-is-sending-sensor-data-but-appears-completely-offline-in-home-assistang-no-mqtt/214025

closed 2020-12-06 by stale[bot]