esphome/issues #1081: M5StickC IR LED stays ON after transmitting

abmantis, 2020-03-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Linux/Pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

MStack M5StickC

**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
After sending data with remote_transmitter, the IR LED on the M5StickC stays ON (it is easy to check with a camera or a IR receiver). Also tried setting the internal LED instead of the IR one, and it also stays on. Using an external LED does not exhibit this issue. Could it be related to the M5StickC internal LEDs using the GPIO9 and 10? Still, I can turn them on and off. It's only when I use the remote_transmitter that it stays on.

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  - pin: GPIO9
    carrier_duty_percent: 50%

switch:
  - platform: template
    name: Raw Code Power Button
    id: irs
    turn_on_action:
      - remote_transmitter.transmit_raw:
        code: [4088, -1542, 1019, -510, 513, -1019, 510, -509, 511, -510, 1020,
                 -1020, 1022, -1019, 510, -509, 511, -510, 511, -509, 511, -510,
                 1020, -1019, 510, -511, 1020, -510, 512, -508, 510, -1020, 1022]
          repeat:
            times: 1
            wait_time: 0ms
```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**
Only having issues with the internal LEDs.

OttoWinter, 2020-07-24:
On ESP32 we use the rmt peripheral and configure the `idle_level`.

After a send is complete the driver should go back to that. If not that would be a bug in esp-idf.

However I think it's more likely that the issue is that the GPIO pin for the LED is inverted.

ekorsanov01, 2020-08-04:
I have the same problem? 
Author, have you solved the problem?

abmantis, 2020-08-08:
@OttoWinter I tried using it inverted, but it did not fix the problem.
@ekorsanov01 I have not.

closed 2020-12-20 by stale[bot]