esphome/issues #1082: Raw IR transmitting not working

Chuckame, 2020-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Odroid XU4
Hass.io: `0.106.6`
Docker: `19.03.5`
EspHome docker image: `esphome/esphome-hassio-armv7:1.14.3`

**ESP (ESP32/ESP8266, Board/Sonoff):**

Nodemcu-v2 Lol1n ESP12E

**Affected component:**

https://esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-raw-action

**Description of problem:**
Unable to transmit raw code previously catched with receiver dump mode.

I want to send 'Power on/off' to my Z906 home cinema, the protocol is an IR NEC-like, but without command/address, just value, so the receiver component printed raw codes...

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: transmitter
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_transmitter:
  pin: D6
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "transmitter"
    on_turn_on:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9211, -4236, 661, -458, 774, -1455, 700, -419, 801, -319, 797, -323, 804, -314, 774, -346, 778, -341, 746, -374, 774, -345, 771, -348, 777, -342, 829, -294, 829, -1396, 773, -347, 688, -1543, 743, -377, 714, -405, 774, -345, 802, -318, 743, -375, 743, -377, 661, -458, 747, -1481, 775, -1456, 687, -1540, 715, -1514, 775, -1454, 803, -1425, 794, -1435, 738, -1490, 747, -366, 746]
```

**Logs (if applicable):**

```
[00:01:18][I][app:100]: ESPHome version 1.14.3 compiled on Mar 10 2020, 20:29:39
[00:01:18][C][wifi:415]: WiFi:
[00:01:18][C][wifi:283]:   SSID: [redacted]
[00:01:18][C][wifi:284]:   IP Address: xxxx
[00:01:18][C][wifi:286]:   BSSID: [redacted]
[00:01:18][C][wifi:287]:   Hostname: 'transmitter'
[00:01:18][C][wifi:291]:   Signal strength: -82 dB ▂▄▆█
[00:01:18][C][wifi:295]:   Channel: xxx
[00:01:18][C][wifi:296]:   Subnet: xxxx
[00:01:18][C][wifi:297]:   Gateway: xxx
[00:01:18][C][wifi:298]:   DNS1: xxx
[00:01:18][C][wifi:299]:   DNS2: xxx
[00:01:18][C][template.switch:058]: Template Switch 'transmitter'
[00:01:18][C][template.switch:059]:   Restore State: NO
[00:01:18][C][template.switch:060]:   Optimistic: NO
[00:01:18][C][logger:175]: Logger:
[00:01:18][C][logger:176]:   Level: DEBUG
[00:01:18][C][logger:177]:   Log Baud Rate: 115200
[00:01:18][C][logger:178]:   Hardware UART: UART0
[00:01:18][C][remote_transmitter:018]: Remote Transmitter...
[00:01:18][C][remote_transmitter:019]:   Carrier Duty: 50%
[00:01:18][C][remote_transmitter:020]:   Pin: GPIO12 (Mode: OUTPUT)
[00:01:18][C][ota:029]: Over-The-Air Updates:
[00:01:18][C][ota:030]:   Address: transmitter.local:8266
[00:01:18][C][api:095]: API Server:
[00:01:18][C][api:096]:   Address: transmitter.local:6053
[00:01:20][D][switch:021]: 'transmitter' Turning ON.
```

**Additional information and things you've tried:**
If I use a simple code using `IRremoteESP8266`, without esphome, I can't receive too.
And if I transmit using `IRremoteESP8266`, all is working using the NEC-like protocol with the value only filled.

Code for receiving : 
```cpp
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <ir_NEC.h>
#include <IRutils.h>

#define IR_IN D1

IRrecv irRecv(IR_IN, 1024, 50, false);
decode_results irRecvResults;

void initIR()
{
  irRecv.enableIRIn();
  Serial.println("IR initialized !");
}

void setup()
{
  Serial.begin(115200);

  initIR();
}

void loop()
{
  if (irRecv.decode(&irRecvResults))
  {
    Serial.println();
    Serial.print("addr: 0x");
    Serial.print(irRecvResults.address, HEX);
    Serial.println();
    Serial.print("cmd:  0x");
    Serial.print(irRecvResults.command, HEX);
    Serial.println();
    Serial.print("value:  0x");
    serialPrintUint64(irRecvResults.value, HEX);
    irRecv.resume(); // Receive the next value
  }
  yield();
}
```

Code for transmit : 
```cpp
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRutils.h>

#define IR_OUT D2
IRsend irSend(IR_OUT);

#define IR_SIGNAL_LENGTH 32
#define IR_SIGNAL_PROTOCOL decode_type_t::NEC
uint64_t IR_POWER_BTN = 0x400501FE;
void initIR()
{
  irSend.begin();
  Serial.println("IR initialized !");
}

void setup()
{
  Serial.begin(115200);

  initIR();
}

void loop()
{
    Serial.println("Remote: power !");
    irSend.send(IR_SIGNAL_PROTOCOL, IR_POWER_BTN, IR_SIGNAL_LENGTH);
  delay(1000);
}
```

glmnet, 2020-03-23:
Ahhh you should see on the log if it is debug that the remote code is trasmitted.

Your problem (or at least part of it) is a bit typical, you are using `on_turn_on` instead of `turn_on_action`

I see docs are outdated anyway. Can you explain why you are using `on_turn_on`? May be you read the docs and it is wrong or confusing somewhere. Thanks

Chuckame, 2020-03-24:
Thank you @glmnet this is just that !!! Working for me ! Maybe update the documentation ?
I used on_turn_on because of the doc I think

Chuckame, 2020-03-24:
For information, here my new configuratin:

```
remote_transmitter:
  pin: D6
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "Home cinema power"
    id: home_cinema_power
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0x4005
          command: 0x01FE
```

closed 2020-03-24 by Chuckame