esphome/issues #1083: Issues with time component(s)

marcelveldt, 2020-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io/Homeassistant with esphome add-on.
Homeassistant version 106.5

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 Nodemcu v3

**Affected component:**
Time component, both sntp and homeassistant

**Description of problem:**

homeassistant time component does not work at all. The current time seems to be the time since boot or something.

sntp time is off by one hour. Tried with both setting the timezone and without.

**Problem-relevant YAML-configuration entries:**
```yaml
time:
  - platform: sntp
    id: sntp_time

```

**The issue can be seen with simply logging the time with strftime**
```
using strftime with home_assistant time now():
[09:08:53][I][current time:274]: 01:05

using strftime with sntp time now():
[09:14:39][I][current time:279]: 08:14

```

I tried all different timezone formats, tried switching between stable, beta and dev version of esphome.

marcelveldt, 2020-03-21:
home assistant time is stuck at January 1 1970

lolouk44, 2020-09-13:
Same issue for me. Time is 1h off (feels like UTC/GMT for me when I'm based in London and should be BST (=GMT+1)
If I add timezone, it's not taken into account (tried Europe/London and Europe/Paris as test)
Same issue or I'm doing something stupid?
```
time:
- platform: sntp
  id: sntp_time
  timezone: "Europe/London"

i2c:
  sda: D1
  scl: D2

display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x3F
    id: sainsmart
    lambda: |-
      auto date = id(sntp_time).now().strftime("%d %b %Y");
      auto time = id(sntp_time).now();
      it.printf("%s %02d:%02d:%02d %d",date.c_str(),time.hour,time.minute,time.second, time.is_dst);
```

randybb, 2020-09-13:
This has been fixed in dev https://github.com/esphome/esphome/pull/1266

glmnet, 2020-09-14:
I've missed the fix in v1.15.0 but it's coming in v1.15.1 soon

lolouk44, 2020-09-14:
Thanks both.
I can confirm that the code change in the PR works for me too 👍

closed 2020-09-14 by marcelveldt

tubalainen, 2020-09-14:
This issue was not solved or included in the 1.15.0 release. - So "timezone" does not work in 1.15.0.

Shoud we open a new issue?

https://github.com/esphome/esphome/pull/1266