esphome/issues #1086: Changing bightness on an effect doesn't change the white channel brightness

Skaronator, 2020-03-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome is installed in Docker.
ESPHome 1.14.3
Home Assistant 0.106.6

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 with SK6812 RGBW Led Strip

**Affected component:**
light, neopixelbus

**Description of problem:**

When the light is running a effect, changing the brightness in Home Assistant only take effect for RGB but not for the white channel. Additionally the white value slider does nothing.

As you can see here, when changing the brightness to 0 the white channel continues to stay on.

[image]
[image]

Not sure what the best option fix for this would be. Either make it that the white value slider works in this case or (even better IMO) apply the brightness for the white channel as well since its all part of the same effect.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: neopixelbus
    variant: SK6812
    type: grbw
    pin: 26
    num_leds: 144
    name: "${friendly_name}"
    gamma_correct: 1.5
    effects:
      - addressable_lambda:
          name: Debugger
          lambda: |-
            it.all() = ESPColor(255, 0, 0, 50);

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

farbefreak, 2020-04-22:
Seems like its the same bug as : #947 

I did try the dev build with the fix, this one handles the color different but still dosed work.

If you set a color in Homeassistant WW/CW turns off completely 
like wise If you set WW/CW brightness and choose a RGB color the WW/CW turns off completely 
so its impossible to mix RGB with WW/CW

jimmycleuren, 2020-12-27:
The same issue happens when using the rest api. Everything is fine when only using the rgb channel and leaving white_value at 0. From the moment you specify white_value and brightness together, the result is always that the white channel is at 100%.

oxan, 2021-07-10:
This should be fixed with esphome/esphome#1895 merged, available in the upcoming v1.20 release.

closed 2021-07-10 by oxan