esphome/issues #1087: Comments in YAML lambdas cause weird issues

fake-name, 2020-03-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

```
odroid@odroid:~/Programming/ICScraper$ uname -a
Linux odroid 4.14.165-172 #1 SMP PREEMPT Wed Jan 15 20:20:27 -03 2020 armv7l armv7l armv7l GNU/Linux
odroid@odroid:~/Programming/ICScraper$ python3 --version
Python 3.6.9
odroid@odroid:~/Programming/ICScraper$ pip --version
pip 19.3.1 from /usr/local/lib/python3.6/dist-packages/pip (python 3.6)
odroid@odroid:~/Programming/ICScraper$ esphome version
Version: 1.14.3
```

 

**Affected component:**

YAML parsing somewhere.

**Description of problem:**

Basically, from what I can tell, the `id()` directives in YAML lambdas are resolved by python code before compilation in a manner that's completely unaware of both YAML and C(++) comments. 

As such, if you have *commented out* `id(invalid_ref)` in a lambda, the code generation process fails.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test"
    password: "testtest"

uart:
  - id: nextuart
    rx_pin: GPIO36
    tx_pin: GPIO4
    baud_rate: 9600

display:
  - platform: nextion
    uart_id: nextuart
    # full_update_every : 30
    lambda: |-
      it.set_component_text_printf("t22", "%.2f", 5);
      # // voff = voff + vstep;
      # // it.printf( 0, voff, id(display_font), "Input: %.1f V",    (input_voltage_sensor)       ->state);

```

**Logs (if applicable):**

```
1 odroid@odroid:~/Programming/ICScraper$ esphome test.yaml compile
INFO Reading configuration test.yaml...
Failed config

display.nextion:
  platform: nextion
  uart_id: nextuart

  Couldn't find ID 'display_font'. Please check you have defined an ID with that name in your configuration.
  lambda: !lambda |-
    it.set_component_text_printf("t22", "%.2f", 5);
    # // voff = voff + vstep;
    # // it.printf( 0, voff, id(display_font), "Input: %.1f V",    (input_voltage_sensor)       ->state);
  update_interval: 5s
```

The printf causing the error is commented out with both C++ AND YAML comments, and yet it still causes failures.

patfelst, 2020-03-17:
if you outdent the lines with # to the same level as the "lambda" statement, they will be ignored. Alternatively, if you leave the current indent, then remove the # leaving the C++ comments //, these should also be ignored

patfelst, 2020-03-17:
by the way, if you use the MS Visual Studio Code, the color syntax highlighting will show you when the lines are being interpreted as YAML or C++
[image]

glmnet, 2020-03-17:
This is YAML, which is nice but not perfect. The issue is explained and workarounds provided.

fake-name, 2020-03-18:
The lambda is C++. Why do C++ comments not work?

brandond, 2020-03-18:
There are three layers here - yaml, python, and c++. You can comment using either, but as you noted, if you use a c style comment, it still needs to make it past the python id rewriting, as the python simply does string replacement within the code block. Implementing a full parser that respected comments would be possible, but is a lot of work just to support a corner case for unused code.

fake-name, 2020-03-18:
Ok, so esphome *is* doing rewriting of the C++?

Just from looking at it's use, I initially assumed it was a proper C++ function. Why on earth was the function invocation operation overloaded for even more stuff? It'd be much nicer to use notation that is unambiguously invalid C++.

If there's a bizarre additional preprocessor that's rewriting stuff, that it exists and what, exactly can trigger it is apparently undocumented. https://esphome.io/guides/automations.html says lambdas are just C++, and my reading of it was that the `id()` function was a C++ function.

glmnet, 2020-03-18:
Ahhh yes, TBH I just looked into your code example and see the # comments at the same level of the lambda code block, and of course it will not work, now there is ANOTHER issue which is the id() being preprocessed by ESPHome and failing because you might have removed the id, as you already know.

Now the real issue here is the lamba processor picking ids from c++ comments

Unfortunately I'm not [python ](https://github.com/esphome/esphome/blob/4cb30a22ac804124e2198784e34bd6c575ed5d96/esphome/core.py#L240) ninja so I'll leave it to sb else.

glmnet, 2020-03-19:
Now I’m tempted to use my copy paste code skills 

https://stackoverflow.com/questions/241327/remove-c-and-c-comments-using-python

fake-name, 2020-03-19:
To be fair, a lot of the stuff you find for this is about *parsing* C++, when all that's really needed here is to *lex* C++, which is a much more limited challenge.

glmnet, 2020-03-20:
Why not striping comments out before splitting the parts? That makes much more sense to me. That way multi line comments can be sorted out easily.

fake-name, 2020-03-20:
Because I didn't think too hard about where to do things?

closed 2020-04-06 by glmnet