esphome/issues #1090: ESP intermittently crashes on WiFi connection

vks007, 2020-03-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker , ESPHome Ver : 1.14.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini
**Affected component:**

[WiFi Component](https://esphome.io/components/wifi.html) (not sure)

**Description of problem:**
My ESP has been crashing for some time. It works for as long as I don't restart it. When it restarts it may or may not boot up well. It keeps on crashing while connecting to WiFi and then enters the safe mode. At times it works fine. I really don't have a way to tell when would it start crashing. 

**Problem-relevant YAML-configuration entries:**
```yaml
# Feature List DONE:
# 1. Turns OFF the motor if the ESP  loses connection to the HA API. An important point to note here is that ESPHome does not
#   know which API client is connected to it. So if a API client to see the log messages is connected to ESPHome
#   it will continue to assume it has a connection. I might want to enhance this in future where I would specifically look
#   for some timestamp or counter form HA to ensure I have a connection to HA or not
# 2. Turns OFF the motor if the motor has been running for more than X min (presently configured as 45)

substitutions:
  relay_gpio: GPIO14
  status_gpio: GPIO15
  st_sp_gpio: GPIO2
  mode_gpio: GPIO12
  sump_gpio: GPIO13
  static_ip: 192.168.1.63
  max_disconnect_time: '15' # max time of disconnection from the api , take action after this
  api_scan_interval: '5' #time in sec to scan for HA api connection

globals:
   - id: api_disconnect_time #time in sec for which ESPHome is disconnected from HA API
     type: int
     restore_value: no
     initial_value: '0'

esphome:
  name: tank_controller
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret secondary_ssid
  password: !secret secondary_ssid_pswd
  manual_ip:
    static_ip: $static_ip
    gateway: !secret default_gateway
    subnet: !secret subnet_mask
  fast_connect: true
#  output_power: 17db
  
# Enable logging
logger:
  level: VERBOSE

api:
  reboot_timeout: 10 min

ota:
  password: !secret ota_pswd

status_led:
  pin: $status_gpio
  
switch:
  - platform: gpio
    name: "Motor Relay"
    id: "motor_relay"
[... 335 lines cut ...]
0x4025bf29: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4025bf39: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4025bf51: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4025bf59: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x40223870: int conststr::parseNthInteger6u>(char const (&) [6u], unsigned int) at ?? line ?
0x4021e5a4: loop_wrapper() at core_esp8266_main.cpp line ?
0x40101399: cont_wrapper at ?? line ?

```

**Additional information and things you've tried:**
I have tried the following but nothing has solved the issue:
1. I tried specifying fast_connect = true
2. I was using both MQTT and API , I removed MQTT altogether , it didnt solve the issue. A point to be noted is that when MQTT was in use it used to fail while connecting to MQTT and not to connecting WiFi. So the point of failure has moved.
3. I have tried changing the Wemos board , it didnt make a difference
4. I have tried specifying the output_power manually, it didnt make a difference.

I am at a point where I am not able to debug it any longer. Given this issue is intermittent, I have to try a fix and then wait for a long time making finding the solution very slow. I have hit a dead end now, somebody please suggest how to proceed.

vks007, 2020-03-19:
Some progress on the issue. I removed the code in the interval section and the ESP has been running without crashing for 3 days now. Although not sure why would that cause an issue. 
Also, I dont know how to do what I was doing in the interval section elsewhere in the code.

glmnet, 2020-07-30:
Ahh yes, I believe your issue is the same that happened to me last night...

I really don't know what is going on here, but not only the interval section is the culprint but the `delayed_on_off` is causing trouble sure there, the one with the long delay.

Try bring back the interval section and change 

```yaml

    filters:
      - delayed_on_off: 5s
```

to delayed_on or delayed_off what suits you better.

I'd need to setup a test environment for this.

htvekov, 2020-08-29:
Hi' guys.

Just wanted to share some of my personal experience regarding use of delayed on/off filters on template sensors and ESPHome stack trace errors.

I've spend last two days trying to figure out what is wrong with my Sonos display set up with a PIR sensor and two filter delays.
Have read up on numerous issues regarding ESP crashes and entering eternal bootloop during WiFi setup.
Error reporting seems to be 'all over the place' and not sharing identical specifics - besides the stack trace errors on ESPHome boot.

My display lambda is quite extensive and time consuming (0,25 sec. according to verbose log). So first step was to bump ic2 speed to 100khz. That helped, but still i couldn't run code with 2 binary template sensors with two different filter off delays. Rewrote/tweaked my switch action that triggered the template sensors and finally got ESPHome to boot almost 100% stable.
But I can't add an on_boot action to turn on the display on each boot without ESP crashes again. Removing one of my  template sensors and then it works ? As I need both my template sensors I had to tweak the code and instead add the turn on display command at my display page control code section.

Finally my current code is stable and boots on almost every attempt. But I have the feeling that I'm at 99,8% CPU usage first seconds after boot. Adding or moving anything in my code could easily cause ESP to crash on boot again.

Two simultaneous template sensors with delayed off schedules seems in most cases to be more than ESPHome can handle during boot ? I believe that ESPHome must have some core timing issues on boot. In my case if feels like to many things are initialized simultaneously on boot than ESPHome can handle.

If my code and error logs can be of any use trying to identify the problem, just let me know.

Ciao !

glmnet, 2020-10-02:
Please leave feedback if this PR is fixing the issue. Thanks

closed 2020-10-15 by glmnet

htvekov, 2020-12-14:
Hi' @glmnet 

Sorry about the really late reply 🙄😒
I've been somewhat out of the tinkering loop last two months or so...

Just tested some of my different Sonos display setups using ESPHome 1.16.0-dev
I'm afraid all, as previous,  still fail to boot, even if just one more sensor is added.
In my specific setup I can't see any change.

Let me know if I could help testing further

Ciao !

glmnet, 2020-12-14:
hmmm, your issue might be different than @vks007 could you open a new issue and post there the full yaml and logs?

htvekov, 2020-12-27:
Hi' @glmnet

Just opened a separate issue for my ESP 'overload' boot issue 👍

Merry X-mas ! 🎁🎄🍻