esphome/issues #1093: MQTT Discovery unique_id - not so unique and causes errors

vvv850, 2020-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker install
HW - Intel NUC
**ESP (ESP32/ESP8266, Board/Sonoff):**

Multiple ESP32
**Affected component:**

MQTT Client Component

**Description of problem:**
While testing various boards and functionalities I stumbled upon a quirk. 
The first hint that something was wrong was that after a HASS restart I received the following error:

```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 425, in _async_add_entity
    raise HomeAssistantError(msg)
homeassistant.exceptions.HomeAssistantError: Entity id already exists: sensor.outside_humidity. Platform mqtt does not generate unique IDs
```
The node was configured to subscribe to a topic and pull data from it while displaying on a HMI.

I searched every core file to find the duplicate entry but nothing caught my eye. After several ours of searching I noticed that two nodes published the same line in the discovery message:

`"unique_id":"ESPsensoroutside_humidity"`

I tried multiple MQTT Component configuration, but ESPHome doesn't seem to change the unique_id.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: tiny_pico_attic_sm_hmi
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xxxx"
  password: "xxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tiny Pico Attic Sm Hmi"
    password: "xxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Valeriu83!@#"

ota:
  password: "Valeriu83!@#"

mqtt:
  broker: nuc.hassio.ro
  username: tinypico04
  password: !secret tinypico04_password
  topic_prefix: first/second/third
  client_id: testid
  id: hopefullyunique111
  
uart:
  rx_pin: GPIO27
  tx_pin: GPIO15
  baud_rate: 9600

display:
  - platform: nextion
    lambda: |-
      it.set_component_text_printf("t0", "%.1f C", id(out_temp).state);
      it.set_component_text_printf("t1", "%.1f %%Rh", id(out_hum).state);
      it.set_component_text_printf("t2", "%.0f hPa", id(out_press).state);

sensor:
  - platform: mqtt_subscribe
    name: "Outside Temp"
    id: out_temp
    topic: tiny_pico_attic_sm_hmi/sensor/hmi_outside_temperature/state
  - platform: mqtt_subscribe
    name: "Outside Humidity"
    id: out_hum
    topic: tiny_pico_attic_sm_hmi/sensor/hmi_outside_humidity/state
  - platform: mqtt_subscribe
    name: "Outside Pressure"
    id: out_press
    topic: tiny_pico_attic_sm_hmi/sensor/hmi_outside_pressure/state

binary_sensor:
  - platform: nextion
    page_id: 1
    component_id: 2
    name: "Hue 1 dec"
  - platform: nextion
    page_id: 1
    component_id: 3
    name: "Hue 1 Touch"
  - platform: nextion
    page_id: 1
    component_id: 4
    name: "Hue 1 inc"
  - platform: nextion
    page_id: 1
    component_id: 5
    name: "Hue 2 dec"
  - platform: nextion
    page_id: 1
    component_id: 6
    name: "Hue 2 Touch"
  - platform: nextion
    page_id: 1
    component_id: 7
    name: "Hue 2 inc"
  - platform: nextion
    page_id: 1
    component_id: 8
    name: "Hue 3 dec"
  - platform: nextion
    page_id: 1
    component_id: 9
    name: "Hue 3 Touch"
  - platform: nextion
    page_id: 1
    component_id: 10
    name: "Hue 3 inc"
  - platform: nextion
    page_id: 1
    component_id: 11
    name: "Hue Strip1 dec"
  - platform: nextion
    page_id: 1
    component_id: 12
    name: "Hue Strip1 Touch"
  - platform: nextion
    page_id: 1
    component_id: 13
    name: "Hue Strip1 inc"

```

**Logs (if applicable):**

```
Using 'COM20' as serial port.
Showing logs:
[03:54:47]ets Jun  8 2016 00:22:57
[03:54:47]
[03:54:47]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[03:54:47]configsip: 188777542, SPIWP:0xee
[03:54:48]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[03:54:48]mode:DIO, clock div:2
[03:54:48]load:0x3fff0018,len:4
[03:54:48]load:0x3fff001c,len:1044
[03:54:48]load:0x40078000,len:8896
[03:54:48]load:0x40080400,len:5828
[03:54:48]entry 0x400806ac
[03:54:48][I][logger:156]: Log initialized
[03:54:48][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[03:54:48][I][app:028]: Running through setup()...
[03:54:48][C][uart:022]: Setting up UART...
[03:54:48][C][wifi:033]: Setting up WiFi...
[03:54:48][D][wifi:304]: Starting scan...
[03:54:50][D][wifi:319]: Found networks:

[03:54:51][I][wifi:423]: WiFi Connected!
[03:54:51][C][wifi:283]:   SSID: 'XXXXX'
[03:54:51][C][wifi:284]:   IP Address: 10.211.200.190
[03:54:51][C][wifi:286]:   BSSID: 78:8A:20:4C:A3:B5
[03:54:51][C][wifi:287]:   Hostname: 'tiny_pico_attic_sm_hmi'
[03:54:51][C][wifi:291]:   Signal strength: -50 dB ▂▄▆█
[03:54:51][C][wifi:295]:   Channel: 6
[03:54:51][C][wifi:296]:   Subnet: 255.255.255.0
[03:54:51][C][wifi:297]:   Gateway: 10.211.200.1
[03:54:51][C][wifi:298]:   DNS1: 10.211.200.1
[03:54:51][C][wifi:299]:   DNS2: 0.0.0.0
[03:54:51][D][wifi:432]: Disabling AP...
[03:54:51][C][ota:029]: Over-The-Air Updates:
[03:54:51][C][ota:030]:   Address: tiny_pico_attic_sm_hmi.local:3232
[03:54:51][C][ota:032]:   Using Password.
[03:54:51][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[03:54:51][C][api:022]: Setting up Home Assistant API server...
[03:54:51][C][mqtt:025]: Setting up MQTT...
[03:54:52][D][mqtt:103]: Resolving MQTT broker IP address...
[03:54:53][D][mqtt:136]: Resolved broker IP address to 10.211.200.178
[03:54:53][I][mqtt:162]: Connecting to MQTT...
[03:54:56][I][mqtt:202]: MQTT Connected!
[03:54:56][I][app:058]: setup() finished successfully!
[03:54:56][I][app:100]: ESPHome version 1.14.3 compiled on Mar 17 2020, 03:30:54
[03:54:56][C][wifi:415]: WiFi:
[03:54:56][C][wifi:283]:   SSID: 'XXXXX'
[03:54:56][C][wifi:284]:   IP Address: 10.211.200.190
[03:54:56][C][wifi:286]:   BSSID: 78:8A:20:4C:A3:B5
[03:54:56][C][wifi:287]:   Hostname: 'tiny_pico_attic_sm_hmi'
[03:54:56][C][wifi:291]:   Signal strength: -51 dB 
[03:54:56][C][wifi:295]:   Channel: 6
[03:54:57][C][wifi:296]:   Subnet: 255.255.255.0
[03:54:57][C][wifi:297]:   Gateway: 10.211.200.1
[03:54:57][C][wifi:298]:   DNS1: 10.211.200.1
[03:54:57][C][wifi:299]:   DNS2: 0.0.0.0
[03:54:57][C][uart:040]: UART Bus:
[03:54:57][C][uart:042]:   TX Pin: GPIO15
[03:54:57][C][uart:045]:   RX Pin: GPIO27
[03:54:57][C][uart:047]:   Baud Rate: 9600 baud
[03:54:57][C][uart:048]:   Stop bits: 1
[03:54:57][C][logger:175]: Logger:
[03:54:57][C][logger:176]:   Level: DEBUG
[03:54:57][C][logger:177]:   Log Baud Rate: 115200
[03:54:57][C][logger:178]:   Hardware UART: UART0
[03:54:57][C][captive_portal:169]: Captive Portal:
[03:54:57][C][ota:029]: Over-The-Air Updates:
[03:54:57][C][ota:030]:   Address: tiny_pico_attic_sm_hmi.local:3232
[03:54:57][C][ota:032]:   Using Password.
[03:54:57][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[03:54:57][C][api:095]: API Server:
[03:54:57][C][api:096]:   Address: tiny_pico_attic_sm_hmi.local:6053
[03:54:57][C][mqtt:051]: MQTT:
[03:54:57][C][mqtt:053]:   Server Address: XXXXXX (10.211.200.178)
[03:54:57][C][mqtt:054]:   Username: 'tinypico04'
[03:54:57][C][mqtt:055]:   Client ID: 'testid'
[03:54:57][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[03:54:57][C][mqtt:058]:   Discovery retain: YES
[03:54:57][C][mqtt:060]:   Topic Prefix: 'first/second/third'
[03:54:57][C][mqtt:062]:   Log Topic: 'first/second/third/debug'
[03:54:57][C][mqtt:065]:   Availability: 'first/second/third/status'

```

**Additional information and things you've tried:**

It's clear that my mistake in not following a naming convention triggered this, but I cannot seem to fix it. 

I tried to delete the node and rename it. 
Deleted the MQTT Broker (although now it's clear that the broker isn't the culprit).
Changed the subscribed topic sensor id (Sensor > platform: mqtt_subscribe), but the Unique_id stays the same.

brandond, 2020-03-17:
You need to clean MQTT discovery data. For best results, run the clean using the original config yaml that was causing the conflict, then rebuild and reflash one of the nodes via UART (not via OTA) in order to avoid having it come online and republish.

https://esphome.io/guides/cli.html#clean-mqtt-command

vvv850, 2020-03-17:
Hi brandond,

I forgot to mention that with each MQTT test I manually deleted the messages and topics corresponding to the node. Even so, the problem isn't the broker and the retained messages, but the node that sends the same MQTT discovery message like this:

```
{
"name":"Outside testhum","state_topic":"first/second/third/sensor/outside_humidy/state",
"availability_topic":"first/second/third/status",
"unique_id":"ESPsensoroutside_humidy",
"device":{
"identifiers":"d8a01d471f90",
"name":"tiny_pico_attic_sm_hmi",
"sw_version":"esphome v1.14.3 Mar 17 2020, 04:01:59",
"model":"Espressif ESP-WROVER-KIT",
"manufacturer":"espressif"
}
}
```

Anyway, I managed to fix it just now by changing the MQTT sensor "friendly" name. It seems that in this case, ESPHome ignore the manually set ID for this type of sensor and constructs the ID from the friendly name:
Before:
```
  - platform: mqtt_subscribe
    name: "Outside Humidity"
    id: out_hum
    topic: tiny_pico_attic_sm_hmi/sensor/hmi_outside_humidity/state
```
After:
 ```
 - platform: mqtt_subscribe
    name: "Outside testhum"
    id: out_hum
    topic: tiny_pico_attic_sm_hmi/sensor/hmi_outside_humidity/state
```

As you can see, even though I have a unique ID for the problem sensor, ESPHome sends the discovery message using the friendly name.

I am curios if this is by design.

Cheers

brandond, 2020-03-17:
Yes, the id is used only for code generation and lambdas. The mqtt id is generated from the name string, hence my recommendation to change the sensor name. This is by design.

vvv850, 2020-03-17:
Sorry for not reading carefully. I haven't noticed the sensor name change recommendation.

It's clear now.

closed 2020-03-17 by vvv850