esphome/issues #1094: Wattage calculation incorrect

dirkoswanepoel, 2020-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Pow R2

**Affected component:**

https://esphome.io/components/sensor/cse7766.html

**Description of problem:**

If we use the first line as an example, it assumes the voltage to be 
From the logs it appears that it assumes the voltage to be somewhere around 100V. Assuming the voltage and current reading is correct, the wattage should be 203.17052W, but it is read as 89.43143W.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: cabinet
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "### SSID ###"
  password: "### PWD ###"
  reboot_timeout: 0s
  fast_connect: true

# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 4800

# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:

sensor:
  - platform: cse7766
    current:
      name: "Cabinet Current"
    voltage:
      name: "Cabinet Voltage"
    power:
      name: "Cabinet Power"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: cabinet_button
    on_press:
      - switch.toggle: relay_1

switch:
  - platform: gpio
    name: "Cabinet"
    pin: GPIO12
    id: relay_1

```

**Logs (if applicable):**

```
[17:49:52][D][cse7766:147]: Got voltage=230.7V current=0.9A power=89.4W
[17:49:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.68161 V with 1 decimals of accuracy
[17:49:52][D][sensor:092]: 'Cabinet Current': Sending state 0.88074 A with 2 decimals of accuracy
[17:49:52][D][sensor:092]: 'Cabinet Power': Sending state 89.43143 W with 1 decimals of accuracy
[17:50:52][D][cse7766:147]: Got voltage=230.9V current=0.9A power=90.0W
[17:50:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.87500 V with 1 decimals of accuracy
[17:50:52][D][sensor:092]: 'Cabinet Current': Sending state 0.88600 A with 2 decimals of accuracy
[17:50:52][D][sensor:092]: 'Cabinet Power': Sending state 89.99859 W with 1 decimals of accuracy
[17:51:52][D][cse7766:147]: Got voltage=230.9V current=0.9A power=89.4W
[17:51:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.90398 V with 1 decimals of accuracy
[17:51:52][D][sensor:092]: 'Cabinet Current': Sending state 0.88677 A with 2 decimals of accuracy
[17:51:52][D][sensor:092]: 'Cabinet Power': Sending state 89.40731 W with 1 decimals of accuracy
[17:52:52][D][cse7766:147]: Got voltage=230.8V current=0.9A power=89.3W
[17:52:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.81093 V with 1 decimals of accuracy
[17:52:52][D][sensor:092]: 'Cabinet Current': Sending state 0.88384 A with 2 decimals of accuracy
[17:52:52][D][sensor:092]: 'Cabinet Power': Sending state 89.25735 W with 1 decimals of accuracy
[17:53:52][D][cse7766:147]: Got voltage=230.9V current=0.9A power=89.6W
[17:53:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.89149 V with 1 decimals of accuracy
[17:53:52][D][sensor:092]: 'Cabinet Current': Sending state 0.87638 A with 2 decimals of accuracy
[17:53:52][D][sensor:092]: 'Cabinet Power': Sending state 89.60061 W with 1 decimals of accuracy
[17:54:52][D][cse7766:147]: Got voltage=230.9V current=0.9A power=89.0W
[17:54:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.87047 V with 1 decimals of accuracy
[17:54:52][D][sensor:092]: 'Cabinet Current': Sending state 0.85762 A with 2 decimals of accuracy
[17:54:52][D][sensor:092]: 'Cabinet Power': Sending state 89.02879 W with 1 decimals of accuracy
[17:55:52][D][cse7766:147]: Got voltage=230.8V current=0.9A power=89.1W
[17:55:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.75470 V with 1 decimals of accuracy
[17:55:52][D][sensor:092]: 'Cabinet Current': Sending state 0.85568 A with 2 decimals of accuracy
[17:55:52][D][sensor:092]: 'Cabinet Power': Sending state 89.08366 W with 1 decimals of accuracy
[17:56:52][D][cse7766:147]: Got voltage=230.6V current=0.9A power=89.3W
[17:56:52][D][sensor:092]: 'Cabinet Voltage': Sending state 230.64859 V with 1 decimals of accuracy
[17:56:52][D][sensor:092]: 'Cabinet Current': Sending state 0.87970 A with 2 decimals of accuracy
[17:56:52][D][sensor:092]: 'Cabinet Power': Sending state 89.25301 W with 1 decimals of accuracy
```

**Additional information and things you've tried:**

brandond, 2020-03-17:
Voltage, current, and power are all read from the chip via serial UART, not calculated in software.

https://github.com/esphome/esphome/blob/dev/esphome/components/cse7766/cse7766.cpp#L84

glmnet, 2020-03-18:
Docs says **active power**. Now, I’m not expert but watts is not Volt x Amps in ac loads with mix of active and reactive types

closed 2020-03-18 by glmnet