esphome/issues #1095: Yaml file got overridden

Lumpusz, 2020-03-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io from Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP 32 wrover kit
**Affected component:**

Yaml editor

**Description of problem:**
I was editing and uploading configuration repeatedly as I was doing some refactoring, and one time the config got overridden with the load text, and I lost it entirely. I don't remember explicitly hitting save, I just hit "Edit" from the upload dialog then I quickly hit "Upload" from the edit dialog, as the errors were in an included '.h' file. Now it just has this:
```
Loading configuration yaml...
```
Whenever the yaml file is being loaded, no actions should be allowed. Especially not this, as one can loose the entire config for good.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jonathanadams, 2020-04-29:
It's likely this has been fixed in esphome/esphome#1025.

glmnet, 2020-04-30:
Yes, as a lot of changes have been made to the dashboard by @jonathanadams we better close this, thanks for your report @Lumpusz if you see some strange behavior please report again.

closed 2020-04-30 by glmnet