esphome/issues #1098: OTA failing for NodeMCU ESP32

bangom, 2020-03-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

* Home Assistant: hassio on Tinkerboard version 0.107.4
* ESPHome version 1.14.3 as HA add-on

* Board: NodeMCU ESP32

**Affected component:**

OTA: https://esphome.io/components/ota.html

**Description of problem:**
After a successful firmware build, OTA fails (randomly at 1-50%) and ESP reboots. USB flashing is successful without any problems. OTA for other boards (ESP8266) in same esphome installation also works without the problem. 

**Problem-relevant YAML-configuration entries:**
```esphome:
  name: esp32_livingr_plant
  platform: ESP32
  board: nodemcu-32s
  includes:
    - tsl2591.h
  libraries:
    - "Adafruit TSL2591 Library"  
    - "Adafruit Unified Sensor"
      
esp32_ble_tracker:

wifi:
  ssid: "********"
  password: "********"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Livingr Plant"
    password: "*************"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "*************"

ota:
  password: "**************"

i2c:
  - id: bus_a
    sda: 16
    scl: 17
    scan: True
    frequency: 100kHz
  - id: bus_b
    sda: 21
    scl: 22
    scan: True
    frequency: 200kHz
    
time:
  - platform: homeassistant
    id: esptime

switch:
  - platform: restart
    name: "esp32_livingr_plant Restart"
  
binary_sensor:
  - platform: ble_presence
    mac_address: ******************
    name: "xxxxxxxxxxxx"
    id: ble_xxx
    device_class: presence

sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new Tsl2591();
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      name: "Room Ambient Light" 
      unit_of_measurement: lux
      id: tsl2891_light

  - platform: bmp085
    temperature:
      name: "Room Temperature BMP085"
      id: bmp085_temp
    pressure:
      name: "Room Pressure"
      id: bmp085_pressure
    update_interval: 60s
    address: 0x77
    i2c_id: bus_a
    
  - platform: htu21d
    temperature:
      name: "Room Temperature HTU21D"
      id: htu21d_temp
    humidity:
      name: "Room Humidity"
      id: htu21d_humidity
    update_interval: 60s
    i2c_id: bus_a

font:
  - file: 'arial.ttf'
    id: font_header
    size: 10
  - file: 'arial.ttf'
    id: font_main
    size: 12
  - file: 'arial.ttf'
    id: font_clock
    size: 10

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
    i2c_id: bus_b
    lambda: |-
      // Print "ESPHome" in top center.
      it.printf(64, 0, id(font_header), TextAlign::TOP_CENTER, "ESPHome");
      it.line(26, 12, 102, 12);

      // Print time in HH:MM format
      it.strftime(128, 64, id(font_clock), TextAlign::BASELINE_RIGHT, "%H:%M", id(esptime).now());
      
      if (id(bmp085_temp).has_state() && id(bmp085_pressure).has_state()  ) {
        it.printf(64, 20, id(font_main), TextAlign::TOP_CENTER , "T:%.1f°C P:%.1f hPa", id(bmp085_temp).state, id(bmp085_pressure).state  );
      }
      
      if (id(htu21d_humidity).has_state() && id(tsl2891_light).has_state() ) {
        it.printf(64, 36, id(font_main), TextAlign::TOP_CENTER , " H:%.1f%% L:%.0f lux ", id(htu21d_humidity).state, id(tsl2891_light).state );
      }
```

**Logs (if applicable):**

esphome build log:
```
INFO Reading configuration /config/esphome/esp32_livingr_plant.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 29 March 02:00:00 to 25 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32_livingr_plant
Processing esp32_livingr_plant (board: nodemcu-32s; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Adafruit TSL2591 Library> 1.2.1
|   |-- <Adafruit Unified Sensor> 1.1.2
|   |-- <Wire> 1.0.1
|-- <Adafruit Unified Sensor> 1.1.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/src/main.cpp.o
Linking /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/firmware.elf
Retrieving maximum program size /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/firmware.elf
Building /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/firmware.bin
Checking size /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/firmware.elf
DATA:    [==        ]  18.9% (used 61948 bytes from 327680 bytes)
PROGRAM: [========= ]  94.2% (used 1543722 bytes from 1638400 bytes)
========================= [SUCCESS] Took 88.03 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esp32_livingr_plant.local
INFO  -> a.b.c.d
INFO Uploading /data/esp32_livingr_plant/.pioenvs/esp32_livingr_plant/firmware.bin (1543840 bytes)
Uploading: [=======================                                     ] 38% 
ERROR Error sending data: [Errno 32] Broken pipe
```

ESP32 serial port log during failed OTA:
```
20:02:41][D][sensor:092]: 'Room Ambient Light': Sending state 154.00000 lux with 0 decimals of accuracy
[20:02:46][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:xx:xx:xx:xx):
[20:02:46][D][xiaomi_ble:173]:   Illuminance: 136lx
[20:02:54][D][ota:072]: Starting OTA Update from a.b.c.d...
[20:02:54][D][ota:243]: OTA in progress: 0.1%
[20:02:55][D][ota:243]: OTA in progress: 1.9%
[20:02:56][D][ota:243]: OTA in progress: 2.4%
[20:02:57][D][ota:243]: OTA in progress: 7.8%
[20:03:04][D][ota:243]: OTA in progress: 8.2%
[20:03:08][D][ota:243]: OTA in progress: 8.5%
[20:03:09][D][ota:243]: OTA in progress: 13.9%
[20:03:10][D][ota:243]: OTA in progress: 15.1%
[20:03:11][D][ota:243]: OTA in progress: 17.9%
[20:03:14][D][ota:243]: OTA in progress: 22.2%
[20:03:23][D][ota:243]: OTA in progress: 22.6%
[20:03:24][D][ota:243]: OTA in progress: 22.9%
[20:03:26][D][ota:243]: OTA in progress: 24.2%
[20:03:32][D][ota:243]: OTA in progress: 24.6%
[20:03:41]/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!
[20:03:41]abort() was called at PC 0x4008e8e7 on core 0
[20:03:41]
[20:03:41]Backtrace: 0x400928e8:0x3ffeadc0 0x40092b19:0x3ffeade0 0x4008e8e7:0x3ffeae00 0x40121c89:0x3ffeae20 0x40121cd6:0x3ffeae40 0x400dfea8:0x3ffeae60 0x40084d5e:0x3ffeaea0 0x400ffa35:0x3ffeaef0 0x40101ac1:0x3ffeaf30 0x40146fad:0x3ffeaf80 0x4014c025:0x3ffeafc0 0x4015130e:0x3ffeafe0 0x4013a797:0x3ffeb000 0x4008effd:0x3ffeb030
[20:03:41]
[20:03:41]Rebooting...
[20:03:41]ets Jun  8 2016 00:22:57
[20:03:41]
[20:03:41]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:03:41]configsip: 0, SPIWP:0xee
[20:03:41]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:03:41]mode:DIO, clock div:2
[20:03:41]load:0x3fff0018,len:4
[20:03:41]load:0x3fff001c,len:1044
[20:03:41]load:0x40078000,len:8896
[20:03:41]load:0x40080400,len:5828
[20:03:41]entry 0x400806ac
[20:03:42][I][logger:156]: Log initialized
[20:03:42][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[20:03:42][I][app:028]: Running through setup()...
```

**Additional information and things you've tried:**
* Wifi signal OK: [20:03:47][C][wifi:291]:   Signal strength: -52 dB ▂▄▆█
* Rebooting Home Assistant - no change.
* Rebooting ESP32 - no change
* Another powersupply for ESP32 - no change
* Measuring ESP32 ICMP reply - no packet-loss between ESP32 and esphome.

ryrzy, 2020-03-23:
i have exactly that same problem with one wemos d1 mini (didnt try on other devices to not kill them ;) ). I use nevest dev branch

wimpie007, 2020-03-23:
same problem here with an ESP32 and OTA update from esphome-dev. (hassio)
8266 ota updates work.

krupaldesai, 2020-03-31:
Hello all, I was facing the same issue on my ESP32 MH ET Live DevKit board. When I enable BLE tracker OTA fails most of the time. Almost impossible to perform OTA. Rarely it worked.

Upon some research I encountered this document https://www.espressif.com/sites/default/files/documentation/ESP32_FAQs__EN.pdf. Under section 5.3.2 I read: 

> Please note that if Option “Software
> controls WiFi/Bluetooth coexistence” is enabled, the BLE scan interval shall not exceed
> 0x100 slots (about 160 ms). 

I'm not entirely sure if this is related but I tried below and it works for me:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 160ms
    window: 15ms
    duration: 60s
```

With the above configuration OTA works every time. I don't understand the values completely nor did I experiment with different values. Above worked for me in the first go so I let it stay. Give it a try.

Hope this helps! Thanks @OttoWinter for this incredible platform - esphome. It can create wonders.

wimpie007, 2020-03-31:
mmm, did't work for me unfortunalety...
problem stays the same.

krupaldesai, 2020-04-01:
> mmm, did't work for me unfortunalety...
> problem stays the same.

Huh.. I guess, I got excited too early. After 2-3 successful OTAs, now with above setup I am getting random errors.

Couple of observations:

- after flashing it once over USB, the first OTA always worked for me
- without `esp32_ble_tracker` specified, OTA always works
- on the latest stable version `1.14.3` I had always seen `[Errno 104] Connection reset by peer` but on the latest dev branch `1.15.0-dev` it is now random - so far came across `Error sending data: timed out` and `[Errno 104] Connection reset by peer`.

wimpie007, 2020-04-01:
yeah, without esp32_ble_tracker, it works every time.
One solution would be to disable bluetooth scanning when doing OTA.

bangom, 2020-04-01:
Do you get wifi beacon timeouts on ESP32? And what about the AP site? I get lot of ```B4:E6:2D:xx:zz:ee@ap-downstairs_b-g-n disconnected, received disassoc: no activity (4)``` on AP log. From my perspective it seems that ESP32 CPU is too busy and can't process wifi beacons, therefore wifi goes down (AP will disconnect ESP32).

bangom, 2020-04-01:
Confirming that disabling ```esp32_ble_tracker:``` fixes OTA failures, improves Wifi stability (connection stable even under heavy ping speed test)  and Wifi beacon timeouts. Will experiment with ```scan_parameters:```... some more info also here: https://github.com/esphome/esphome/pull/855.

bangom, 2020-04-07:
When ```esp32_ble_tracker:``` is enabled I get those errors:

With ESPHome version 1.14:
a) 
```
[22:11:44]/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!
[22:11:44]abort() was called at PC 0x4008e8e7 on core 0
[22:11:44]
[22:11:44]Backtrace: 0x400928e8:0x3ffec830 0x40092b19:0x3ffec850 0x4008e8e7:0x3ffec870 0x4011de51:0x3ffec890 0x4011de9e:0x3ffec8b0 0x400dfbf0:0x3ffec8d0 0x40084d5e:0x3ffec910 0x400fdb89:0x3ffec960 0x401404bd:0x3ffec9b0 0x40145535:0x3ffec9f0 0x4014a81e:0x3ffeca10 0x40136093:0x3ffeca30 0x4008effd:0x3ffeca60
[22:11:44]
[22:11:44]Rebooting...
[22:11:44]ets Jun  8 2016 00:22:57
[22:11:44]
[22:11:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:11:44]configsip: 0, SPIWP:0xee
[22:11:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:11:44]mode:DIO, clock div:2
[22:11:44]load:0x3fff0018,len:4
[22:11:44]load:0x3fff001c,len:1044
[22:11:44]load:0x40078000,len:8896
[22:11:44]load:0x40080400,len:5828
[22:11:44]entry 0x400806ac
```
b) 
```
[22:12:04][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.x.y)
[22:12:07][W][wifi_esp32:336]: Event: Disconnected ssid='********' bssid=6C:3B:6B:**:**:** reason='Beacon Timeout'
[22:12:07][W][wifi:100]: WiFi Connection lost... Reconnecting...
[22:12:07][W][wifi:488]: Restarting WiFi adapter...
```

With ESPhome dev:

a)
```
[23:12:37][D][ota:072]: Starting OTA Update from 192.168.xx.yy...
[23:12:40][E][WiFiClient.cpp:392] write(): fail on fd 55, 	
[23:12:47][W][ota:311]: Error client disconnected while receiving data!
[23:12:47][W][ota:126]: Auth: Reading cnonce failed!
[23:12:47][W][wifi_esp32:336]: Event: Disconnected ssid='*********' bssid=6C:3B:6B:**:**:** reason='Beacon Timeout'
[23:12:47][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.xx.yy)
```
b)
```
[16:33:30][D][sensor:092]: 'Miflora04 Illuminance': Sending state 187.00000 lx with 0 decimals of accuracy
[16:33:31][D][ota:072]: Starting OTA Update from 192.168.xx.yy...
[16:33:31][D][ota:243]: OTA in progress: 0.1%
[16:33:32][D][ota:243]: OTA in progress: 2.9%
[16:33:43][D][ota:243]: OTA in progress: 5.1%
[16:33:44][D][ota:243]: OTA in progress: 6.0%
[16:33:45][D][ota:243]: OTA in progress: 6.4%
[16:33:46][D][ota:243]: OTA in progress: 11.0%
[16:33:57][W][ota:321]: Timeout waiting for data!
[16:33:57][W][ota:278]: Update end failed! Error: No Error
[16:33:57]
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[16:33:57][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.xx.yy)
[16:33:57][W][esp32_ble_tracker:061]: Too many BLE events to process. Some devices may not show up.
[16:33:57][D][xiaomi_ble:161]: Got Xiaomi HHCCJCY01 (80:EA:CA:**:**:**):
[16:33:57][D][xiaomi_ble:176]:   Illuminance: 2050lx
```

Any idea how to fix this?

bangom, 2020-04-07:
@OttoWinter  could you please advise on this?

Found old issue https://github.com/esphome/issues/issues/29 so I've tried to compile with ```arduino_version: dev``` in yaml config:
```
esphome:
  name: esp32_livingr_plant
  platform: ESP32
  arduino_version: dev
```
but got error:

```
INFO Reading configuration /config/esphome/esp32_livingr_plant.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32_livingr_plant
Processing esp32_livingr_plant (board: nodemcu-32s; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/stage)
--------------------------------------------------------------------------------
PlatformManager: Installing platform-espressif32
git version 2.17.1
Cloning into '/root/.platformio/platforms/_tmp_installing-bma8sybd-package'...
warning: Could not find remote branch feature/stage to clone.
fatal: Remote branch feature/stage not found in upstream origin
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'feature/stage', 'https://github.com/platformio/platform-espressif32.git', '/root/.platformio/platforms/_tmp_installing-bma8sybd-package']
```

Have hassio with ESPHome 1.15.0-dev plugin.

Any idea? Thanks!

marrold, 2020-05-20:
I have the same issue with ble_presence enabled. It would be interesting if it could be automatically disabled pre-OTA update.

littlerain14, 2020-06-11:
> yeah, without esp32_ble_tracker, it works every time.
> One solution would be to disable bluetooth scanning when doing OTA.

Dear friend, How to disable bluetooth scanning without doing upload function?

puuu, 2020-07-03:
I also came across this issue. Using the `dev` branch and having `esp32_ble_tracker` enabled, I am unable to come over 5% with OTA.

But for me it works after setting `arduino_version: 1.0.2` at `esphome` in the yaml config.

`scan_parameters` in `esp32_ble_tracker` is not set. I am using the default values.

Maybe someone can test, if this solves the issue.

krupaldesai, 2020-07-08:
> But for me it works after setting `arduino_version: 1.0.2` at `esphome` in the yaml config.
> 
> Maybe someone can test, if this solves the issue.

Didn't work for me on latest ESPHome v1.14.5 :(. It still fails the OTA same way. I will try using `dev` branch build soon if that helps.

djmaze, 2020-10-10:
Having a similar problem with `esp32_ble_beacon` activated. Getting timeouts almost all the time when trying to upload OTA.

polluz, 2020-11-30:
Same problem HERE.

I have 2 esp32 dev board acting as RF and BLE gateways with mqtt (for RF sensors and BLE xiaomi thermometers).

HELP PLZ

RF SECTION:
```
remote_receiver:
  pin: 
    number: GPIO27
  dump: rc_switch
  filter: 4us
  tolerance: 50%
  idle: 4ms
  on_rc_switch:
    - logger.log:
        format: "on_rc_switch: %i %llx"
        args: [x.protocol, x.code]
    - lambda: !lambda |-
        char buffer [10];
        id(${devicename}).publish("${devicename}/rx", itoa(x.code,buffer,16));
```

BLE (tried parameters)
```
esp32_ble_tracker:
  scan_parameters:
    interval: 160ms
    window: 15ms
    duration: 60s
```

BLE SENSORS 
```
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:7D:DB:DC" 
    bindkey: "ccc"
    temperature:
      name: "${devicename}.ble.Bagno Temperatura"
    humidity:
      name: "${devicename}.ble.Bagno Umidità"
    battery_level:
      name: "${devicename}.ble.Bagno Batteria"
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:0C:C2:67" 
    bindkey: "aaa"
    temperature:
      name: "${devicename}.ble.Disimpegno Temperatura"
    humidity:
      name: "${devicename}.ble.Disimpegno Umidità"
    battery_level:
      name: "${devicename}.ble.Disimpegno Batteria"  
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:07:44:65" 
    bindkey: "xxx"
    temperature:
      name: "${devicename}.ble.Camera Lucia Temperatura"
    humidity:
      name: "${devicename}.ble.Camera Lucia Umidità"
    battery_level:
      name: "${devicename}.ble.Camera Lucia Batteria"    
```

deftdawg, 2021-01-11:
Maybe try setting  
```
wifi: 
 power_save_mode: none
```
If you haven't already

polluz, 2021-01-13:
Hi, thank you, Tried but no changes... :-( still failing ota. Tried various parameters. No way

eperdeme, 2021-05-21:
Sorry to reply to an old thread but though it might help.

My ESP32 uptime varied from few mins I think it's due to using Bluetooth + WIFI possibly. Tried various things but uptime was between 5-60mins but varied.

I've moved to a newer espressif32 lib now and it's rock solid. 18hours uptime and rising, it might be worth a try ? 

Key bit is
```
platform: espressif32@3.2.0
```
```
esphome:
  name: ${device}
  platform: ESP32
  board: esp32dev
  platformio_options: 
    platform: espressif32@3.2.0
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
```

Prior to this I was ready to ditch Esphome and go back to micropython as that seem to not suffer from these issues. Now I'm happy it finally works after many weeks of intermittency.

DrBlokmeister, 2021-06-13:
That didn't solve the issue for me. Not only is OTA not working, but the connection to wifi also drops out every now and then. Apperently it's still not reliable.

the78mole, 2021-10-03:
Same here. With BLE tracker active, OTA stops working realiably. 
[image]

On the serial console, I receive the following error during OTA:
[image]

I believe, it would work fine again, if BLE would be switched off during OTA.

robloh, 2021-10-08:
Same issues on ESP32 with ethernet. Very annoying to have to take grab devices and plug in to a computer every time you need to make a change (like adding another BLE device to scan for). BLE is a pretty big thing for ESP devices, any chance this could get addressed by disabling BLE scanning during upgrade @jesserockz @OttoWinter ?

puuu, 2021-10-09:
I also suffer from an OTA that does not work reliably when the EPS32 BLE Tracker component is activated.
I have a total of 6 EPS32s with epshome, 3 of them with BLE Tracker and these 3 all have problems with OTA (but in different strong appearances). 
Debugging is very difficult because it is very sporadic and usually only occurs after a few days.
Since my devices are scattered around my house and screwed into cases, it is difficult to connect a serial adapter.

Overall there seems to be a memory problem, e.g.: memory leak, memory fragmentation, ...

I do not have any special configuration in `esp32_ble_tracker`, `wifi` or `ota`.

Here are some workarounds that can help reduce the issues: 
1) Reduce memory consumption by disabling other (not just BLE) components 
2) If you are using the Remote Receiver component, reduce `buffer_size` (2 KB works fine for me even with the Daikin IR Remote Climate component) 
3) Try to reboot/power cylce the ESP32 before OTA. (The Restart Switch component may be helpful, but power cycle seems to be more efficient)
4) Resetting (Reset button/power cylce) multiple times to force safe mode.
5) Consider splitting your configuration across multiple ESP32s

I hope this might be helpful. 

For the developer (@jesserockz @OttoWinter):
I haven't looked at esphome's BLE component, but I've seen other projects favoring NimBLE over the Arduino/ESP-IDF BLE library. 
[They predict up to 100kB less RAM consumption.](https://github.com/h2zero/NimBLE-Arduino#resource-use-improvement).
Please consider switching to [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp).

jds11111, 2021-11-05:
And it seems to have gotten worse with 2021.10.x Devices that had been working fine for months are now unstable.

nagyrobi, 2022-03-24:
Follow-up from https://discord.com/channels/429907082951524364/956262153705373716

Decoded stack trace:
```
[19:47:33][D][ota:139]: Starting OTA Update from ::FFFF:AC16:D7FC...
[19:47:33][V][ota:170]: OTA features is 0x01
[19:47:33][V][ota:190]: Auth: Nonce is 6453e5edc330effcf5d04b6c23ace45c
[19:47:33][V][ota:210]: Auth: CNonce is 808fe8f2f2415fe3cd5298f4bd954e51
[19:47:33][V][ota:217]: Auth: Result is cd2258a2eb23b4faae6ee0e4f000170e
[19:47:33][V][ota:225]: Auth: Response is cd2258a2eb23b4faae6ee0e4f000170e
[19:47:33][V][ota:253]: OTA size is 1585536 bytes
[19:47:33][D][Updater.cpp:132] begin(): OTA Partition: app0
[19:47:33][V][ota:270]: Update: Binary MD5 is 607db74ea1e3d5c38573ee97fe927224
[19:47:33][D][ota:308]: OTA in progress: 0.1%
[19:47:34][D][ota:308]: OTA in progress: 3.6%
[19:47:35][D][ota:308]: OTA in progress: 7.3%
[19:47:36][D][ota:308]: OTA in progress: 10.9%
[19:47:37][D][ota:308]: OTA in progress: 14.5%
[19:47:38][D][ota:308]: OTA in progress: 18.4%
[19:47:39][D][ota:308]: OTA in progress: 22.0%
[19:47:40][D][ota:308]: OTA in progress: 25.6%
[19:47:41][D][ota:308]: OTA in progress: 29.2%
[19:47:42][D][ota:308]: OTA in progress: 32.8%
[19:47:43][D][ota:308]: OTA in progress: 36.4%
[19:47:44][D][ota:308]: OTA in progress: 40.1%
[19:47:46][D][ota:308]: OTA in progress: 43.7%
[19:47:47][D][ota:308]: OTA in progress: 47.3%
[19:47:48][D][ota:308]: OTA in progress: 50.9%
[19:47:49][D][ota:308]: OTA in progress: 54.5%
[19:47:50][D][ota:308]: OTA in progress: 58.2%
[19:47:51][D][ota:308]: OTA in progress: 61.8%
[19:47:52][D][ota:308]: OTA in progress: 65.6%
[19:47:53][D][ota:308]: OTA in progress: 69.5%
[19:47:54][D][ota:308]: OTA in progress: 73.1%
[19:47:55][D][ota:308]: OTA in progress: 76.8%
[19:47:56][D][ota:308]: OTA in progress: 80.6%
[19:47:59]abort() was called at PC 0x401e116b on core 0
[19:47:59]
[19:47:59]ELF file SHA256: 0000000000000000
[19:47:59]
[19:47:59]Backtrace: 0x4008f938:0x3ffd8170 0x4008fbb5:0x3ffd8190 0x401e116b:0x3ffd81b0 0x401e11b2:0x3ffd81d0 0x401e087b:0x3ffd81f0 0x401e0bb2:0x3ffd8210 0x400da431:0x3ffd8230 0x40130fb1:0x3ffd8250 0x4012c0d2:0x3ffd8270 0x40090c2a:0x3ffd82a0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008f938: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4008fbb5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x401e116b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e11b2: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e087b: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
WARNING Decoded 0x401e0bb2: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400da431: __gnu_cxx::new_allocator<esphome::esp32_ble_tracker::ServiceData>::allocate(unsigned int, void const*) at /config/esphome/.esphome/build/wt32-eth01-bletest/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:406
 (inlined by) std::allocator_traits<std::allocator<esphome::esp32_ble_tracker::ServiceData> >::allocate(std::allocator<esphome::esp32_ble_tracker::ServiceData>&, unsigned int) at /data/cache/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/alloc_traits.h:360
 (inlined by) std::_Vector_base<esphome::esp32_ble_tracker::ServiceData, std::allocator<esphome::esp32_ble_tracker::ServiceData> >::_M_allocate(unsigned int) at /data/cache/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:170
 (inlined by) void std::vector<esphome::esp32_ble_tracker::ServiceData, std::allocator<esphome::esp32_ble_tracker::ServiceData> >::_M_emplace_back_aux<esphome::esp32_ble_tracker::ServiceData const&>(esphome::esp32_ble_tracker::ServiceData const&) at /data/cache/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/vector.tcc:412
WARNING Decoded 0x40130fb1: btc_gap_ble_cb_to_app at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:1070
 (inlined by) btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:914
WARNING Decoded 0x4012c0d2: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c:163
WARNING Decoded 0x40090c2a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[19:48:00]
[19:48:00]Rebooting...
[19:48:00]ets Jun  8 2016 00:22:57
[19:48:00]
[19:48:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:48:00]configsip: 0, SPIWP:0xee
[19:48:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:48:00]mode:DIO, clock div:2
[19:48:00]load:0x3fff0018,len:4
[19:48:00]load:0x3fff001c,len:1044
[19:48:00]load:0x40078000,len:8896
[19:48:00]load:0x40080400,len:5828
[19:48:00]entry 0x400806ac
[19:48:00][I][logger:214]: Log initialized
[19:48:00][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[19:48:00][D][esp32.preferences:114]: Saving preferences to flash...
[19:48:00][I][app:029]: Running through setup()...
[19:48:00][V][app:030]: Sorting components by setup priority...
```

danijelt, 2023-06-29:
This appears to have been resolved on its own over time by updating espressif library in ESPHome. Following the suggestion [in this comment](https://github.com/esphome/issues/issues/1098#issuecomment-845789058) by @eperdeme, I had forced v5.3.0 for some time which solved the issue.

As v5.3.0 is now default in ESPHome, I decided to try without the workaround on Sonoff Dual R3 and OTA passed multiple times on firmware without platformio overrides.

I must note that I don't use BLE tracker so maybe there are still issues on devices which have it enabled.

Janhouse, 2023-06-29:
The whole point was about using BLE.

joe-sydney, 2023-06-29:
This stopped being an issue for me when I changed to from framework type arduino to edp-idf.
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf #arduino

oldjiberjaber, 2023-11-02:
This issue still exists, even with esp-idf as the framework. 

I'm getting a bit fed up of having to demount my devices and bring them to the study to flash them, surely there must be some sort of fix to disable BLE when the OTA is triggered and alow it to come back on when the device resets?

RAM:   [==        ]  16.6% (used 54236 bytes from 327680 bytes)
Flash: [========  ]  78.9% (used 1448229 bytes from 1835008 bytes)
Building .pioenvs/esphome-kitchen-323934/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esphome-kitchen-323934/firmware.bin"], [".pioenvs/esphome-kitchen-323934/firmware.elf"])
Wrote 0x172fa0 bytes to file /data/build/esphome-kitchen-323934/.pioenvs/esphome-kitchen-323934/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 426.80 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome-kitchen-323934.lan
INFO  -> 192.168.1.43
INFO Uploading /data/build/esphome-kitchen-323934/.pioenvs/esphome-kitchen-323934/firmware.bin (1453984 bytes)
Uploading: [===================================                         ] 58% 
ERROR Error sending data: [Errno 104] Connection reset by peer

joe-sydney, 2023-11-02:
The safe mode switch or button does that.
https://esphome.io/components/switch/safe_mode.html
https://esphome.io/components/button/safe_mode.html

I used them when I still had problems. Before doing the OTA update, use them to put the esp32 into safe mode.

oldjiberjaber, 2023-11-19:
Thanks for the suggestion, but I'm afraid that didn't work.
Device requested to be put in to Safe mode, booted and verified it was in safe mode, initiated an install OTA - time out again.
I've tried this on 2 different devices now, So still not working.  Getting a bit fed up of having to remove devices from their location in order to flash them.  I have an ESP8266 about 120ft away which accepts OTA with no problem, however the ESP32 devices running BT fail every time (Lolin and MHT devices)

**Log from ESP32 device:**
```
[16:44:05][D][sensor:094]: 'kitchen Ambient Light': Sending state 28.40000 lx with 2 decimals of accuracy
[16:44:13][D][button:010]: 'Safe Mode Boot' Pressed.
[16:44:13][I][safe_mode.button:014]: Restarting device in safe mode...
[16:44:13][I][app:134]: Rebooting safely...
[16:44:13][D][esp-idf:000]: I (495051) MQTT_CLIENT: Client asked to disconnect

INFO Processing expected disconnect from ESPHome API for esphome-kitchen-323934.lan
WARNING Disconnected from API
INFO Successfully connected to esphome-kitchen-323934.lan in 6.872s
INFO Successful handshake with esphome-kitchen-323934.lan in 2.312s
[16:44:28][C][i2c.idf:061]: I2C Bus:
[16:44:28][C][i2c.idf:062]:   SDA Pin: GPIO21
[16:44:28][C][i2c.idf:063]:   SCL Pin: GPIO22
[16:44:28][C][i2c.idf:064]:   Frequency: 50000 Hz
[16:44:28][C][i2c.idf:067]:   Recovery: bus successfully recovered
[16:44:28][I][i2c.idf:077]: Results from i2c bus scan:
[16:44:28][I][i2c.idf:083]: Found i2c device at address 0x45
[16:44:28][C][bluetooth_proxy:088]: Bluetooth Proxy:
[16:44:28][C][bluetooth_proxy:089]:   Active: YES
[16:44:28][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[16:44:28][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[16:44:28][C][sht3xd:033]: SHT3xD:
[16:44:28][C][sht3xd:034]:   Address: 0x45
[16:44:28][C][sht3xd:038]:   Update Interval: 60.0s
[16:44:28][C][sht3xd:040]:   Temperature 'Kitchen Temperature'
[16:44:28][C][sht3xd:040]:     Device Class: 'temperature'
[16:44:28][C][sht3xd:040]:     State Class: 'measurement'
[16:44:28][C][sht3xd:040]:     Unit of Measurement: '°C'
[16:44:28][C][sht3xd:040]:     Accuracy Decimals: 2
[16:44:28][C][sht3xd:041]:   Humidity 'kitchen Humidity'
[16:44:28][C][sht3xd:041]:     Device Class: 'humidity'
[16:44:28][C][sht3xd:041]:     State Class: 'measurement'
[16:44:28][C][sht3xd:041]:     Unit of Measurement: '%'
[16:44:28][C][sht3xd:041]:     Accuracy Decimals: 2
[16:44:28][C][adc:097]: ADC Sensor 'kitchen Ambient Light'
[16:44:28][C][adc:097]:   Device Class: 'voltage'
[16:44:28][C][adc:097]:   State Class: 'measurement'
[16:44:28][C][adc:097]:   Unit of Measurement: 'lx'
[16:44:28][C][adc:122]:  Attenuation: 11db
[16:44:28][C][esp32_ble:374]: ESP32 BLE:
[16:44:28][C][esp32_ble:376]:   MAC address: 30:AE:A4:32:39:36
[16:44:28][C][mdns:115]: mDNS:
[16:44:28][C][mdns:116]:   Hostname: esphome-kitchen-323934
[16:44:28][C][ota:097]: Over-The-Air Updates:
[16:44:28][C][ota:098]:   Address: esphome-kitchen-323934.lan:3232
[16:44:28][C][api:139]: API Server:
[16:44:28][C][api:140]:   Address: esphome-kitchen-323934.lan:6053
[16:44:28][C][api:142]:   Using noise encryption: YES
[16:44:28][C][mqtt:133]: MQTT:
[16:44:28][C][mqtt:135]:   Server Address: bb.lan:1883 (192.168.1.6)
[16:44:28][C][mqtt:136]:   Username: ''[redacted]
[16:44:28][C][mqtt:137]:   Client ID: 'esphome-kitchen-323934-30aea4323934'[redacted]
[16:44:28][C][mqtt:139]:   Discovery prefix: 'homeassistant'
[16:44:28][C][mqtt:140]:   Discovery retain: YES
[16:44:28][C][mqtt:142]:   Topic Prefix: 'esphome-kitchen-323934'
[16:44:28][C][mqtt:144]:   Log Topic: 'esphome-kitchen-323934/debug'
[16:44:28][C][mqtt:147]:   Availability: 'tele/kitchen/LWT'
[16:44:28][C][improv_serial:032]: Improv Serial:
[16:44:28][C][mqtt.button:029]: MQTT Button 'Safe Mode Boot': 
[16:44:28][C][mqtt.button:030]:   State Topic: 'esphome-kitchen-323934/button/safe_mode_boot/state'
[16:44:28][C][mqtt.button:030]:   Command Topic: 'esphome-kitchen-323934/button/safe_mode_boot/command'
[16:44:28][C][mqtt.sensor:028]: MQTT Sensor 'Kitchen Temperature':
[16:44:28][C][mqtt.sensor:032]:   State Topic: 'esphome-kitchen-323934/sensor/kitchen_temperature/state'
[16:44:28][C][mqtt.sensor:028]: MQTT Sensor 'kitchen Humidity':
[... 56 lines cut ...]
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling .pioenvs/esphome-kitchen-323934/src/main.o
Linking .pioenvs/esphome-kitchen-323934/firmware.elf
RAM:   [==        ]  16.6% (used 54260 bytes from 327680 bytes)
Flash: [========  ]  79.1% (used 1450941 bytes from 1835008 bytes)
Building .pioenvs/esphome-kitchen-323934/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esphome-kitchen-323934/firmware.bin"], [".pioenvs/esphome-kitchen-323934/firmware.elf"])
Wrote 0x173a30 bytes to file /data/build/esphome-kitchen-323934/.pioenvs/esphome-kitchen-323934/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 82.05 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome-kitchen-323934.lan
INFO  -> 192.168.1.43
INFO Uploading /data/build/esphome-kitchen-323934/.pioenvs/esphome-kitchen-323934/firmware.bin (1456688 bytes)
Uploading: [=============================                               ] 48% 
ERROR Error sending data: timed out
```

oldjiberjaber, 2023-11-29:
So I noticed the ability to turn BLE ON and OFF and placed that in to the OTA Start.  Hoping this had resolved the issue, I find whilst it now reports Stopping BLE scanning it still fails again around 50 to 70%

`
ota:
  on_begin:
    then:
      - logger.log: "OTA start"
      - ble.disable
`

> [21:01:11][D][sht3xd:064]: Got temperature=23.50°C humidity=39.35%
> [21:01:11][D][sensor:094]: 'Living Room Temperature': Sending state 21.49928 °C with 2 decimals of accuracy
> [21:01:11][D][sensor:094]: 'Living Room Humidity': Sending state 39.35302 % with 2 decimals of accuracy
> [21:01:20][D][sensor:094]: 'livingrm Ambient Light': Sending state 109.20000 lx with 2 decimals of accuracy
> [21:01:30][D][sensor:094]: 'livingrm Ambient Light': Sending state 106.60000 lx with 2 decimals of accuracy
> [21:01:32][D][ota:151]: Starting OTA Update from 192.168.1.6...
> [21:01:32][D][main:157]: OTA start
> [21:01:32][D][esp32_ble_tracker:233]: Stopping scan.
> [21:01:36][D][ota:320]: OTA in progress: 0.1%
> [21:01:37][D][ota:320]: OTA in progress: 7.9%
> [21:01:38][D][ota:320]: OTA in progress: 15.4%
> [21:01:39][D][ota:320]: OTA in progress: 22.9%
> [21:01:40][D][ota:320]: OTA in progress: 29.8%
> [21:01:41][D][ota:320]: OTA in progress: 36.9%
> [21:01:42][D][ota:320]: OTA in progress: 44.6%
> [21:01:43][D][ota:320]: OTA in progress: 51.8%
> WARNING esp32-bluetooth-proxy-322da0.lan: Connection error occurred: [Errno 104] Connection reset by peer
> INFO Processing unexpected disconnect from ESPHome API for esp32-bluetooth-proxy-322da0.lan
> WARNING Disconnected from API
> INFO Successfully connected to esp32-bluetooth-proxy-322da0.lan in 0.023s
> INFO Successful handshake with esp32-bluetooth-proxy-322da0.lan in 0.174s
> [21:02:11][D][sensor:094]: 'livingrm Ambient Light': Sending state 109.80000 lx with 2 decimals of accuracy
> [21:02:13][I][bluetooth_proxy:278]: [0] [26:9C:95:1B:D6:58] Connecting v3 with cache
> [21:02:13][D][esp32_ble_tracker:215]: Pausing scan to make connection...
> [21:02:13][I][esp32_ble_client:069]: [0] [26:9C:95:1B:D6:58] 0x00 Attempting BLE connection
> [21:02:14][I][esp32_ble_client:149]: [0] [26:9C:95:1B:D6:58] Connected
> [21:02:14][D][esp32_ble_tracker:266]: Starting scan...

> INFO ESPHome 2023.11.6
> INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-322da0.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing esp32-bluetooth-proxy-322da0 (board: mhetesp32minikit; framework: espidf; platform: platformio/espressif32@5.4.0)
> --------------------------------------------------------------------------------
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
>  - framework-espidf @ 3.40405.230623 (4.4.5) 
>  - tool-cmake @ 3.16.4 
>  - tool-ninja @ 1.7.1 
>  - toolchain-esp32ulp @ 2.35.0-20220830 
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> Reading CMake configuration...
> Dependency Graph
> |-- noise-c @ 0.1.4
> |-- ArduinoJson @ 6.18.5
> |-- Improv @ 1.2.3
> Compiling .pioenvs/esp32-bluetooth-proxy-322da0/src/main.o
> Linking .pioenvs/esp32-bluetooth-proxy-322da0/firmware.elf
> RAM:   [==        ]  16.6% (used 54300 bytes from 327680 bytes)
> Flash: [========  ]  79.4% (used 1456221 bytes from 1835008 bytes)
> Building .pioenvs/esp32-bluetooth-proxy-322da0/firmware.bin
> Creating esp32 image...
> Successfully created esp32 image.
> esp32_create_combined_bin([".pioenvs/esp32-bluetooth-proxy-322da0/firmware.bin"], [".pioenvs/esp32-bluetooth-proxy-322da0/firmware.elf"])
> Wrote 0x174ed0 bytes to file /data/build/esp32-bluetooth-proxy-322da0/.pioenvs/esp32-bluetooth-proxy-322da0/firmware-factory.bin, ready to flash to offset 0x0
> ======================== [SUCCESS] Took 175.13 seconds ========================
> INFO Successfully compiled program.
> INFO Resolving IP address of esp32-bluetooth-proxy-322da0.lan
> INFO  -> 192.168.1.17
> INFO Uploading /data/build/esp32-bluetooth-proxy-322da0/.pioenvs/esp32-bluetooth-proxy-322da0/firmware.bin (1461968 bytes)
> Uploading: [===============================                             ] 52% 
> ERROR Error sending data: [Errno 104] Connection reset by peer
> 

```