esphome/issues #1099: ESPHome Time Component - .timestamp updating every 128 seconds

ChrisKraw, 2020-03-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io - ESPhome 1.14.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - board esp01_1m
**Affected component:**

Time component, both sntp and homeassistant

**Description of problem:**
Time component .timestamp is updating its value every 128 seconds (I'm calling it evey 1 second)
I have tried both platform: homeassistant and sntp.  The  .second is working fine.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
`esphome:
  name: adcvoltagetest
  platform: ESP8266
  board: esp01_1m

# WiFi connection
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

    
# Enable logging
logger: 
  level: debug
  #Only needed on esp8266
  esp8266_store_log_strings_in_flash: False

# Enable Home Assistant API
api:
  password: !secret esphome_api_password

# Enable over-the-air updates
ota:
  password: !secret esphome_ota_password

time:
  - platform: homeassistant
    id: hastime

interval:
  - interval: 1s
    then:
      - lambda: |-
          auto time_t = id(hastime).utcnow().timestamp;
          auto time_s = id(hastime).utcnow().second;
          if (id(hastime).utcnow().is_valid()) {
            id(timestamp_sec).publish_state(time_t);
            id(time_seconds).publish_state(time_s);
          }

sensor: 

  - platform: template
    name: "Current time stamp"
    id: timestamp_sec
  
  - platform: template
    name: "Current time"
    id: time_seconds
    `
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
`[09:59:51][C][template.sensor:021]: Template Sensor 'Current time stamp'
[09:59:51][C][template.sensor:021]:   Unit of Measurement: ''
[09:59:51][C][template.sensor:021]:   Accuracy Decimals: 1
[09:59:51][C][template.sensor:022]:   Update Interval: 60.0s
[09:59:51][C][template.sensor:021]: Template Sensor 'Current time'
[09:59:51][C][template.sensor:021]:   Unit of Measurement: ''
[09:59:51][C][template.sensor:021]:   Accuracy Decimals: 1
[09:59:51][C][template.sensor:022]:   Update Interval: 60.0s
[09:59:51][C][logger:175]: Logger:
[09:59:51][C][logger:176]:   Level: DEBUG
[09:59:51][C][logger:177]:   Log Baud Rate: 115200
[09:59:51][C][logger:178]:   Hardware UART: UART0
[09:59:51][C][homeassistant.time:010]: Home Assistant Time:
[09:59:51][C][homeassistant.time:011]:   Timezone: 'EST5EDT4,M3.2.0/2,M11.1.0/2'
[09:59:51][C][ota:029]: Over-The-Air Updates:
[09:59:51][C][ota:030]:   Address: adcvoltagetest.local:8266
[09:59:51][C][ota:032]:   Using Password.
[09:59:51][C][api:095]: API Server:
[09:59:51][C][api:096]:   Address: adcvoltagetest.local:6053
[09:59:51][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:51][D][sensor:092]: 'Current time': Sending state 50.00000  with 1 decimals of accuracy
[09:59:52][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:52][D][sensor:092]: 'Current time': Sending state 51.00000  with 1 decimals of accuracy
[09:59:53][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:53][D][sensor:092]: 'Current time': Sending state 52.00000  with 1 decimals of accuracy
[09:59:54][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:54][D][sensor:092]: 'Current time': Sending state 53.00000  with 1 decimals of accuracy
[09:59:55][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:55][D][sensor:092]: 'Current time': Sending state 54.00000  with 1 decimals of accuracy
[09:59:56][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:56][D][sensor:092]: 'Current time': Sending state 55.00000  with 1 decimals of accuracy
[09:59:57][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:57][D][sensor:092]: 'Current time': Sending state 56.00000  with 1 decimals of accuracy
[09:59:58][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:58][D][sensor:092]: 'Current time': Sending state 57.00000  with 1 decimals of accuracy
[09:59:59][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[09:59:59][D][sensor:092]: 'Current time': Sending state 58.00000  with 1 decimals of accuracy
[10:00:00][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[10:00:00][D][sensor:092]: 'Current time': Sending state 59.00000  with 1 decimals of accuracy
[10:00:01][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[10:00:01][D][sensor:092]: 'Current time': Sending state 0.00000  with 1 decimals of accuracy
[10:00:02][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[10:00:02][D][sensor:092]: 'Current time': Sending state 1.00000  with 1 decimals of accuracy
[10:00:03][D][sensor:092]: 'Current time stamp': Sending state 1584972032.00000  with 1 decimals of accuracy
[10:00:03][D][sensor:092]: 'Current time': Sending state 2.00000  with 1 decimals of accuracy

... cont 

[10:01:38][D][sensor:092]: 'Current time stamp': Sending state 1584972160.00000  with 1 decimals of accuracy
[10:01:38][D][sensor:092]: 'Current time': Sending state 37.00000  with 1 decimals of accuracy
[10:01:39][D][sensor:092]: 'Current time stamp': Sending state 1584972160.00000  with 1 decimals of accuracy
[10:01:39][D][sensor:092]: 'Current time': Sending state 38.00000  with 1 decimals of accuracy
[10:01:40][D][sensor:092]: 'Current time stamp': Sending state 1584972160.00000  with 1 decimals of accuracy
[10:01:40][D][sensor:092]: 'Current time': Sending state 39.00000  with 1 decimals of accuracy
[10:01:41][D][sensor:092]: 'Current time stamp': Sending state 1584972160.00000  with 1 decimals of accuracy
[10:01:41][D][sensor:092]: 'Current time': Sending state 40.00000  with 1 decimals of accuracy
[10:01:42][D][sensor:092]: 'Current time stamp': Sending state 1584972160.00000  with 1 decimals of accuracy`
```

**Additional information and things you've tried:**

Tried this also on ESP32, same issue

Vedeneb, 2020-11-28:
I can confirm this issue, took me hours to find why my automation was not working. Is there any workaround? I'm trying to measure the interval between two pulses on a GPIO Pin. Pulse Width doesn't work since it only updates based on update_interval, I need the duration since the last pulse the moment the pin state changes. So I tried it with the Time component to store the current timestamp in a global variable and on the next pulse measure the difference. But now I'm stuck as the timestamp only updates every 128 seconds...

Vedeneb, 2020-11-28:
I just tried to use `millis()` inside lambda, and it worked! That's a way better solution anyway since I now have sub-seconds-precission and also don't need the time component anylonger. Why is it nowhere documented that one can use millis() in lambda?

chemelli74, 2021-03-20:
Is this issue still open ?

oxan, 2021-07-27:
The problem here isn't in the time component, but in the state publishing: ESPHome uses single-precision floats to represent states, and a single-precision float isn't accurate enough to fully represent enough digits for a timestamp. As long as you store the timestamp in an integer variable it works fine.

closed 2021-07-27 by oxan

behnam85, 2021-11-03:
I am also facing same issue