esphome/issues #1100: Can't upload an OTA in ESP32

sermayoral, 2020-03-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.107.5
ESPHome 1.14.3 (release)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/ota.html

**Description of problem:**

I can't perform an OTA update. I have uploaded lots of firmware of this device but, suddenly, the OTA update does not work anymore. I have put a SSD1306 screen yesterday, after that i uploaded a firmware with the screen with success. Ten minutes later i fixed a problem with the screen (to show pzems data), but when i tried to upload the firmware again, i got this error. I tried many times without success. Today i have tried once again, but same result :-(

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: portero
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret nombre_wifi_1
  password: !secret passwd_wifi
  
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.36
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.30
    dns2: 8.8.8.8
  
  # Necesario establecer esta linea en el cambio de IP de *.37 a la nueva
  # use_address: 192.168.1.37
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Portero Fallback Hotspot"
    password: !secret passwd_esphome

captive_portal:

# Enable logging
logger:
  # Necesario si queremos usar UART0
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret passwd_esphome

# Habilita el acceso web al ESP
# web_server:
#   port: 80

ota:
  password: !secret passwd_esphome
  
binary_sensor:
  # Sensor para detectar llamada al porterillo
  - platform: gpio
    pin: 
      number: 18
      mode: INPUT_PULLUP
      inverted: True
    name: "Portero"
    device_class: sound
    filters:
      - delayed_on: 100ms
  # Boton de gestion de pantalla consumos
  - platform: gpio
    pin: 
      number: 25
      mode: INPUT_PULLUP
      inverted: True
    id: boton_gestion_pantalla
    on_press:
      then:
      - display.page.show_next: my_display
      - component.update: my_display

# Definimos la pantalla   
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    id: my_display
    pages:
      - id: pagina_apagado
        lambda: |-
          it.fill(COLOR_OFF);
      - id: pagina_general
        lambda: |-
          it.printf(63, 0, id(cabecera), TextAlign::TOP_CENTER, "GENERAL");
          it.printf(63, 15, id(cuerpo), TextAlign::TOP_CENTER, "%.1f W", id(pzem_general_power).state);
      - id: pagina_calefaccion
        lambda: |-
          it.printf(63, 0, id(cabecera), TextAlign::TOP_CENTER, "CALEFACCION");
          it.printf(63, 15, id(cuerpo), TextAlign::TOP_CENTER, "%.1f W", id(pzem_calefaccion_power).state);
      
# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

# Definimos las fuentes que iran en la pantalla
font:
  - file: 'fuentes/slkscr.ttf'
    id: cabecera
    size: 9
  - file: 'fuentes/BebasNeue-Regular.ttf'
    id: cuerpo
    size: 40

# Habilitamos el bus I2C para el control de la pantalla
i2c:
  sda: 21
  scl: 22
      
sensor:
  # Termometros
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:30:AF:2D
    temperature:
      name: "Temperatura 3"
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 15 #3
            send_every: 3 #2
    humidity:
      name: "Humedad 3"
    battery_level:
      name: "Bateria Higrometro Peques"
  - platform: xiaomi_lywsd02
    mac_address: 3F:59:C8:63:38:99
    temperature:
      name: "Temperatura 4"
    humidity:
      name: "Humedad 4"
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:38:39:8E
    temperature:
      name: "Temperatura 5"
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 15 #3
            send_every: 3 #2
    humidity:
      name: "Humedad 5"
    battery_level:
      name: "Bateria Higrometro Salon"
  # Pzem General
  - platform: pzemac
    modbus_id: bus_general
    current:
      name: "General Intensidad"
      internal: true
    voltage:
      name: "General Voltaje"
      internal: true
    power:
      name: "General Consumo Instantaneo"
      id: pzem_general_power
    frequency:
      name: "General Frecuencia"
      internal: true        
    power_factor:
      name: "General Factor Potencia"
      internal: true        
    update_interval: 30s
  # Pzem Calefaccion
  - platform: pzemac
    modbus_id: bus_calefaccion
    current:
      name: "Calefaccion Intensidad"
      internal: true      
    voltage:
      name: "Calefaccion Voltaje"
      internal: true      
    power:
      name: "Calefaccion Consumo Instantaneo"
      id: pzem_calefaccion_power
    frequency:
      name: "Calefaccion Frecuencia"
      internal: true      
    power_factor:
      name: "Calefaccion Factor Potencia"
      internal: true      
    update_interval: 30s
      
switch:
  - platform: gpio
    pin: 19
    id: portal_no_usar
  - platform: template
    name: "Portal"
    turn_on_action:
    - switch.turn_on: portal_no_usar
    - delay: 600ms
    - switch.turn_off: portal_no_usar
  - platform: gpio
    pin: 23
    name: "Zumbador"
    restore_mode: ALWAYS_ON
  
# Necesario para el PZEM
uart:
  - id: uart_general
    tx_pin: 1
    rx_pin: 3
    baud_rate: 9600
    stop_bits: 2
  - id: uart_calefaccion
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
    stop_bits: 2
modbus:
  - id: bus_general
    uart_id: uart_general
  - id: bus_calefaccion
    uart_id: uart_calefaccion
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/portero.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/portero
Processing portero (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/portero/.pioenvs/portero/src/main.cpp.o
Linking /data/portero/.pioenvs/portero/firmware.elf
Retrieving maximum program size /data/portero/.pioenvs/portero/firmware.elf
Checking size /data/portero/.pioenvs/portero/firmware.elf
Building /data/portero/.pioenvs/portero/firmware.bin
DATA:    [==        ]  18.9% (used 62068 bytes from 327680 bytes)
PROGRAM: [==========]  95.0% (used 1557106 bytes from 1638400 bytes)
======================== [SUCCESS] Took 190.58 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.36
INFO Uploading /data/portero/.pioenvs/portero/firmware.bin (1557216 bytes)
Uploading: [=                                                           ] 2% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

**Additional information and things you've tried:**

krupaldesai, 2020-03-31:
Try this out, it worked for me: https://github.com/esphome/issues/issues/1098#issuecomment-606784815

sermayoral, 2020-03-31:
@krupaldesai thanks.

I'm compiling it, but i think it is not going to work for me. I haven't never had problem with ble + wifi. I think memory is full and that is the problem. 

Unfortunately i don't know how to fix this problem. Using ESPHome Flasher with an USB don't work either :-(

sermayoral, 2020-03-31:
Same problem:

```
INFO Uploading /data/portero/.pioenvs/portero/firmware.bin (1557216 bytes)
Uploading: [                                                            ] 0% 
ERROR Error sending data: timed out
```

xppx99, 2020-04-01:
I have exactly the same problem, in 3 ESP32 that I had. The last version that worked was 1.13.6. 

[image]

from the device:
[image]

krupaldesai, 2020-04-01:
> Same problem:
> 
> ```
> INFO Uploading /data/portero/.pioenvs/portero/firmware.bin (1557216 bytes)
> Uploading: [                                                            ] 0% 
> ERROR Error sending data: timed out
> ```

I see. I also got to the similar issue after flashing 2-3 times. Sorry about the false alarm. But for me it is always via OTA. USB flash using ESPHome flasher works 100% time for me. I noticed that always I was getting "Connection reset by peer" on the last stable version `1.14.3` but since yesterday when I switched to dev `1.15.0-dev` I'm now getting randomly `Connection reset` and `Error sending data: timed out`.

bangom, 2020-04-07:
What does serial port (console) of ESP32 says when you get "ERROR Error sending data: timed out"?

randybb, 2020-04-08:
esp32_ble_tracker is killing your wifi conection (radio is switching between WiFi and BT protocol). If you don't need it, disable it.

bangom, 2020-04-08:
Duplicate of #1098

bangom, 2020-04-08:
Duplicate of #1098

sermayoral, 2020-04-08:
@bangom is not like #1098, in my case it always fails...

@randybb i need esp32_ble_tracker to control two Xiaomi BLE Thermometers. I have flashed this ESP many times without problems. The problems start when i added a ssd1306 screen, using the i2c bus. I could flash first time, but i had to modify a parameter, and this second time it was impossible to upload. it has not been possible to do it by usb with esphome flasher neither

I think, after upload the code using the i2c bus, the esp memory go to the end and it cannot save the firmware bin, or something like that

randybb, 2020-04-08:
What is your RSSI? Many upload problems and drops are caused by weak WiFi signal. ESP32 has much worse reception than ESP8266 with the same pcb antenna.

sermayoral, 2020-04-08:
> 
> 
> What is your RSSI? Many upload problems and drops are caused by weak WiFi signal. ESP32 has much worse reception than ESP8266 with the same pcb antenna.

[12:33:12][I][app:100]: ESPHome version 1.14.3 compiled on Mar 21 2020, 23:36:25
[12:33:12][C][wifi:415]: WiFi:
[12:33:12][C][wifi:283]:   SSID: 

[redacted]
[12:33:12][C][wifi:284]:   IP Address: 192.168.1.36
[12:33:12][C][wifi:286]:   BSSID: 

[redacted]
[12:33:12][C][wifi:287]:   Hostname: 'portero'
[12:33:12][C][wifi:291]:   Signal strength: -50 dB ▂▄▆█

randybb, 2020-04-08:
-50 dB is ideal.
Based on [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf ) pins 6-11. 16 and 17 are not "safe to use" as these are used by internal memory:
> External connections can be made to any GPIO except for GPIOs in the range 6-11, 16, or 17. GPIOs 6-11 areconnected to the module’s integrated SPI flash and PSRAM. GPIOs 16 and 17 are connected to the module’s integrated PSRAM.

sermayoral, 2020-04-08:
@randybb thanks!! I will see it.

But I'm afraid the problem is not going to be solved with this... :-(

randybb, 2020-04-08:
I am using i2c all over the place for displays, gpio, temperature sensors, light sensors,... and no problems, Sure, no BT that causes headache of many people. (I just hope all peripherals and sleep is disabled during OTA...)

sermayoral, 2020-04-08:
Thanks @randybb.

Do you think it could be an ESP32 memory problem? I see the firmware size is:
```
Building /data/portero/.pioenvs/portero/firmware.bin
DATA:    [==        ]  18.9% (used 62068 bytes from 327680 bytes)
PROGRAM: [==========]  95.0% (used 1557106 bytes from 1638400 bytes)
```
Maybe it has no size to save the firmware until it can flash it. 

I don't think it's the ble because it also fails using USB connection via ESPHome Flasher

randybb, 2020-04-08:
This should be fine - it has two same partitions, otherwise it would raise an error about insufficient space. One is Active and the other one is used for OTA. After validation it will change register and boot from the other one.

bangom, 2020-04-08:
> @bangom is not like #1098, in my case it always fails...
> 
@sermayoral , for me OTA also always fails if esp32_ble_tracker is enabled.

So we know that there is btle + wifi coexistence problem on ESP32... how do we solve this? Disable dynamically esp32_ble_tracker during OTA? But generally, wifi is not stable all the time, I see a lot of disconnects due to esp32_ble_tracker being enabled (tried various scan_parameters without any luck).

mikalauskas, 2020-09-26:
ESPHome version 1.15.2
the issue still persists

```pi@rpi:~/esphome $ esphome rf_bridge.yaml upload
INFO Reading configuration rf_bridge.yaml...
INFO Resolving IP address of rf_bridge.lan
INFO  -> 10.32.56.113
INFO Uploading rf_bridge/.pioenvs/rf_bridge/firmware.bin (1517024 bytes)
Uploading: [=                                                           ] 2%
ERROR Error sending data: [Errno 104] Connection reset by peer
```

```[14:47:33][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[14:47:37][D][ota:072]: Starting OTA Update from 10.32.56.49...
[14:47:37][D][ota:243]: OTA in progress: 0.1%
[14:47:47][I][mqtt:202]: MQTT Connected!
[14:47:48][D][binary_sensor:036]: 'RF Bridge Status': Sending state ON
```

xppx99, 2020-09-26:
I also confirm that the issue persists

JoeSydney, 2020-09-26:
I am not sure if my problems was the same as others, but I went from zero success with OTA (always had to resort to serial flashing) to now 100% success with OTA. It was working on 1.15.1,then encountered issues moving from 1.15.1 to 1.15.2 but I did a fresh install of ESPhome including deleting the platformio stuff, and now OTA is working on 1.15.2 as well...

Maybe try a fresh install of everything?

pieterdevriesch, 2021-02-04:
Not stale. Issue still persists in 1.15.3, making it difficult to upgrade to 1.16. The fix should be really simple, just set ble_tracker and rf_bridge to off before initiating the firmware upload.

DrBlokmeister, 2021-03-30:
Yep. Same issue here. I actually destroyed an ESP32 board trying to find out why OTA always fails. Removed ble tracker and OTA just works. I was going crazy over this.

Now that ESPHome is becoming more mature as a part of Home Assistant, we should really get this issue fixed. Basically the ble_tracker component cannot be used together with the OTA component at the moment. At least we should disable OTA together with ble_tracker or warn the user if both are present in the config. Next step is to temporarily disable ble when an OTA is ongoing.

joe-sydney, 2021-03-30:
I had the same issues, but now OTA works consistently and I don't know how I got it to work. This is on two different brands of esp32 boards, and on quite a few esphome releases. Failure is very occasional. The only thing I did was tweak the following. Maybe give it a try to see if it works for you. I don't really know why it started working for me.

esp32_ble_tracker:
  scan_parameters:
    interval: 320ms
    window: 320ms
    active: false

DrBlokmeister, 2021-04-03:
> I had the same issues, but now OTA works consistently and I don't know how I got it to work. This is on two different brands of esp32 boards, and on quite a few esphome releases. Failure is very [...]

Thanks for the tip. Gonna give that a try.

joe-sydney, 2021-04-04:
> Thanks for the tip. Gonna give that a try.

Did it work? I am interested to see why OTA suddenly started working for me.

fimholz, 2021-04-08:
> I added it to the config and could flash it OTA no problem. However, when I tried to reflash, it didn't work. My ESP32 node had trouble connecting to wifi. Connection was very intermittent. Just removed the config and now it works again.

I think the setting with interval=window is problematic since the RF is shared between BLE and WiFi. When esp32_ble_tracker is configured like this, the WiFi stack has no time window left to receive packages like OTA data. I would try, step by step, lowering the window time (e.g. 180ms) and then remove all but one sensor which uses the esp32_ble_tracker component. This could get an idea if just the WiFi stack has no time to receive or if the esp32_ble_tracker component uses too much CPU power because of too many sensors. I had a similar problem with the xiaomi_lywsdcqq platform when 8 sensors and 8 ble_rssi levels were defined in a single ESP32 board. After removing the rssi_levels and splitting up to 5+3 sensors on two boards, the OTA problem was gone.

[Here](https://esphome.io/components/esp32_ble_tracker.html), under Configuration variables-->scan_parameters, is a Link to a guide from TI regarding the time setting for scanning.

analogue, 2021-05-26:
FYI, https://github.com/esphome/issues/issues/1098#issuecomment-845789058 fixed the issue for me.