esphome/issues #1101: Dallas Temperature Sensor error:found no sensors in ESP32

RewinKant, 2020-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Home Assistant:
arch: aarch64
docker: true
hassio: true
os_name: Linux
os_version: 4.19.106-v8
python_version: 3.7.7
version: 0.107.5

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU-32S ESP32 - error [AliExpress](https://aliexpress.ru/item/32864722159.html?spm=a2g0s.9042311.0.0.234733edwPgZ4E)
NodeMCU ESP8266 - test [AliExpress](https://aliexpress.ru/item/32656401198.html?spm=a2g0s.9042311.0.0.234733edwPgZ4E)

**Affected component:**
[Dallas Temperature Sensor](https://esphome.io/components/sensor/dallas.html)

**Description of problem:**
I couldn't connect this module to ESP32. I started looking for a problem in Dallas by dismantling it to the end. When I connected this module to the ESP8266, I realized that the sensor is working. So, in my opinion, there is an error in the ESPHome program or it does not work correctly. The photo below shows the comparative modifications of ESP32 and ESP8266. And a photo with the connected circuit.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_esp32
  platform: ESP32
  board: nodemcu-32s
wifi:
  ssid: "Natali"
  password: "11197683"
  ap:
    ssid: "Test Esp32 Fallback Hotspot"
    password: "LN77gfxpxdNG"
captive_portal:
logger:
  level: DEBUG
api:
  password: "1234"
ota:
  password: "1234"

dallas:
  - pin: GPIO34
    update_interval: 5s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
in photo
```
[image]

**Additional information and things you've tried:**
module DS18B20 [AliExpress](https://aliexpress.ru/item/4000143479592.html?spm=a2g0s.9042311.0.0.234733edwPgZ4E)

Connection diagram:
[image]
[image]

xheronimo, 2020-03-24:
dallas:
  - pin: GPIO34
    update_interval: 5s
   index : 0

randybb, 2020-03-24:
GPIO34 is input only ([ref](https://i2.wp.com/randomnerdtutorials.com/wp-content/uploads/2018/08/ESP32-DOIT-DEVKIT-V1-Board-Pinout-36-GPIOs-updated.jpg?ssl=1)). And don't forget you can have only one master on one-wire bus.

RewinKant, 2020-03-25:
> пишите
> индекс: 0
> 
> Для вас, чтобы выбрать датчик, который вы найдете в первую очередь на этом pin-код
> 
> даллас:
> 
> * pin: GPIO34
>   update_interval: 5s
>   индекс : 0

RewinKant, 2020-03-25:
1) You gave an example of a sensor with adress: 0, this does not apply to the error, but it did not give a positive result
2) I also changed the pin, but this should not affect performance.
```
INFO Connecting to test_esp32.local:6053 (192.168.0.54)
INFO Successfully connected to test_esp32.local
[11:01:38][I][app:100]: ESPHome version 1.14.3 compiled on Mar 25 2020, 11:00:41
[11:01:38][C][wifi:415]: WiFi:
[11:01:38][C][wifi:283]:   SSID: [redacted]
[11:01:38][C][wifi:284]:   IP Address: 192.168.0.54
[11:01:38][C][wifi:286]:   BSSID: [redacted]
[11:01:38][C][wifi:287]:   Hostname: 'test_esp32'
[11:01:38][C][wifi:291]:   Signal strength: -64 dB ▂▄▆█
[11:01:38][C][wifi:295]:   Channel: 1
[11:01:38][C][wifi:296]:   Subnet: 255.255.255.0
[11:01:38][C][wifi:297]:   Gateway: 192.168.0.21
[11:01:38][C][wifi:298]:   DNS1: 192.168.0.21
[11:01:38][C][wifi:299]:   DNS2: 0.0.0.0
[11:01:38][C][logger:175]: Logger:
[11:01:38][C][logger:176]:   Level: DEBUG
[11:01:38][C][logger:177]:   Log Baud Rate: 115200
[11:01:38][C][logger:178]:   Hardware UART: UART0
[11:01:38][C][dallas.sensor:070]: DallasComponent:
[11:01:38][C][dallas.sensor:071]:   Pin: GPIO32 (Mode: INPUT)
[11:01:38][C][dallas.sensor:072]:   Update Interval: 5.0s
[11:01:38][W][dallas.sensor:075]:   Found no sensors!
[11:01:38][C][dallas.sensor:085]:   Device 'Livingroom Temperature'
[11:01:38][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[11:01:38][C][dallas.sensor:085]:     Accuracy Decimals: 1
[11:01:38][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[11:01:38][C][dallas.sensor:093]:     Address: 0x0000000000000000
[11:01:38][C][dallas.sensor:094]:     Resolution: 12
[11:01:38][C][captive_portal:169]: Captive Portal:
[11:01:38][C][ota:029]: Over-The-Air Updates:
[11:01:38][C][ota:030]:   Address: test_esp32.local:3232
[11:01:38][C][ota:032]:   Using Password.
[11:01:38][C][api:095]: API Server:
[11:01:38][C][api:096]:   Address: test_esp32.local:6053
```

randybb, 2020-03-25:
`[02:34:18][C][dallas.sensor:072]: DallasComponent:
[02:34:18][C][dallas.sensor:073]:   Pin: GPIO32 (Mode: INPUT)
[02:34:18][C][dallas.sensor:074]:   Update Interval: 60.0s
[02:34:18][D][dallas.sensor:079]:   Found sensors:
[02:34:18][D][dallas.sensor:082]:     0x2D0516B398F4FF28`

closed 2020-07-24 by OttoWinter