esphome/issues #1104: componen climate bang_bang don't work in mode heat or cool

xheronimo, 2020-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home assistant, docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32
**Affected component:**

**Description of problem:**
Component bang_bang don't work . Only in mode auto or off.
In mode heat or cool don't work

**Problem-relevant YAML-configuration entries:**
```yaml
        - type: state-label
          entity: climate.nombre_calefaccion_01
          attribute: current_temperature
          suffix: ' ºC'
          style:
            top: 65%
            left: 22%

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
file
esphome/esphome/components/bang_bang/bang_bang_climate.cpp
[bang_bang_climate.zip](https://github.com/esphome/issues/files/4375334/bang_bang_climate.zip)

I think this problem is in return 
void BangBangClimate::compute_state_() {
  if (this->mode != climate::CLIMATE_MODE_AUTO) {
    // in non-auto mode
    this->switch_to_action_(static_cast<climate::ClimateAction>(this->mode));
//  return; THE PROBLEM TO SOLVED 
  }
Now the problem is that it either behaves like a shutdown or always like a car, I think because of the code that follows

xheronimo, 2020-03-24:
It code work fine.
In mode auto, off, heat or cool

void BangBangClimate::compute_state_() {
//  if (this->mode != climate::CLIMATE_MODE_AUTO) {
    // in non-auto mode
//   this->switch_to_action_(static_cast<climate::ClimateAction>(this->mode));
//    return;
  }
  if (isnan(this->current_temperature) || isnan(this->target_temperature_low) || isnan(this->target_temperature_high)) {
    // if any control values are nan, go to OFF (idle) mode
    this->switch_to_action_(climate::CLIMATE_ACTION_OFF);
    return;
  }
  const bool too_cold = this->current_temperature < this->target_temperature_low;
  const bool too_hot = this->current_temperature > this->target_temperature_high;

  climate::ClimateAction target_action;
  if (too_cold) {
    // too cold -> enable heating if possible, else idle
    if ((this->supports_heat_) && ((this->mode == climate::CLIMATE_MODE_AUTO) || (this->mode == climate::CLIMATE_MODE_HEAT)))
      target_action = climate::CLIMATE_ACTION_HEATING;
    else
      target_action = climate::CLIMATE_ACTION_OFF;
  } else if (too_hot) {
    // too hot -> enable cooling if possible, else idle
    if ((this->supports_cool_) && ((this->mode == climate::CLIMATE_MODE_AUTO) || (this->mode == climate::CLIMATE_MODE_COOL)))
      target_action = climate::CLIMATE_ACTION_COOLING;
    else
      target_action = climate::CLIMATE_ACTION_OFF;
  } else {
    // neither too hot nor too cold -> in range
    if (this->supports_cool_ && this->supports_heat_) {
      // if supports both ends, go to idle mode
      target_action = climate::CLIMATE_ACTION_OFF;
    } else {
      // else use current mode and don't change (hysteresis)
      target_action = this->action;
    }
  }

  this->switch_to_action_(target_action);

xheronimo, 2020-03-26:
The correct file that works fine
[bang_bang_climate.zip](https://github.com/esphome/issues/files/4385963/bang_bang_climate.zip)

glmnet, 2020-03-26:
can you make a fork and push this changes to a branch you can then provide a link to? that way we can review the code change.
Thanks

xheronimo, 2020-05-12:
I push changes in the branch
https://github.com/esphome/esphome/pull/1005
I dont know . how put the link with fix

closed 2020-06-17 by glmnet