esphome/issues #1111: Cover is blocking a button

Chuckame, 2020-03-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass 0.107.5, docker environment, odroid xu4 with ubuntu

**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp8266 (tuya switch 2 ways with tywe3s chip) 

**Affected component:**

https://esphome.io/components/cover/time_based.html
Or maybe generic cover? 

**Description of problem:**
- Expected
The cover is closed. 
When `btnL` released first time, the cover is opening. 
When `btnL` released the second time, the cover should be stopped. 
- Actual
When `btnL` released the second time, the cover is not stopped, but if I hit the `btnR`, the cover is closing.
The `btnL` is "frozen" while the cover is opening or until stopped or closing. 
On my first try, I tried without scripts. Actually tested with scripts for the "async" way, same issue. 
So I think is not a configuration bug, but a feature bug? Or I'm open for some configuration propositions!

**Problem-relevant YAML-configuration entries:**
Actually, the corresponding led is blinking while the cover is opening or closing, nice effect in real 😁😁
```yaml
substitutions:
  node_id: switch_tuya2_1
  blinking_interval: 500ms
  interlock_time: 200ms

esphome:
  name: ${node_id}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

cover:
  - platform: time_based
    name: "Volet salon"
    id: ${node_id}_cover
    device_class: shade
    open_action:
      - switch.turn_off: go_close
      - delay: ${interlock_time}
      - switch.turn_on: go_open
    open_duration: 27s
    close_action:
      - switch.turn_off: go_open
      - delay: ${interlock_time}
      - switch.turn_on: go_close
    close_duration: 25s
    stop_action:
      - switch.turn_off: go_close
      - switch.turn_off: go_open

script:
  - id: close_cover_script
    then:
      - cover.close: ${node_id}_cover
  - id: open_cover_script
    then:
      - cover.open: ${node_id}_cover
  - id: stop_cover_script
    then:
      - cover.stop: ${node_id}_cover

binary_sensor:
  - platform: gpio
    id: btnL
    internal: yes
    pin:
      number: 3
      inverted: yes
    on_release:
      then:
        - if:
            condition:
              - script.is_running: open_cover_script
            then:
              - script.stop: open_cover_script
              - script.execute: stop_cover_script
            else:
              - if:
                  condition:
                    lambda: 'return id(switch_tuya2_1_cover).position != COVER_OPEN;'
                  then:
                    - script.execute: open_cover_script
                  else:
                    - script.execute: stop_cover_script

  - platform: gpio
    id: btnR
    internal: yes
    pin:
      number: 5
      inverted: yes
    on_release:
      then:
        - if:
            condition:
              - script.is_running: close_cover_script
            then:
              - script.stop: close_cover_script
              - script.execute: stop_cover_script
            else:
              - if:
                  condition:
                    lambda: 'return id(switch_tuya2_1_cover).position != COVER_CLOSED;'
                  then:
                    - script.execute: close_cover_script
                  else:
                    - script.execute: stop_cover_script

switch:
  - platform: template
    id: go_close
    turn_on_action:
      - switch.turn_on: relayL1
      - while:
          condition:
            lambda: 'return true;'
          then:
            - switch.turn_on: ledR
            - delay: ${blinking_interval}
            - switch.turn_off: ledR
            - delay: ${blinking_interval}
    turn_off_action:
      - switch.turn_off: relayL1
      - switch.turn_off: ledR
  - platform: template
    id: go_open
    internal: yes
    turn_on_action:
      - switch.turn_on: relayL2
      - while:
          condition:
            lambda: 'return true;'
          then:
            - switch.turn_on: ledL
            - delay: ${blinking_interval} 
            - switch.turn_off: ledL
            - delay: ${blinking_interval}
    turn_off_action:
      - switch.turn_off: relayL2
      - switch.turn_off: ledL

  - platform: gpio
    id: relayL1
    internal: yes
    pin: 13
    interlock: relayL2
    interlock_wait_time: ${interlock_time}
  - platform: gpio
    id: relayL2
    internal: yes
    pin: 4
    interlock: relayL1
    interlock_wait_time: ${interlock_time}
  - platform: gpio
    id: ledL
    internal: yes
    pin:
      number: 14
      inverted: yes
  - platform: gpio
    id: ledR
    internal: yes
    pin:
      number: 1
      inverted: yes

status_led:
  pin:
    number: 0
    inverted: yes
```

**Logs (if applicable):**
No particular logs, the `btnL` press & release is detected, but has no effect. 

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-04-27:
scripts execution ends after sending the cover the command.

you could check if cover is moving like so...

````yaml
cover:
  - platform: time_based
    name: "Volet salon"
    id: ${node_id}_cover
    device_class: shade
    open_action:
      - switch.turn_off: go_close
      - delay: ${interlock_time}
      - switch.turn_on: go_open
    open_duration: 27s
    close_action:
      - switch.turn_off: go_open
      - delay: ${interlock_time}
      - switch.turn_on: go_close
    close_duration: 25s
    stop_action:
      - switch.turn_off: go_close
      - switch.turn_off: go_open

binary_sensor:
  - platform: gpio
    id: btnL
    internal: yes
    pin:
      number: 3
      inverted: yes
    on_release:
      then:
        - if:
            condition:
              lambda: 'return id(switch_tuya2_1_cover).current_operation == COVER_OPERATION_IDLE;'
            then:
              - cover.open: ${node_id}_cover
            else:
              - cover.stop: ${node_id}_cover

  - platform: gpio
    id: btnR
    internal: yes
    pin:
      number: 5
      inverted: yes
    on_release:
      then:
        - if:
            condition:
              lambda: 'return id(switch_tuya2_1_cover).current_operation == COVER_OPERATION_IDLE;'
            then:
              - cover.close: ${node_id}_cover
            else:
              - cover.stop: ${node_id}_cover

switch:
  - platform: template
    id: go_close
    turn_on_action:
      - switch.turn_on: relayL1
      - while:
          condition:
            lambda: 'return true;'
          then:
            - switch.turn_on: ledR
            - delay: ${blinking_interval}
            - switch.turn_off: ledR
            - delay: ${blinking_interval}
    turn_off_action:
      - switch.turn_off: relayL1
      - switch.turn_off: ledR
  - platform: template
    id: go_open
    internal: yes
    turn_on_action:
      - switch.turn_on: relayL2
      - while:
          condition:
            lambda: 'return true;'
          then:
            - switch.turn_on: ledL
            - delay: ${blinking_interval}
            - switch.turn_off: ledL
            - delay: ${blinking_interval}
    turn_off_action:
      - switch.turn_off: relayL2
      - switch.turn_off: ledL

  - platform: gpio
    id: relayL1
    internal: yes
    pin: 13
    interlock: relayL2
    interlock_wait_time: ${interlock_time}
  - platform: gpio
    id: relayL2
    internal: yes
    pin: 4
    interlock: relayL1
    interlock_wait_time: ${interlock_time}
  - platform: gpio
    id: ledL
    internal: yes
    pin:
      number: 14
      inverted: yes
  - platform: gpio
    id: ledR
    internal: yes
    pin:
      number: 1
      inverted: yes

status_led:
  pin:
    number: 0
    inverted: yes

````

closed 2020-04-27 by glmnet

Chuckame, 2020-04-27:
Working!