esphome/issues #1112: MQTT not connecting to mosquitto broker

andreixk, 2020-03-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Compiling on: Mac OS 10.15.4
Python: 3.7.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

Running on: ESP-01S

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**
I'm trying to connect to a mosquitto broker i've set up in google compute engine; I can connect to it from my raspberry pi zero W that's on the same network as ESP-01S. However the device with esphome is trying and failing to connect.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: aquanode
  platform: ESP8266
  board: esp01

wifi:
  ssid: Homer
  password: zzzzzzzzzzz
  ap:
    ssid: "HarveyStub"
    password: "zzzzzzzzzzzzz"

logger:

mqtt:
  broker: my-mqtt-host.name
  port: 20961
  username: mquser
  password: mqpass
  topic_prefix: 'water'

```

**Logs (if applicable):**

```
[22:30:59][D][wifi:319]: Found networks:
[22:30:59][I][wifi:365]: - 'Homer' (B6:BE:76:18:77:22) ▂▄▆█
[22:30:59][D][wifi:366]:     Channel: 2
[22:30:59][D][wifi:367]:     RSSI: -63 dB
[22:30:59][D][wifi:369]: - 'BurnsHome' (B0:BE:76:18:77:22) ▂▄▆█
[22:30:59][D][wifi:369]: - 'houzz' (BA:BE:76:18:77:22) ▂▄▆█
[22:30:59][D][wifi:369]: - 'iHat' (BE:BE:76:18:77:22) ▂▄▆█
[22:30:59][I][wifi:193]: WiFi Connecting to 'Homer'...
[22:31:00][I][wifi:423]: WiFi Connected!
[22:31:00][C][wifi:283]:   SSID: 'Homer'
[22:31:00][C][wifi:284]:   IP Address: 10.0.1.28
[22:31:00][C][wifi:286]:   BSSID: B6:BE:76:18:77:22
[22:31:00][C][wifi:287]:   Hostname: 'aquanode'
[22:31:00][C][wifi:291]:   Signal strength: -66 dB ▂▄▆█
[22:31:00][C][wifi:295]:   Channel: 2
[22:31:00][C][wifi:296]:   Subnet: 255.255.255.0
[22:31:00][C][wifi:297]:   Gateway: 10.0.1.1
[22:31:00][C][wifi:298]:   DNS1: 8.8.8.8
[22:31:00][C][wifi:299]:   DNS2: (IP unset)
[22:31:00][D][wifi:432]: Disabling AP...
[22:31:00][C][web_server:066]: Setting up web server...
[22:31:00][C][mqtt:025]: Setting up MQTT...
[22:31:00][I][mqtt:162]: Connecting to MQTT...
[22:31:19][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[22:31:19][I][mqtt:162]: Connecting to MQTT...
[22:31:38][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[22:31:38][I][mqtt:162]: Connecting to MQTT...
...and so on forever
```

**Additional information and things you've tried:**

brandond, 2020-03-30:
Do you have any logs from the broker, or perhaps a packet capture? The MQTT stuff is generally pretty stable, which makes me suspect the problem is not on the esphome side.

andreixk, 2020-03-30:
the broker did not even receive the attempt at a connection, so there's nothing to show there.

maoterodapena, 2020-04-01:
I agree with brandond about stability on the mqtt part. I see that you defined the google DNS statically, but the mqtt server name seems to be a local name. Can you try to put in the yaml file the static ip of the mqtt server instead of the name?

andreixk, 2020-04-01:
yup, i tried putting the IP address too - no difference.
I'm sure the code is stable, it's probably the matter of some configuration i'm missing? is there any firewall caveats that i may be missing? currently i just have firewall open TCP port 20961

maoterodapena, 2020-04-01:
Can you post your mosquitto.conf (redacting names, etc..)?

andreixk, 2020-04-01:
```
allow_anonymous false
listener 20961 0.0.0.0
pid_file /var/run/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log

include_dir /etc/mosquitto/conf.d
password_file /etc/mosquitto/pwfile
````

maoterodapena, 2020-04-01:
I just tested a similar configuration with port 20961, and user/passwd autentication, but with all devices in the same network, and it works. Even I added the web server component you seem to have.

Just try a
```

logger:
  level: VERY_VERBOSE
```

and maybe you will get a hint... but certainly it seems to be firewall related. This is what I get:
  

```
[22:18:31][C][wifi:295]:   Channel: 6
[22:18:31][C][wifi:296]:   Subnet: 255.255.255.0
[22:18:31][C][wifi:297]:   Gateway: 192.168.1.1
[22:18:31][C][wifi:298]:   DNS1: 80.58.61.254
[22:18:31][C][wifi:299]:   DNS2: 80.58.61.250
[22:18:31][D][wifi:432]: Disabling AP...
[22:18:31][C][web_server:066]: Setting up web server...
[22:18:31][VV][scheduler:056]: set_interval(name='', interval=10000, offset=3908)
[22:18:31][C][mqtt:025]: Setting up MQTT...
[22:18:31][I][mqtt:162]: Connecting to MQTT...
[22:18:31][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=4294961517 (now=8141)
[22:18:31][I][mqtt:202]: MQTT Connected!
[22:18:31][I][app:058]: setup() finished successfully!
[22:18:31][V][mqtt:372]: Publish(topic='water/status' payload='online' retain=1)
```

Just for your reference, if I configure an erroneus IP on the yaml file (to a non existing mqtt server), I get this (with very_verbose logger), and seems to be your problem.

```

[22:25:01][C][mqtt:025]: Setting up MQTT...
[22:25:01][I][mqtt:162]: Connecting to MQTT...
[22:25:01][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=4294965239 (now=8374)
[22:25:10]pm open,type:0 0
[22:25:10][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=7943 (now=17943)
[22:25:20][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=17943 (now=27943)
[22:25:30][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=27943 (now=37943)
[22:25:31][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[22:25:31][I][mqtt:162]: Connecting to MQTT...
[22:25:40][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=37943 (now=47943)
[22:25:50][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=47943 (now=57943)
```
Sorry I can not help you.

andreixk, 2020-05-04:
i have no idea why, but when i turned on verbose logging, it started working. maybe it was just a matter of reflashing it a few times...

closed 2020-05-04 by andreixk