esphome/issues #1115: Unable to use a DC motor with encoder because its too fast.

AalianKhan, 2020-03-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

wemos D1 mini clone
**Affected component:**

https://esphome.io/components/sensor/rotary_encoder.html
https://esphome.io/components/output/esp8266_pwm.html
**Description of problem:**
I am trying to use a 12v DC [motor](https://www.banggood.com/GM4632-370-DC-12V-30RPM-High-torque-Turbo-Encoder-Motor-Worm-Geared-Motor-Reducer-Motor-p-1069461.html?rmmds=search&cur_warehouse=CN&p=KF2715792233201409DJ&custlinkid=223122) with encoder alongside an L298N motor driver. the problem is that when I code in a statement where it says to turn off the motor at EXACTLY 1000 steps, esphome is unable to turn it off because the encoder is too fast for esphome to process. if I type in a statement where it says turn off the motor when it's over 1000 steps it does turn it off but at ~1080 steps or so. I have tried limiting this problem by making the encoder component internal and reduced logger level. now it stops around ~1030 steps or so. 

**Problem-relevant YAML-configuration entries:**
```yaml
api:
  services:
    - service: duty_cycle
      variables:
        cycle: int
      then:
        - output.set_level:
            id: pwm_output
            level: !lambda 'return cycle;'
    - service: pwm
      variables:
        pwm: int
      then:
        - output.esp8266_pwm.set_frequency:
            id: pwm_output
            
            #id: pwm_output
            frequency: !lambda 'return pwm;'
ota:

sensor:
  - platform: rotary_encoder
    name: "Rotary Encoder"
    pin_a: D1
    pin_b: D2
    id: current_state
    internal: true
#    on_value:
#      then:
#      - if:
#          condition: 
#            or:
#              - lambda: |-
#                  return id(current_state).state = 1000;
#              - lambda: |-
#                  return id(current_state).state = 0;
#          then:
#            - switch.turn_off: clk
#            - switch.turn_off: cclk
#            - logger.log:
#                format: "Steps at %.1f"
#                args: [ x ]
#                level: INFO

output:
  - platform: esp8266_pwm
    pin: D7
    id: pwm_output
    frequency: 1000

switch:
  - platform: gpio
    pin: D4
    name: "motor 1"
    id: clk
  - platform: gpio
    pin: D3
    name: motor 2
    id: cclk
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have tried slowing it down by using a PWM single and connecting it to the PWM pin on L298N, but I am unable to do this for some reason. at a frequency of 1000hz (default) with a duty cycle of 99% the motor is off but when I set it 100% its at full power. Tried changing the frequency but the same result.

glmnet, 2020-04-18:
Well seems like two problems

* You should first figure out why PWM is not working for you, you'll need to get this working reliably otherwise you'll never fix your other issue

* Motor has inertia, and this inertia depends on load, so most likely your motor is not driving any load now so when it does behavior will change, for good or for bad but will change, to deal with desired position, output value to the motor and actual feedback value I recommend you to teach yourself a bit of PID control algorithms, they are very complicated but for what you're trying to do here it is not too complicated.

AalianKhan, 2020-04-18:
How would I add a PID in esphome?

glmnet, 2020-04-18:
Esphome is a code generator, it takes yaml and creates a c++ file which is compiled and then this is your final program. 
So you can add any c++ in the yaml itself. 

I would use the interval running at maybe 100 or 50 msec then run the pid there 

But I’ve never tried it. 

https://esphome.io/guides/automations.html#interval

closed 2020-08-30 by stale[bot]