esphome/issues #1118: remote_transmitter raw work Incorrect

nmweizi, 2020-03-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3 homeassistant v0.107.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

Processing fr (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
Unable to transmit nec code previously catched with receiver dump mode.

I want to send 'Power on/off' to TV, the protocol is an IR dump raw,TV not power on or off.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: fr
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxxx"
  password: "xxxxx"
  reboot_timeout: 0s
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Asd Fallback Hotspot"
    password: "xxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxx"

ota:
  password: "xxxxx"

mqtt:
  broker: 192.x.x.x
  username: xxx
  password: xxxxx
  discovery: True
  discovery_retain: True
  reboot_timeout: 0s

remote_receiver:
  pin: 
    number: D1
    inverted: True
  dump: all

  #tolerance: 50%
  #filter: 250us
  #idle: 4ms
  buffer_size: 4kb

            
  
remote_transmitter:
  pin: D8
  
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:
  - platform: template
    name: "TV power"
    # assumed_state: True
    turn_on_action: 
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [308, -1788, 305, -765, 337, -710, 283, -742, 305, -764, 286, -1785, 306, -767, 280, -764, 283, -1790, 303, -765, 283, -1811, 283, -1811, 283, -1811, 336, -711, 282, -1811, 284]
          repeat: 3
            
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

chaoranxie, 2020-03-29:
i am actually having the same issue with raw not working.

debsahu, 2020-04-09:
Comment out the receiver part when using for transmitting. Took me all afternoon to figure out out.

nmweizi, 2020-04-09:
thank you. @debsahu

closed 2020-04-09 by nmweizi