esphome/issues #1120: captive_portal: throw reason='Probe Request Unsuccessful' on Sonoff basic

mguaylambert, 2020-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

8266 Sonoff basic
**Affected component:**

https://esphome.io/components/captive_portal.html

**Description of problem:**
When enabling captiveportal: on the sonoff basic (8266), it cannot connect to the Wi-Fi showing this error in the console : reason='Probe Request Unsuccessful' while trying to connect to my AP. If I remove the captiveportal:, it works fine. Also, adding debug: seem's to fix the issue weirdly.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lampe_chevet
  platform: ESP8266
  board: esp8285

wifi:
  ssid: "Amenophis"
  password: "***"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lampe Chevet Fallback Hotspot"
    password: "***"
    
captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "***"

ota:
  password: "***"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Lampe de chevet"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Lampe de chevet"
    icon: "mdi:lamp"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
Using 'COM9' as serial port.
Showing logs:
[11:56:47][W][wifi_esp8266:334]: Event: Disconnected ssid='Amenophis' reason='Probe Request Unsuccessful'
[11:56:47][W][wifi:451]: Error while connecting to network.
[11:56:47][W][wifi:488]: Restarting WiFi adapter...
[11:56:47][I][wifi:193]: WiFi Connecting to 'Amenophis'...
[11:56:53][W][wifi_esp8266:334]: Event: Disconnected ssid='Amenophis' reason='Probe Request Unsuccessful'
[11:56:53][W][wifi:451]: Error while connecting to network.
[11:56:53][W][wifi:488]: Restarting WiFi adapter...
[11:56:58][D][wifi:304]: Starting scan...
[11:57:04][D][wifi:319]: Found networks:
[11:57:04][I][wifi:365]: - 'Amenophis' (E4:F4:C6:12:61:36) ▂▄▆█
[11:57:04][D][wifi:366]:     Channel: 7
[11:57:04][D][wifi:367]:     RSSI: -34 dB
[11:57:04][D][wifi:369]: - 'VIRGIN806' (2C:79:D7:02:31:3E) ▂▄▆█
[11:57:04][D][wifi:369]: - 'EBOX_0568_EXT' (9C:3D:CF:FB:89:98) ▂▄▆█
[11:57:04][D][wifi:369]: - 'GigiNet' (68:8F:2E:D0:E7:18) ▂▄▆█
[11:57:04][D][wifi:369]: - 'VIDEOTRON0031' (90:EF:68:7B:22:24) ▂▄▆█
[11:57:04][D][wifi:369]: - 'EBOX_0568' (00:23:6A:D2:05:69) ▂▄▆█
[11:57:04][D][wifi:369]: - 'luarbima' (68:FF:7B:F6:C1:48) ▂▄▆█
[11:57:04][D][wifi:369]: - 'Los Pollos Locos' (0C:B6:D2:E1:BE:0C) ▂▄▆█
[11:57:04][D][wifi:369]: - 'SDC' (40:F2:01:EE:B3:63) ▂▄▆█
[11:57:04][D][wifi:369]: - '/dev/null' (00:11:32:A5:56:D4) ▂▄▆█
[11:57:04][D][wifi:369]: - '/dev/null/guest' (02:11:32:A5:56:D4) ▂▄▆█
[11:57:04][D][wifi:369]: - '' (16:11:32:A5:56:D4) ▂▄▆█
[11:57:04][D][wifi:369]: - 'TOTOLINK N302R Plus' (78:44:76:98:86:34) ▂▄▆█
[11:57:04][D][wifi:369]: - 'oh dear!' (08:62:66:97:01:28) ▂▄▆█
[11:57:04][D][wifi:369]: - 'frito-O-poulet' (AC:3B:77:A6:A9:06) ▂▄▆█
[11:57:04][D][wifi:369]: - 'BELL677' (18:90:D8:DF:1E:B6) ▂▄▆█
[11:57:04][D][wifi:369]: - 'VIDEOTRON4827' (C0:A0:BB:9F:E3:F0) ▂▄▆█
[11:57:04][D][wifi:369]: - '/dev/null/guest' (02:11:32:93:47:99) ▂▄▆█
[11:57:04][D][wifi:369]: - 'LifeisGolden' (70:8B:CD:60:4A:90) ▂▄▆█
[11:57:04][D][wifi:369]: - 'Marowak' (24:F5:A2:C0:B3:F9) ▂▄▆█
[11:57:04][D][wifi:369]: - 'Boomer568' (94:6A:77:6C:97:D5) ▂▄▆█
[11:57:04][D][wifi:369]: - '' (94:6A:77:66:7E:B4) ▂▄▆█
[11:57:04][D][wifi:369]: - 'guybrushthreepwood' (00:23:6A:EE:14:42) ▂▄▆█
[11:57:04][D][wifi:369]: - 'Mac4Ever' (B8:C7:5D:0C:1B:F7) ▂▄▆█
[11:57:05][D][wifi:369]: - 'VIDEOTRON0651' (04:BF:6D:5C:15:CC) ▂▄▆█
[11:57:05][D][wifi:369]: - 'Helix2019' (94:6A:77:66:7E:B3) ▂▄▆█
[11:57:05][D][wifi:369]: - 'Grace&Frankie' (34:DB:9C:82:44:0E) ▂▄▆█
[11:57:05][D][wifi:369]: - 'LoftStory' (04:D4:C4:C3:1E:D0) ▂▄▆█
[11:57:05][D][wifi:369]: - 'Motha-Fuckaaaaaaaaaaa-visiteur' (20:AA:4B:D9:20:1E) ▂▄▆█
[11:57:05][D][wifi:369]: - 'VIDEOTRON3856' (B8:EC:A3:D2:E9:90) ▂▄▆█
[11:57:05][D][wifi:369]: - 'Simba' (E4:F0:42:D2:6B:9A) ▂▄▆█
[11:57:05][D][wifi:369]: - 'BELL457' (B8:EE:0E:BB:1E:7A) ▂▄▆█
[11:57:05][D][wifi:369]: - 'Cedrik-EXT' (60:63:4C:B3:0A:F3) ▂▄▆█
[11:57:05][D][wifi:369]: - 'RGL-EXT' (80:26:89:42:02:20) ▂▄▆█
[11:57:05][D][wifi:369]: - 'VIDEOTRON2535' (04:BF:6D:59:C3:14) ▂▄▆█
[11:57:05][I][wifi:193]: WiFi Connecting to 'Amenophis'...
[11:57:10][W][wifi_esp8266:334]: Event: Disconnected ssid='Amenophis' reason='Probe Request Unsuccessful'
[11:57:10][W][wifi:451]: Error while connecting to network.
[11:57:10][W][wifi:488]: Restarting WiFi adapter...
[11:57:11][I][wifi:193]: WiFi Connecting to 'Amenophis'...
[11:57:16][W][wifi_esp8266:334]: Event: Disconnected ssid='Amenophis' reason='Probe Request Unsuccessful'
[11:57:16][W][wifi:451]: Error while connecting to network.
[11:57:16][W][wifi:488]: Restarting WiFi adapter...

```

**Additional information and things you've tried:**

darylsargent, 2020-05-27:
Same issue experienced, worked fine untill adding a timer and condition so maybe a memory issue?

Removed captive_portal:, all good

andordavoti, 2020-08-20:
having the same issue, this bricked a lot of my sonoff devices. any news on how to fix this?

MaxWh1t3, 2020-12-06:
https://github.com/esphome/issues/issues/952 worked for me on 3 boards!

BuildTheRobots, 2021-03-04:
Had the same issue, but until I found this post I was convinced it was enabling/disabling a DHT sensor that was causing it. Now that I've disabled the captive_portal wifi and sensor are both working.

sielenk, 2021-04-25:
Changing the configuration for my device made this problem reappear for one of my devices.
Adding or removing "captive_portal" has no effect, the same goes for "status_led".

It works when using "switch" for the relay and fails when replacing that with "output" and "fan".

andriej, 2021-07-05:
Encountered this on one of my sonoffs. It was "down" since half of year (1.15), I've reflashed it and put newest 1.19.4 on - still "Probe request" and was not able to login to WiFi.
I've found one of issues regarding removing "captive portal" - did that, removed one line, reflashed and it works like a charm.

DoctorOctagonapus, 2021-11-08:
Welp, I'm stuck. Got a Sonoff Basic that's now totally unusable since updating to 2021.10.3.

Logs:
```
INFO Reading configuration /config/esphome/[name].yaml...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:21:24]scandone
[15:21:24][W][wifi_esp8266:479]: Event: Disconnected ssid='[name]' reason='Probe Request Unsuccessful'
[15:21:24]wifi evt: 1
[15:21:24]STA disconnect: 201
[15:21:24][W][wifi:535]: Error while connecting to network.
[15:21:24][W][wifi:570]: Restarting WiFi adapter...
[15:21:24][V][wifi_esp8266:060]: Disabling STA.
[15:21:24]del if0
[15:21:24]usl
[15:21:24]mode : null
[15:21:24][V][wifi_esp8266:535]: Event: Changed Mode old=STA new=OFF
[15:21:24]wifi evt: 8
[15:21:30][I][wifi:247]: WiFi Connecting to '[name]'...
[15:21:30][V][wifi:249]: Connection Params:
[15:21:30][V][wifi:250]:   SSID: '[name]'
[15:21:30][V][wifi:255]:   BSSID: Not Set
[15:21:30][V][wifi:273]:   Password: [redacted]
[15:21:30][V][wifi:280]:   Channel: Not Set
[15:21:30][V][wifi:287]:   Using DHCP IP
[15:21:30][V][wifi:289]:   Hidden: NO
[15:21:30][V][wifi_esp8266:058]: Enabling STA.
[15:21:30]mode : sta([Mac])
[15:21:30]add if0
[15:21:30][V][wifi_esp8266:535]: Event: Changed Mode old=OFF new=STA
[15:21:30]wifi evt: 8
[15:21:32]scandone
[15:21:32][W][wifi_esp8266:479]: Event: Disconnected ssid='[name]' reason='Probe Request Unsuccessful'
[15:21:32]wifi evt: 1
[15:21:32]STA disconnect: 201
[15:21:32][W][wifi:535]: Error while connecting to network.
[15:21:32][W][wifi:570]: Restarting WiFi adapter...
Eventually it starts up the fallback AP and continues on a loop of failed connection attempts.
```
Config:
```esphome:
  name: [redacted]
  platform: ESP8266
  board: esp01_1m

wifi:
  # Hidden SSID
  ssid: !secret wifi_ssid
  password: !secret wifi_pw
  fast_connect: true
  output_power: 10db
  
  ap:
    ssid: [name]
    password: [redacted]

#captive_portal:

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

...
```
I've already disabled captive_portal after this device broke with the same error once before. It was fine with previous versions of ESPHome, but the latest one has broken it completely now apparently. All my other ESP devices work, including another Sonoff Basic that still has captive_portal enabled. This one has apparently just decided it doesn't wanna live any more.

Any ideas?

tmarkson, 2021-11-22:
> Any ideas?

@DoctorOctagonapus I suggest simplifying your config to barebones, as others have before.  Try removing the `ap`, `output_power`, and `level` items.  Remove any other components below `ota` and try flashing -- if that works, then add those components individually, uploading firmware in between adds.  I've found compatibility issues this way, especially when using ESP8266 devices that seem to choke on complex items.  Some variants may be running at reduced clock or another handicap.

pironic, 2022-04-09:
here is my config

```
substitutions:
  id: bes02
  ip: 10.1.50.121
  class: door
  version: 22.04.09.02

esphome:
  name: ${id}
  platform: esp8266
  board: nodemcu

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: true
  manual_ip:
    static_ip: "${ip}"
    subnet: 255.255.255.0
    gateway: 10.1.50.1

  ap:
    ssid: "${id} Fallback"
    password: !secret wifi_pass

logger:
  level: VERBOSE
  baud_rate: 76800
  hardware_uart: UART0
```

And i'm getting the same problems on 2022.3.2. Did anyone ever figure out a true cause for this? 

[image]

alnavasa, 2022-04-12:
Robotdyn 2x relay board,
ESP8266
same issue with wireless as soon as the code was a little bit interesting. 
managed to solve the issue by commenting the lines of captive portal and fallback ap

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  power_save_mode: none
  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "$device_name Fallback Hotspot"
#    password: "lghjfd5rfsU"

#captive_portal:
```

hcooper, 2023-01-17:
+1, I found a fresh install of ESPHome on a Sonoff Basic R2 fails with 

```
[W][wifi_esp8266:479]: Event: Disconnected ssid=... reason='Probe Request Unsuccessful'
```

**However I can confirm adding `debug:` to the config makes it work!**

Also of note, a fresh install of Tasmota works fine (v12.3.1)

vks007, 2023-01-27:
I am also getting the same error with many of my modules. Is there a solution to this? The other thread suggested to use framework version 1.8.0 but that's too old now, it doesnt even compile now.
@hcooper  are you sure adding debug helps because logger level debug is anyway the default?

johny-mnemonic, 2023-03-12:
This must be some kind of conflict in the compilation.
I just hit this exact issue after using two Sonoff basics with same configuration for a long time without any issue. What triggered it was adding the DHT sensor definition to one of them. 
After going through these threads I came to a conclusion it must be a combination of modules that results to either filling too much RAM or using too much of some other resource of this poor little bugger.
And sure enough, removing some modules fixed it. So, my Sonoff basic modules were both running with captive portal, mqtt client, binary_sensor, switch and output defined.
The one where I added the DHT sensor is now running just fine with captive portal and mqtt removed. Not sure which one was the culprit but after removing these and flashing it using serial link it works just fine 🤷‍♂️

myLens, 2023-04-19:
I also ended up in a messed runtime. Adding and removing modules leads to unpredictable results - no matter it it is webserver, a simple sensor, the captive portal ....
I suppose the wifi_esp8266 has memory issues or some combination of modules causes malfunction.
Sometimes it works like a charm but even the slightest change can become a game changer - very disappointing.

aiexz, 2023-09-22:
Removing `captive_portal` from config worked for HW-628 board (nodemcu clone)

kjake, 2023-10-03:
> **However I can confirm adding `debug:` to the config makes it work!**

+1 this worked for me on two of my D1 boards and got them back online with the latest build of ESPhome (2023.9.3)

mguaylam, 2024-11-26:
Hey there guys! Since I gave away my Sonoff Basics, I never encountered this issue again.
Let me know if it's still a thing, I could re-open.