esphome/issues #1122: DF Player Mini option "dfplayer.is_playing"

Electronlibre2012, 2020-04-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassos 3.12 Supervisor 214 HA0.107.7
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32Cam
**Affected component:**

DF PLAYER MINI option dfplayer.is_playing

**Description of problem:**
Hello,

sorry its not a problem but i dont find any information if its possible to make the option "dfplayer.is_playing" true / false visible in HA? 

I try to put a "text_sensor" to be report in HA but i dont succeed...sorry i am a noob.

In simple words : I want to report the state of "dfplayer.is_playing" in HA.
I know this is an option in EspHome and it cant be seen in HA, but with the use of a text_sensor or anything else? 

Please i need some help.

**Problem-relevant YAML-configuration entries:**
```yaml
text_sensor:
  name: State_DFPlayerMini
  icon: "mdi:player"
  id: df_player
  if:
    condition:
      dfplayer.is_playing
    then:
      logger.log: 'Playback is active!'
   
   on_value:
      then:
        - lambda: |-
            ESP_LOGD("main", "The state of DF Player is %s", x );

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-04-01:
try to make your way using the [template binary sensor](https://esphome.io/components/binary_sensor/template.html)

glmnet, 2020-04-01:
_not tested code..._

```yaml

binary_sensor:
  - platform: template
    name: "DFPlayer playing"
    lambda: |-
      return id(df_player).is_playing();
````

Electronlibre2012, 2020-04-01:
> 
> 
> _not tested code..._
> 
> ```yaml
> binary_sensor:
>   - platform: template
>     name: "DFPlayer playing"
>     lambda: |-
>       return id(df_player).is_playing();
> ```

A great thank you @glmnet ! it works : here is the code : 

```
dfplayer:
  id: dfplayerState
  on_finished_playback:
    then:
      logger.log: 'Playback finished event'

binary_sensor:   
  - platform: template
    name: "DFPlayer State"
    lambda: |-
      return id(dfplayerState).is_playing();

```
i make a doorbell with a ESP32cam + capacitive touch button : [https://www.thingiverse.com/thing:4209618](url)

And now i upgrade with a vocal synthesis : DFPlayer mini.

Thanks for your help!

closed 2020-04-02 by glmnet