esphome/issues #1123: ESP8266 not registering hostname in DHCP

nickidw, 2020-04-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Raspberry pi 2 with ESPHome 1.14.3, all latest versions. Mikrotik wifi

**ESP (ESP32/ESP8266, Board/Sonoff):** 

ESP8266, d1 mini, ESPHome version 1.14.3

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
I would like to register a DNS entry in the Mikrotik to enable ESPHome on hass.io to be able to connect to the ESP8266, but for that to work I need the ESP8266 to send a hostname with DHCP. I cannot use mDNS in this instance because the two devices are on different networks.

I noticed that the d1 mini I'm trying to get running does not register a hostname in Mikrotik DHCP Leases. I captured the DHCP packets for a sonoff running tasmota on bootup, as well as for the ESP8266 running ESPHome, it is attached.

When comparing the two different DHCP conversations it is clear that the sonoff sends a hostname with the DHCP discover request (the 1st request in the sequence), but the ESP9266 does not. This seems to be the cause why the DHCP does not have a hostname recorded for the lease.

[dhcp.zip](https://github.com/esphome/issues/files/4422986/dhcp.zip)

 I've tried it in a better position with the same results.

**Problem-relevant YAML-configuration entries:**
esphome:
  name: d1minisensor
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "dewha"
  password: "password"
  use_address: d1minisensor.dewha.local
  domain: .dewha.local
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "D1Minisensor Fallback Hotspot"
    password: "4fFGUXRXuihI"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

**Logs (if applicable):**

```
[20:21:11][I][app:100]: ESPHome version 1.14.3 compiled on Apr  2 2020, 19:44:45
[20:21:11][C][wifi:415]: WiFi:
[20:21:11][C][wifi:283]:   SSID: 'dewha'
[20:21:11][C][wifi:284]:   IP Address: 192.168.10.249
[20:21:11][C][wifi:286]:   BSSID: D6:CA:6D:13:C8:9D
[20:21:11][C][wifi:287]:   Hostname: 'd1minisensor'
[20:21:11][C][wifi:291]:   Signal strength: -72 dB ▂▄▆█
[20:21:11][C][wifi:295]:   Channel: 1
[20:21:11][C][wifi:296]:   Subnet: 255.255.255.0
[20:21:12][C][wifi:297]:   Gateway: 192.168.10.1
[20:21:12][C][wifi:298]:   DNS1: 192.168.10.1
[20:21:12][C][wifi:299]:   DNS2: 208.67.222.222
[20:21:12][C][logger:175]: Logger:
[20:21:12][C][logger:176]:   Level: DEBUG
[20:21:12][C][logger:177]:   Log Baud Rate: 115200
[20:21:12][C][logger:178]:   Hardware UART: UART0
[20:21:12][C][captive_portal:169]: Captive Portal:
[20:21:12][C][ota:029]: Over-The-Air Updates:
[20:21:12][C][ota:030]:   Address: d1minisensor.dewha.local:8266
[20:21:12][C][api:095]: API Server:
[20:21:12][C][api:096]:   Address: d1minisensor.dewha.local:6053

```

**Additional information and things you've tried:**

meijerwynand, 2020-04-07:
Greetings, 

I am having the same challenges running Mikrotik hardware myself. Speaking under correction, due to the nature of mDNS it is not a feature Mikrotik will be making available on their hardware. 

I too have noticed, once a device has been flashed with esphome, on first boot the hostname is blank in the DHCP leases. However, once the lease is renewed, the hostname is then updated in the leases table.

To address the 2 issues at hand, blank hostname and then no mDNS I have done the following:

1. For the blank hostname I have just set my DHCP lease time low (5min) that on the next request it resolves the blank.
2. My Mikrotik hardware for my IoT network facilitates a "mDNS" workaround by creating a static DNS entry once a DHCP lease is issued and also cleaning up after itself. My HA server uses my Mikrotik DNS for the .home lookups. 

Notes: 
- Keeping DNS on 1d TTL creates issues if the device if and you are doing a looking, the HA server caches the A record then as 127.0.0.1 and your ESP device is "missing" for 1 day... set TTL low (example in code below is for 10minutes)
- I also did run Pi-hole for a while and it worked well, until there is disconnect / reboot. Then you have to manage your network on 2 different devices, with different subsets and different vlans... The same with running avahi on the local HA server.. you end up managing your network on different devices making troubleshooting more troublesome. 

If all goes up in smoke, I plug in a new/spare Mikrotik, load config and continue life as usual. Creating separation of concerns by leaving the complex networking on the hardware which already has a low power consumption.

[image]

[image]

## Mikrotik DHCP leases script
```
# DNS record for DHCP lease

:local topdomain;
:local FullHostName;
:local NoUpdate false;

# Configure your domain
:set topdomain "home";

:set FullHostName ($"lease-hostname" . "." . $topdomain)

/ip dns static;

:if ($leaseBound = 1) do={

:foreach n in [find] do={
# If a static DNS entry is the same as the lease then leave it and set a flag to avoid DNS Update
:if (([get $n name] = $"FullHostName") and ([get $n address] = $leaseActIP)) do={
:set NoUpdate true;
} else={
# If some DNS entry with same fully qualified domain name or same address already exist remove it
:if (([get $n name] = $"FullHostName") or ([get $n address] = $leaseActIP)) do={
   :log info ("Removing from Static DNS : " . [get $n name] .  " @ " . [get $n address]);
    remove $n;
  }
 }
}
# Add new Static DNS Entry if necessary
:if ($NoUpdate = false) do={
:log info ("Adding to Static DNS : " . $"FullHostName" .  " @ " . $leaseActIP);
add name=($"lease-hostname" . "." . $topdomain) address=$leaseActIP ttl=600;
 }
# remove DNS entry at DHCP release :
} else={
:log info ("Clear static DNS entry at DHCP Release : " . $"FullHostName" . " @ " . $leaseActIP);
:foreach n in [find] do={
:if (([get $n name] = $"FullHostName") and ([get $n address] = $leaseActIP)) do={
remove $n;
  }
 }
}
```
This may not resolve your highlighted and identified issues, however I trust this may be of some use to you or others.

nickidw, 2020-04-08:
Thank you. I did put a DNS script in place, but don't want to implement all kinds of clunky as workarounds.

DAVe3283, 2020-08-18:
This affects the ESP32 as well. It looks like [it is being fixed upstream](https://github.com/espressif/esp-lwip/pull/6#issuecomment-607344793), though I am not sure what esptool version(s) got the fix and which are still waiting.

There also seems to be [a workaround](https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-508558849), if the upstream fix is not available.

ilium007, 2021-05-23:
Still an issue in 1.18.0

githubcdr, 2021-06-09:
Same on ESP32 here

ilium007, 2021-06-09:
This issue on ESP32 can be resolved by changed Arduino framework version - read through this whole issue thread. Can't be fixed on ESP8266 at the moment.

ilium007, 2021-06-10:
> Same on ESP32 here

I got my ESP32 working by bumping Arduino framework version

```
esphome:
  name: ${device_name}
  comment: ${friendly_name}
  platform: ESP32
  board: esp-wrover-kit
  arduino_version: espressif32@3.2.0
```

ilium007, 2021-06-13:
How do we open this again so that an ESP8266 fix can be implemented? It's a real issue.

davet2001, 2021-07-20:
This seems to be a problem for me also. I want to enter the dhcp hostname when setting up in HA but it doesn't work.  mydevice.local does let me set up the device, but restart of HA causes all sensors to go unavailable. Static IP is a last resort.  I'm using a sonoff esp8266.

ilium007, 2021-07-20:
> This seems to be a problem for me also. I want to enter the dhcp hostname when setting up in HA but it doesn't work.  mydevice.local does let me set up the device, but restart of HA causes all sensors to go unavailable. Static IP is a last resort.  I'm using a sonoff esp8266.

What router are you using?

davet2001, 2021-07-20:
> What router are you using?

UniFi routers with isc dhcp and bind9.

ilium007, 2021-07-21:
I feel like I am just talking to myself here but I have done more investigation and packet captures during the DHCP process. The ESP8266 running ESPHome version 1.20.0-dev.

From a cold started (no power applied) the ESP8266 (Adafruit Huzzah breakout) only sends a DHCP DISCOVER packet followed by a DHCP REQUEST packet. The DISCOVER packet has no hostname included. By just sending these two packets the device successfully obtains an IP address.

[routeros_6_dhcp_esp8266.pcap.zip](https://github.com/esphome/issues/files/6852472/routeros_6_dhcp_esp8266.pcap.zip)

This is in contrast to a MacBook Pro connected to the same router where I see the expected DISCOVER, OFFER, REQUEST and ACK packets.

[routeros_6_dhcp_macbook.pcap.zip](https://github.com/esphome/issues/files/6852484/routeros_6_dhcp_macbook.pcap.zip)

Until this gets resolved I can go no further with ESPHome and designing custom hardware based on the ESP8266 / ESP32.

davet2001, 2021-07-27:
I believe I have found the root cause of why this was occurring on my setup.
### Short version:
My device name contained an illegal '_' character.  This was rejected by the local dns server. Renaming fixed it.

### Long version:
My device is for a domestic hot water tank, called 'dhw_tank'
Server logs during esp device initialisation:
`$ journalctl -f -u bind9.service`
```
Jul 27 21:40:33 myserver.mydomain.lan named[831]: zone mydomain.lan/IN: dhw_tank.mydomain.lan/A: bad owner name (check-names)
```

A quick google told me that [some host name characters are illegal](https://support.microfocus.com/kb/doc.php?id=7001908).
After [renaming my device](https://community.home-assistant.io/t/esphome-change-device-name/145936/6) to 'dhw-tank', then re-installing (flashing) it via the esp dashboard, I got this out of bind9 which looks much more healthy:
```
Jul 27 21:46:13 myserver.mydomain.lan named[831]: client @0xaf125e80 127.0.0.1#38555/key rndc-key: updating zone 'mydomain.lan/IN': adding an RR at 'dhw-tank.mydomain.lan' A 192.168.0.160
Jul 27 21:46:13 myserver.mydomain.lan named[831]: client @0xaf125e80 127.0.0.1#38555/key rndc-key: updating zone 'mydomain.lan/IN': adding an RR at 'dhw-tank.mydomain.lan' TXT "00548e72e291766c4cfe87690746a3ccbc"
Jul 27 21:46:14 myserver.mydomain.lan named[831]: client @0xaf3fb250 127.0.0.1#51525/key rndc-key: signer "rndc-key" approved
Jul 27 21:46:14 myserver.mydomain.lan named[831]: client @0xaf3fb250 127.0.0.1#51525/key rndc-key: updating zone '0.168.192.in-addr.arpa/IN': deleting rrset at '160.0.168.192.in-addr.arpa' PTR
Jul 27 21:46:14 myserver.mydomain.lan named[831]: client @0xaf3fb250 127.0.0.1#51525/key rndc-key: updating zone '0.168.192.in-addr.arpa/IN': adding an RR at '160.0.168.192.in-addr.arpa' PTR dhw-tank.mydomain.lan.
Jul 27 21:46:14 myserver.mydomain.lan named[831]: zone 0.168.192.in-addr.arpa/IN: sending notifies (serial 869)
Jul 27 21:46:14 myserver.mydomain.lan named[831]: resolver priming query complete
```
In HA I removed and re-added the device, entering the full domain name then restarted HA.
All sensors still working and the device is pingable via dhw-tank.mydomain.lan.
Also restarted the RPi that is running HA and bind9, everything still working fine after reboot.

Suggestion @ilium007: maybe try hardcoding a string in place of `${device_name}`, to see if that fixes or changes the problem.

ilium007, 2021-07-27:
I have already removed and tested without the hyphen as I suspected that may have caused an issue. This did not fix the issue in my case.

ayavilevich, 2021-08-12:
I have the same symptoms, no hostname on first dhcp lease but then hostname registers once the lease expires.
Similar setup: ESP-01 (ESP8266), MikroTik router BUT I am running Tasmota. Specifically I tried Tasmota 9.5 and 8.6 .

@nickidw do you have a record of what tasmota version you tried in your test where it registered ok on the first try? Maybe newer builds of tasmota also got this problem from ESP's wifi implementation.

No underscores in the hostname in my case.

ilium007, 2021-08-12:
The underscore in the hostname has nothing to do with the problem. The issue in the Mikrotik case is that their implementation requires the dhcp option 12 hostname to be sent in both the dhcp discovery and request packets. The ESP8266 does not send the hostname in the dhcp discovery packet.

ayavilevich, 2021-08-12:
@ilium007 Ok. Strange that it works fine on a renew... Did you get a response from Mikrotik?

here are your other cases for reference:
https://github.com/esphome/issues/issues/2069
https://forum.mikrotik.com/viewtopic.php?t=176761

p.s. for now I worked around this by defining the hostname manually for the right mac address in the dhcp->dns script of the router.

ilium007, 2021-08-12:
I am still waiting on a response from Mikrotik. Can you share your DHCP lease script?

ayavilevich, 2021-08-12:
> 
> 
> I am still waiting on a response from Mikrotik. Can you share your DHCP lease script?

sure. See https://pastebin.com/mU1zAVk0
You would set any manual overrides in the top.

foorschtbar, 2021-08-13:
Yes, thats a behavior i could see with all ESP8266 (NodeMCU, Wemos, ESP-07 Chips etc.) and with all devices that had a ESP8266 inside (some Smart Sockets or Light Bulbs). I contacted Mikrotik support in June with several Wireshark logs, but they can't or won't help me. I have now sent them a link to this thread and hope they will fix this behavior.

iksiu6, 2022-04-10:
I have sollution for Mikrotik users who have problem with ESP8266 where hostname is empty.
In Your Mikrotik router go to IP > DNS > Static and add name(were it will be host name of Your ESP) and IP of Your ESP.

It`s not resolve problem with ESPhome/heostname but will be working.