esphome/issues #1125: No "INFO" log level, ESP_LOGI shown as "DEBUG" but sometimes should be "WARN" or "ERROR"

PieBru, 2020-04-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**Affected component:** logger

**Description of problem:**
1. A logger::level `INFO` is missing?
2. Some `ESP_LOGI()` sould be `WARN` or `ERROR`?

**Problem-relevant YAML-configuration entries:**
```yaml
logger:           # https://esphome.io/components/logger.html
  level: DEBUG    # Otherwise no [I] messages are shown :(
```

**Logs (if applicable):**

The `[I]` lines are very useful, but are enabled only with `DEBUG` ore more verbosity.
```
PASTE DEBUG LOG HERE
[19:36:28][I][sgp30:112]: Current eCO2 baseline: 0x8C64, TVOC baseline: 0x86AC
[19:36:36][D][sensor:092]: 'TAQ Uptime': Sending state 4776.76221 s with 0 decimals of accuracy
[19:36:37][D][bme680.sensor:314]: Got temperature=17.3°C pressure=1010.7hPa humidity=43.7% gas_resistance=33937.0Ω
[19:36:38][D][sgp30:237]: Got eCO2=400.0ppm TVOC=0.0ppb
[19:36:38][D][sgp30:131]: External compensation data received: Humidity 39.97%
[19:36:38][D][sgp30:141]: External compensation data received: Temperature 15.21°C
[19:36:38][D][sgp30:150]: Calculated Absolute humidity: 5.180 g/m³ (0x052E)
[19:36:38][I][sgp30:112]: Current eCO2 baseline: 0x8C64, TVOC baseline: 0x86AC
[19:36:47][D][bme680.sensor:314]: Got temperature=17.3°C pressure=1010.7hPa humidity=42.0% gas_resistance=34053.0Ω
[19:36:48][D][sgp30:237]: Got eCO2=400.0ppm TVOC=0.0ppb
[19:36:48][D][sgp30:131]: External compensation data received: Humidity 39.97%
[19:36:48][D][sgp30:141]: External compensation data received: Temperature 15.21°C
[19:36:48][D][sgp30:150]: Calculated Absolute humidity: 5.180 g/m³ (0x052E)
...
```

**Additional information and things you've tried:**
As an example, look at ESPHome `i2c.cpp` :
IMO the row `ESP_LOGI(TAG, "Unknown error at address 0x%02X", address);` should be an 'ERROR`, while the row ` ESP_LOGI(TAG, "Found no i2c devices!");` should be a `WARN`.
```C++
void I2CComponent::dump_config() {
  ESP_LOGCONFIG(TAG, "I2C Bus:");
  ESP_LOGCONFIG(TAG, "  SDA Pin: GPIO%u", this->sda_pin_);
  ESP_LOGCONFIG(TAG, "  SCL Pin: GPIO%u", this->scl_pin_);
  ESP_LOGCONFIG(TAG, "  Frequency: %u Hz", this->frequency_);
  if (this->scan_) {
    ESP_LOGI(TAG, "Scanning i2c bus for active devices...");
    uint8_t found = 0;
    for (uint8_t address = 1; address < 120; address++) {
      this->wire_->beginTransmission(address);
      uint8_t error = this->wire_->endTransmission();

      if (error == 0) {
        ESP_LOGI(TAG, "Found i2c device at address 0x%02X", address);
        found++;
      } else if (error == 4) {
        ESP_LOGI(TAG, "Unknown error at address 0x%02X", address);
      }

      delay(1);
    }
    if (found == 0) {
      ESP_LOGI(TAG, "Found no i2c devices!");
    }
  }
}
```

PieBru, 2020-04-04:
The problem nr. 1 does not exist, I probably screwed up something in my heavuly-structured yaml so i thought it was `INFO` but it actually was `WARN`. Sorry.

The problem nr. 2 is as described.

PieBru, 2020-04-04:
Just to clarify, IMO the `ESP_LOG?` in the C++ snippet above would be more effective this way:
```C++
  if (this->scan_) {
    ESP_LOGD(TAG, "Scanning i2c bus for active devices...");
    uint8_t found = 0;
    for (uint8_t address = 1; address < 120; address++) {
      this->wire_->beginTransmission(address);
      uint8_t error = this->wire_->endTransmission();

      if (error == 0) {
        ESP_LOGI(TAG, "Found i2c device at address 0x%02X", address);
        found++;
      } else if (error == 4) {
        ESP_LOGE(TAG, "Unknown error at address 0x%02X", address);
      }

      delay(1);
    }
    if (found == 0) {
      ESP_LOGW(TAG, "Found no i2c devices!");
    }
  }
```

PieBru, 2020-04-04:
I just found the root cause of the problem nr. 1, it isn't a fully false positive and it may fool other ESPHomers, here is why.

This is the relevant snippet of my ESP32 device yaml config:
```YAML
<<: !include _test_air_quality/substitutions.yaml
<<: !include _test_air_quality/globals.yaml
<<: !include _common/default_esphome.yaml
<<: !include _test_air_quality/logger.yaml
...
```
This is my `logger.yaml`:
```
---
logger:                           # https://esphome.io/components/logger.html
  level: ${logger_level}
  hardware_uart: ${logger_hardware_uart}
  baud_rate: ${logger_baud_rate}
  tx_buffer_size: ${logger_tx_buffer_size}

  logs:                           # https://esphome.io/components/logger.html#logger-manual-tag-specific-levels
    text_sensor.component: ${logger_level_text_sensor}
    ads1115.component: ${logger_level_ads1115}
    bme680.component: ${logger_level_bme680}
    sgp30.component: ${logger_level_sgp30}
    sds011.component: ${logger_level_sds011}
    mics6814.component: ${logger_level_mics6814}
    adc.component: ${logger_level_adc}
```

And this is its corresponding `substitutions.yaml`:
```
  logger_hardware_uart: UART0     # Set logger_baud_rate to 0 to disable logging if using this UART for sensors
  logger_baud_rate: '0'
  logger_tx_buffer_size: '512'    # Default = 512
  logger_level: INFO              # https://esphome.io/components/logger.html#log-levels - NONE, ERROR, WARN, INFO, DEBUG (default), VERBOSE, VERY_VERBOSE
  logger_level_text_sensor: INFO
  logger_level_adc: INFO
  logger_level_ads1115: INFO
  logger_level_bme680: INFO
  logger_level_sgp30: INFO
  logger_level_sds011: INFO
  logger_level_mics6814: INFO
```

So I thought the log level was `INFO`, but the ESP32 boot config log shows `DEBUG'.

That was the quirk: in the `_common/default_esphome.yaml` include there was a `logger:` line without any further parameter.

Ii didn't raise my attention, because I thought the second `logger:` parameters should overwrite the first lone `logger:`, or give at least a WARNING. But I was wrong.

Summary of the problem:
A) More than one `logger:` does not trigger any warning.
B) No warning also if the second `logger:` declares parameters incompatible with the first one.

It may be a YAML parser "feature", but it fooled me as it can fool others as well.

PieBru, 2020-04-04:
Regarding problem nr. 2, here is a log snippet of the same ESP32 device, which further explains my opinion about the WARN/ERROR log level attribution inconsistency:
```
...
[18:01:24][C][bme680.sensor:208]: BME680:
[18:01:24][C][bme680.sensor:209]:   Address: 0x76
[18:01:24][E][bme680.sensor:211]: Communication with BME680 failed!
...
[18:01:24][C][sgp30:178]: SGP30:
[18:01:24][C][sgp30:179]:   Address: 0x58
[18:01:24][W][sgp30:183]: Communication failed! Is the sensor connected?
...
```

OttoWinter, 2020-08-06:
Ah yes that bme680 message should be a warning (error is for messages that prevent the ESP from functioning altogether). This often gets caught in code review, but apparently was missed here (probably my own fault).

Yes the issue you saw is due to the way the `<<` key works in YAML, but now there are `packages` to help with that.

closed 2020-12-18 by stale[bot]