esphome/issues #1126: ESP8266 i2c components stop working after few hours

trylika, 2020-04-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3 on Raspbian. Tested versions of ESPHome - develop and 1.14.3
Tested arduino_version: 2.5.2 and 2.4.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Wemos D1 Mini v3.)

**Affected component:**

https://esphome.io/components/i2c.html

**Description of problem:**
After some odd number of hours (around 7). i2c communication stops working with any sensors connected to ESP8266. Rebooting ESP does not help, only full power off/on makes sensor working again. Log shows only message `Unknown transmit error 4 for address *****` for every connected sensor.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: *
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "*"
  password: "*"
  fast_connect: true
  power_save_mode: HIGH
  manual_ip:
    static_ip: *
    gateway: *
    subnet: *
    dns1: *

logger:
  level: DEBUG

ota:
  password: '*'

mqtt:
  broker: *
  username: *
  password: *
  discovery: True
  birth_message:
  will_message:

i2c:
  id: bus_a
  sda: 4
  scl: 5
  scan: False

sensor:
  - platform: bmp280
    temperature:
      id: temperature
      name: "Temperature"
      oversampling: 16x
      accuracy_decimals: 3
      filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    pressure:
      id: pressure
      name: "Pressure"
      oversampling: 16x
      accuracy_decimals: 3
      filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    address: 0x76
    iir_filter: 16x
    update_interval: 6s

  - platform: ccs811
    eco2:
      id: co2
      name: "CO2"
      accuracy_decimals: 3
      filters:
        - filter_out: 65021.0
        - filter_out: 0.0
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    tvoc:
      id: tvoc
      name: "TVOC"
      accuracy_decimals: 3
      filters:
        - filter_out: 65021.0
        - filter_out: 0.0
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    address: 0x5A
    baseline: 0xDD74
    temperature: temperature
    humidity: humidity
    update_interval: 6s

  - platform: htu21d
    temperature:
      id: temperature2
      name: "Temperature 2"
      accuracy_decimals: 3
      filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    humidity:
      id: humidity
      name: "Humidity"
      accuracy_decimals: 3
      filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    address: 0x40
    update_interval: 6s
```

**Logs (if applicable):**

```
Unknown transmit error 4 for address *****
```

**Additional information and things you've tried:**
* Tried different versions of ESPHome (develop and 1.14.3) - same result
* Tried different versions of arduino_version: 2.5.2 and 2.4.2 - same result
* Tried different boards - sensor works without any problem for extended period of time with ESP32 nodemcu-32s or esp32doit-devkit-v1
* Tried other ESP8266 - same result
* ESP8266 with deep sleep works correctly (but in this case i don't want to use deep sleep)
* Tested separate 3.3v regultor - same result
* Tested different WiFi power saving options - same result
* Tested different length of wires connecting to sensor - same result

akakiz, 2020-05-02:
I have similar issue with ESP32 and SHTC3.
It's works until I shutdown homeassistance vm. All other sensors, not related to i2c continues to works perfect (with mqtt).
After shutdown (or just loss connection between ESP32 and HA) HA VM - SHTC3 continues to works for some time (maybe 5 minutes or more) then blue light on ESP32 (I use it as status indicator) starts fast flashing, which means problems with ESP32.
If I restore HA VM, it doesn't helps until I completely power off/on my ESP32 (resetting ESP32 is not helpful).
If I start my ESP32 without HA it's starts to works, I receiving messages on MQTT broker, but not for long time (problem only on device with i2c).

Is there any suggestion?

```
part of project:

status_led:
  pin: 
    number: GPIO2
    inverted: true

mqtt:
  broker: mqtt.myserver.my
  username: meme
  password: momo

i2c:
  sda: 21
  scl: 22

sensor:
  - platform: shtcx
    temperature:
      name: "SHTC3 temp"
    humidity:
      name: "SHTC3 hum"
    update_interval: 30s

```

brambo123, 2020-07-20:
I run into the same problems.
It seems to have some problems in the ccs811 library, but the i2c library also seems to have some problems.

The i2c library uses the underlying Wire library for many functions.
In the Wire library, some issues are resolved when the status() function is called.
However, this function is never called by the i2c library.
One simple fix to try:
In i2c.cpp, line 132, call "this->wire_->status();" before "return false;".

The ccs811 misses some error handling and passes invalid data during startup phase.
I will soon rewrite some things for the ccs811 library.

closed 2020-12-01 by stale[bot]

bedenko, 2020-12-23:
The problem is still present.
I tried the proposal from @brambo123, but compilation failed with:
```
src/esphome/components/i2c/i2c.cpp: In member function 'bool esphome::i2c::I2CComponent::read_bytes(uint8_t, uint8_t, uint8_t*, uint8_t, uint32_t)':
src/esphome/components/i2c/i2c.cpp:132:18: error: 'class TwoWire' has no member named 'status'
     this->wire_->status();
                  ^
*** [.pioenvs/stopnice/src/esphome/components/i2c/i2c.cpp.o] Error 1
```
My error is:
```
i2c:076]: Unknown transmit error 4 for address 0x68
i2c:076]: Unknown transmit error 5 for address 0x68
```

brambo123, 2020-12-23:
> The problem is still present.
> I tried the proposal from @brambo123, but compilation failed with:
>
> ```
> src/esphome/components/i2c/i2c.cpp: In member function 'bool [...]

I just tried the fix again, and it still compiles without problems.
I'm guessing you are running on an ESP32 and not an ESP8266?

bedenko, 2020-12-23:
Yes, that is correct.
And now i see Issue title :facepalm: 

Do you perhaps have a fix for ESP32?