esphome/issues #1130: Deepsleep above 3h48 consumption bug

mbt45, 2020-04-07:
Hello

I have been trying to make a deepsleep project with 1 wake-up per day unfortunately my battery was running out in 14h max. After careful tests of consumption of the d1mini I discovered the deepsleep function of the esphome doesn't put the d1mini in "REEL" deepsleep when the setting is above 13676869ms.
consumption levels are
13676869ms = 0.2mA
13676870ms = 68mA

Quite a difference

Although the ESP will not be accessible during the time of FAKE deepsleep, it will still consume as much power as if it was fully awake

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

wizu, 2020-04-08:
@mbt45 that's not ESPHome bug, but ESP8266 (which D1mini is) limit. Please read:
https://thingpulse.com/max-deep-sleep-for-esp8266/

mbt45, 2020-04-08:
@wizu  Ok I though ESPHOME used to do the same technique  @stefanbode  made for Tasmota deepsleep. [https://github.com/stefanbode/Sonoff-Tasmota/wiki/Deepsleep](url)

stefanbode, 2020-04-08:
Let me check what’s going wrong. If really there is an overflow we can fix this

stefanbode, 2020-04-08:
Uups this is esphome. In this case not. I just created the driver for tasmota. Here it should work.

stefanbode, 2020-04-08:
For tasmota there is a special procedure that wakes up the device every hour for a few ms and send to sleep again to overcome the limit. Anyhow this is a special implementation I did for tasmota and not standard.

mbt45, 2020-04-08:
Hi @stefanbode 
Sorry I mentioned your work in this esphome post because I though they worked with you to develop the deepsleep component to overcome this limitation. Thanks for showing up though

mbt45, 2020-04-09:
@stefanbode do you think it would be possible to adapt your special procedure to ESPHOME? What would be needed?

stefanbode, 2020-04-09:
The procedure is just easy. You can store the remaining time in rtc memory without loosing it. If wakeup at least every hour you see the remaining time. Subtract 3600 second and goto sleep again. This is on high level the solution.

mbt45, 2020-04-13:
Hi everyone. 
After talking with @stefanbode and looking around I came up with a solution that does not require re-encoding ESPHOME itself. 
There is probably a way to get the hours required automatically to the right multiplier but I don't really have time to search that. If anyone is interested this is the code to overcome the 3h47min limitation

`esphome:
  on_boot:
   - priority: 750  # (This is the Max value, above this priority the reboot counter will not set )
     then:
      - if:
          condition:
            lambda: 'return id(reboot_count) < id(deepsleep_multiplier);'
          then:
          - lambda: 'id(reboot_count) = id(reboot_count)+1;'
          - delay: 70ms # (no not touch, below this delay the reboot counter will not set )
          - deep_sleep.enter: deep_sleep_1
          else:
          - lambda: 'id(reboot_count) = 0;'

   - priority: 300
   # This put the ESP to deepsleep again in case wifi is not available at wake up to save battery
     then:
      - delay: 30s
      - if:
          condition:
            wifi.connected:
          then:
          else:
            - deep_sleep.enter: deep_sleep_1

globals:
   - id: reboot_count
     type: int
     restore_value: yes (keep to yes for the value to be incremented after boot)
     initial_value: '0'

   - id: deepsleep_multiplier
     type: int
     initial_value: '0'  # (set the value you want to extend the deepsleep, no decimal allowed)
   
   

deep_sleep:
    id: deep_sleep_1
    run_duration: 20s # (set this as you much need for all your data to be sent)
    sleep_duration: 10s  #( This value should excede 3h so stay safe. It will be multiplied by the deepsleep_multiplier )
    
    `

closed 2020-04-18 by mbt45