esphome/issues #1133: Gamma Correction breaks constant brightness

Skaronator, 2020-04-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Docker Running the master branch with commit https://github.com/esphome/esphome/commit/8613c02d5c3e40938e39def1671fb37c69c9db11

**ESP (ESP32/ESP8266, Board/Sonoff):**
H801 which is a ESP8266 with 5 channels

**Affected component:**
light

**Description of problem:**

The Constant Brightness Option added in https://github.com/esphome/esphome/pull/1007 doesn't respect the gamma correction setting. With the default value you can notice the dip in brightness but when you cranking up the gamma_correct value to 40 then your light just goes out for a few second before turning on in the other color.

I made a quick video, the exposure is locked on the camera so it doesn't autoadjust: https://streamable.com/2h944w

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: esp8266_pwm
    pin: 12
    id: pwm_b
  - platform: esp8266_pwm
    pin: 13
    id: pwm_g
  - platform: esp8266_pwm
    pin: 15
    id: pwm_r
  - platform: esp8266_pwm
    pin: 14
    id: pwm_cw
  - platform: esp8266_pwm
    pin: 4
    id: pwm_ww

light:
  - platform: cwww
    name: "Deckenlicht Kurz"
    gamma_correct: 40
    cold_white: pwm_cw
    warm_white: pwm_ww
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    constant_brightness: true
  - platform: rgb
    name: "Deckenlicht Kurz RGB"
    gamma_correct: 1.5
    red: pwm_r
    green: pwm_g
    blue: pwm_b

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

First PR by @OttoWinter  https://github.com/esphome/esphome/pull/827
Later merged via https://github.com/esphome/esphome/pull/1007

kroimon, 2020-04-10:
Sorry for not replying earlier and thanks for the detailed description and the video! I have a few questions for you to narrow down the issue:

- What LED strip are you using?
- From https://github.com/esphome/esphome/pull/1007#issuecomment-611225161 it looks like your strip is capable of both RGB and CW/WW? Why are you using a combination of the `rgb` and `cwww` components instead of the `rgbww` component?
- The default `gamma_correct` value is 2.8. I had a similar problem with inconsistent brightness on the Mi Desk Lamp because the two white channels seem to already have gamma correction done in hardware, so I had to set `gamma_correct` to 0. Can you please retry with no gamma correction at all?
- If the issue persists, please set the logger level to at least `VERBOSE` or even `VERY_VERBOSE` and post the output here when changing the color temperature.
- Is the same brightness drop happening when you slowly change the color temperature from cold to white, in small steps at a time?

Just as a reference, this is my config for the Mi Desk Lamp that produces a super smooth color temperature change with absolutely no brightness change:

```yaml
# PWM Outputs
output:
  - platform: esp8266_pwm
    id: pwm_coldwhite
    pin: GPIO4
  - platform: esp8266_pwm
    id: pwm_warmwhite
    pin: GPIO5

# Cold White + Warm White Light
light:
  - platform: cwww
    id: lamp
    name: "Desk Lamp"
    cold_white: pwm_coldwhite
    warm_white: pwm_warmwhite
    cold_white_color_temperature: 153 mireds # 6500 K
    warm_white_color_temperature: 370 mireds # 2700 K
    gamma_correct: 0
    constant_brightness: true
```

Skaronator, 2020-04-10:
> What LED strip are you using?

These are just 2 dumb LED Strips. One for [RGB ](https://www.led-konzept.de/mediafiles//Datasheets/datenblatt-led-streifen-11770.pdf)and another one for [CCT](https://www.led-konzept.de/mediafiles//Datasheets/datenblatt-led-streifen-11930.pdf). They look like this IRL:

[image]

Both strips are connected to a H801 which is just a ESP8266 that is connected to 5 MOSFETs.

> Why are you using a combination of the rgb and cwww components instead of the rgbww component?

I'm not a big fan how home assistant handle the white value in thier frontend. There is a ongoing architecture discussion but until then I just keep these strips seperated https://github.com/home-assistant/architecture/issues/123

> The default gamma_correct value is 2.8. I had a similar problem with inconsistent brightness on the Mi Desk Lamp because the two white channels seem to already have gamma correction done in hardware, so I had to set gamma_correct to 0. Can you please retry with no gamma correction at all?

Setting gamma correction to 0 fixed this issue which confirmes that the gamma correction is breaking the constant_brightness option.

> Is the same brightness drop happening when you slowly change the color temperature from cold to white, in small steps at a time?

Yes. In the video posted above you can see at the very end that I set the color temperature between cold and warm white and the light just turns off.

Skaronator, 2020-04-10:
@kroimon @glmnet I nailed down the issue. There are 2 problems with the current code. The first is that the gamma_correction is applied **after** the constant brightness calculation as you can see here:

https://github.com/esphome/esphome/blob/835079ad43593258cd38988209f303ccb4efac7c/esphome/components/light/light_state.cpp#L734-L737

The second issue is that the gamma correction for both channels is calculated individually. They must be calculated together like this:
```c++
const float gamma_correction = gamma_correct(this->brightness_ * this->white_, gamma_correct_);
*cold_white = this->state_ * gamma_correction * cw_fraction;
*warm_white = this->state_ * gamma_correction * ww_fraction;
```

I'll make a PR but I'm not really happy with the code since I now have to pass a lot of arguments though functions.

closed 2020-06-12 by glmnet