esphome/issues #1134: Not connecting to WiFi (such as in issue 455)

prabbit237, 2020-04-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.IO with Docker/ESPHome addon

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**

tl;dr; One solution in issue 445 (and the accepted one) is "erase flash before flashing new firmware." This is for those where that didn't work.

There was a HUGE thread at https://github.com/esphome/issues/issues/455 about WiFi issues and how people were having issues with it not connecting, not seeing the signal, giving errors when it did see the SSID, etc and people were trying many things such as randomly adding to the yaml or removing components, etc. and really no good solutions in many cases. This almost wound up being in the same line as many of those but had a solution that wasn't obvious and yet wasn't really due to the WiFi itself. So I'm posting this for anyone who may wind up in the same boat and hopefully heading them off before they get trapped in the rabbit-hole that 455 wound up becoming (and that's NO fault of OttoWinter but simply that there was issues that may have looked like they were the same but really weren't, such as here.

See below for the explanation...

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: dallas
    address: 0x9E03119779975528
    id: temperature
    filters:
      - lambda: id(f_temp) = x * 1.8 + 32.0; return id(f_temp);
    unit_of_measurement: "°F"
    on_value:
      - if:
          condition:
            for:
              time: 3min
              condition:
                lambda: |-
                  //
                  //..........................this is the line that caused all my issues
                  id(mqtt_client).publish("mikes_ac_sensor/stat/Temp", id(temp_string));
                  //..........................
                  //
                  return id(disabled);		

```
(I snipped some other lines in that section that were above the "on_value" to tidy it up some for here.)

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

In the above yaml snippet, I had an MQTT message being published in the YAML. When that was in there, the WiFi gave the same issues with dropping, not connecting, etc. By removing the one line reading:
````
                  id(mqtt_client).publish("mikes_ac_sensor/stat/Temp", id(temp_string));
````
It started working fine. Apparently that part was being ran more often than I realized and it was overloading the networking even before the networking could get started up. Taking it out of there and adding it to an interval: section to be run every several seconds, the issue was resolved. So if anyone has such an issue with the networking, look for lambda code that may be running more frequently than expected and that may be over-lloading the networking stack itself.

So simple solution but not one that was obvious at first. And other people who posted in #455 that "adding the web_server:" or "removing a sensor" or "doing this seemingly random change" fixed their issue, they may have been slowing down the processing of various sections or removing the offending code or otherwise shifting the timing of networking access without even realizing **why** it was having an effect.

glmnet, 2020-04-27:
Thanks for sharing.

Actually the “random” issue was related to WiFi power and flash reading via spi and bad design on the hardware. 

It’s very common to people to say their issue is the same than otters but many times it’s not.

glmnet, 2020-04-27:
Btw if you posted that code here many of us would have spotted the issue right away.

closed 2020-04-27 by glmnet