esphome/issues #1136: bug in ESPBTUUID::to_string()

mgithubreq, 2020-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

to_string() produces wrong BLE UUID values.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

```
# diff -Naur /tmp/original.cpp ./.local/lib/python3.7/site-packages/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp
--- /tmp/original.cpp   2020-04-12 08:38:43.420505356 +0000
+++ ./.local/lib/python3.7/site-packages/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp     2020-04-12 08:39:27.563563014 +0000
@@ -246,11 +246,11 @@
   char sbuf[64];
   switch (this->uuid_.len) {
     case ESP_UUID_LEN_16:
-      sprintf(sbuf, "%02X:%02X", this->uuid_.uuid.uuid16 >> 8, this->uuid_.uuid.uuid16);
+      sprintf(sbuf, "%02X:%02X", (this->uuid_.uuid.uuid16 >> 8) & 0xff, this->uuid_.uuid.uuid16 & 0xff);
       break;
     case ESP_UUID_LEN_32:
-      sprintf(sbuf, "%02X:%02X:%02X:%02X", this->uuid_.uuid.uuid32 >> 24, this->uuid_.uuid.uuid32 >> 16,
-              this->uuid_.uuid.uuid32 >> 8, this->uuid_.uuid.uuid32);
+      sprintf(sbuf, "%02X:%02X:%02X:%02X", (this->uuid_.uuid.uuid32 >> 24) & 0xff, (this->uuid_.uuid.uuid32 >> 16) & 0xff,
+              (this->uuid_.uuid.uuid32 >> 8) & 0xff, this->uuid_.uuid.uuid32 & 0xff);
       break;
     default:
     case ESP_UUID_LEN_128:

```

mgithubreq, 2020-08-10:
Any reason this wasn't taken up by anyone?

OttoWinter, 2020-08-11:
Looks like https://github.com/esphome/esphome/pull/1177 fixes it

closed 2020-12-23 by stale[bot]