esphome/issues #1141: Unable to Compile ESPHome Firware

mdframe, 2020-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

```
INFO Reading configuration /config/esphome/softenersalt.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/softenersalt
Processing softenersalt (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22Update%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fea27dcbe10>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22Update%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fea27d6b0b8>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=2)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22Update%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fea27dcb470>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=3)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22Update%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fea27d6b2e8>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=4)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22Update%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fea27d6b9b0>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=5)
APIRequestError: [API] Could not connect to PlatformIO API Service. Please try later.:
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif8266/builder/main.py", line 203:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 122:
    _build_project_deps(env)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 1043:
    project.install_dependencies()
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 878:
    lm.install(uri)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 352:
    interactive=interactive)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 253:
    cache_valid="1h")
  File "/usr/local/lib/python3.6/dist-packages/platformio/util.py", line 377:
    "Could not connect to PlatformIO API Service. "
========================= [FAILED] Took 190.09 seconds =========================

```

**Additional information and things you've tried:**

mdframe, 2020-04-12:
Closing as I tracked it down to a blocked IP from from ISP.

Misiu, 2020-04-16:
@mdframe please close this issue then :)

closed 2020-04-16 by mdframe