esphome/issues #1143: rmt: RMT RX BUFFER FULL

TheEris, 2020-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on VM on Unraid
Version 1.14.3 

**ESP (ESP32/ESP8266, Board/Sonoff):**

nodemcu-32s

**Affected component:**

[https://esphome.io/components/remote_receiver.html](url)
probably (?)

**Description of problem:**

In the log, there is spam of errors and the ESP managed to crash (no logs from that) and get stuck. Had to do hard reboot. After reboot, I've noticed hundreds of lines of "RMT RX BUFFER FULL"

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: announcer
  platform: ESP32
  board: nodemcu-32s

wifi:
  networks:
    - ssid: "xxx"
      password: "xxx"
    - ssid: "xxx"
      password: "xxx"
    - ssid: "xxx"
      password: "xxx"
  power_save_mode: none
  manual_ip:
    static_ip: xxx
    gateway: xxx
    subnet: xxx
  ap:
    ssid: "Announcer Fallback Hotspot"
    password: "xxxxxxx"

captive_portal:

# Enable logging
logger:

ota:

status_led:
  pin: GPIO2

remote_receiver:
  pin: 
    number: GPIO21
    inverted: True
  dump:
    - rc_switch
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb

sensor:
  - platform: dht
    pin: GPIO18
    model: DHT22
    temperature:
      name: "temperature 2"
    humidity:
      name: "humidity 2"
    update_interval: 60s
    
binary_sensor:
  - platform: gpio
    name: "MP3 playing"
    pin: 
      number: GPIO19
      inverted: True

  - platform: remote_receiver
    name: "RF switch 1"
    rc_switch_raw:
      code: '001110101101100000010010'
      protocol: 6
    filters:
      - delayed_off: 1s
  
  - platform: remote_receiver
    name: "RF switch 2"
    rc_switch_raw:
      code: '001110101101100000011000'
      protocol: 6    
    filters:
      - delayed_off: 1s

  - platform: remote_receiver
    name: "zvonek"
    rc_switch_raw:
      code: '100000100000100010100001'
      protocol: 6
    filters:
      - delayed_off: 1s
    
  - platform: remote_receiver
    name: "dvere_1"
    rc_switch_raw:
      code: '001000100101110100010110'
      protocol: 6
    filters:
      - delayed_off: 1s    
      
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600

dfplayer:

api:
  services:
  - service: dfplayer_next
    then:
      - dfplayer.play_next:
  - service: dfplayer_previous
    then:
      - dfplayer.play_previous:
  - service: dfplayer_play
    variables:
      file: int
    then:
      - dfplayer.play: !lambda 'return file;'
      
  - service: dfplayer_play_loop
    variables:
      file: int
      loop_: bool
    then:
      - dfplayer.play:
          file: !lambda 'return file;'
          loop: !lambda 'return loop_;'

  - service: dfplayer_play_folder
    variables:
      folder: int
      file: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          file: !lambda 'return file;'

  - service: dfplayer_play_loop_folder
    variables:
      folder: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          loop: True

  - service: dfplayer_set_device_tf
    then:
      - dfplayer.set_device: TF_CARD

  - service: dfplayer_set_device_usb
    then:
      - dfplayer.set_device: USB

  - service: dfplayer_set_volume
    variables:
      volume: int
    then:
      - dfplayer.set_volume: !lambda 'return volume;'
  - service: dfplayer_set_eq
    variables:
      preset: int
    then:
      - dfplayer.set_eq: !lambda 'return static_cast<dfplayer::EqPreset>(preset);'

  - service: dfplayer_sleep
    then:
      - dfplayer.sleep

  - service: dfplayer_reset
    then:
      - dfplayer.reset

  - service: dfplayer_start
    then:
      - dfplayer.start

  - service: dfplayer_pause
    then:
      - dfplayer.pause

  - service: dfplayer_stop
    then:
      - dfplayer.stop

  - service: dfplayer_random
    then:
      - dfplayer.random

```

**Logs (if applicable):**

```
[20:31:43]E (111) rmt: RMT[0] ERR
[20:31:43]E (111) rmt: status: 0x14000040
...
[20:49:02]E (1039860) rmt: RMT RX BUFFER FULL
[20:49:02]E (1039886) rmt: RMT RX BUFFER FULL
[20:49:02]E (1039927) rmt: RMT RX BUFFER FULL
[20:49:02]E (1039945) rmt: RMT RX BUFFER FULL
[20:49:02]E (1039966) rmt: RMT RX BUFFER FULL
...

full log here: [https://pastebin.com/FCPXeM4A](url)

```

**Additional information and things you've tried:**

Reflashing with the newest firmware didn't help. Same problem.

glmnet, 2020-04-14:
Is it happening with other config a of remote receiver?

@puu any ideas?

TheEris, 2020-04-14:
tried the minimal setup with this configuration

```yalm
esphome:
  name: announcer
  platform: ESP32
  board: nodemcu-32s

wifi:
  networks:
    - ssid: "xxxxxxx"
      password: "xxxxxxx"
    - ssid: "xxxxxxx"
      password: "xxxxxxx"
    - ssid: "xxxxxxx"
      password: "xxxxxxx"
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.0.85
    gateway: 192.168.0.1
    subnet: 255.255.254.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Announcer Fallback Hotspot"
    password: "xxxxxxx"

captive_portal:

# Enable logging
logger:

ota:

status_led:
  pin: GPIO2

remote_receiver:
  pin: 
    number: GPIO21
    inverted: True
  dump:
    - rc_switch
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb
```

but not much changed, still getting this in the log (serial):
```
E (9016) rmt: RMT[0] ERR
E (9016) rmt: status: 0x00000000
```

but it doesn't spam it, just this.

also, it sais that I don't have a logger component setup? (in the hassio)
"ERROR No remote or local logging method configured (api/mqtt/logger)" ?????

glmnet, 2020-04-14:
yes, you need `api:` or `mqtt:`

glmnet, 2020-04-14:
it might be related to your receiver or hw configuration, maybe picking too much noise in the 433mhz causes the receiver component to not find a "silence" to understand the transmission ended and this fills the buffer.

Does it shows you anything when you press buttons?

You can try redicing `idle: 4ms`, e.g. `2ms` but I'm not sure which value is needed for the thing to work properly.

glmnet, 2020-04-14:
if you are not missing buttons key presses you can ignore the error may be?

TheEris, 2020-04-14:
Ah yes, I missed the api: ...
Yes, I have a lot of noise around here. But I didn't have much of a problem picking up the buttons. 
I've tried the 2ms idle and still the same problem.

as for the hardware, I'm using some generic RXB6 receiver of an eBay. I can plug in an oscilloscope and take the picture of the noise coming into the ESP if it will help?

glmnet, 2020-04-14:
Yeah maybe some little pull up or pull down will help. I have never connected the receivers to the scope but the cheapest ones are really bad. I’ve only used them to learn codes from remotes

TheEris, 2020-04-14:
I'll try to hook up some resistors to it tomorrow and see if it helps. Also any tips on good RF receiver?

TheEris, 2020-05-18:
update on this issue:
I've tried pullup/down on the input. I've tried to remove everything but the receiver in the code but nothing. The esp keeps throwing "RMT RX BUFFER FULL" in the serial log.

as for now, my solution is to switch to esp8266 where it works without any problems and its noticeably faster to act/do something with HA automations

sara-patel, 2020-08-11:
Hello, even I m facing the same issue even after changing the  .mem_block_num = 4,   I get an error as
 E (311723) rmt: RMT RX BUFFER ERROR . Please anyone suggest any solution for the same and if I change the me-block_num to 4 in RX config then is there any changes to be done on the TX config ?

sara-patel, 2020-08-13:
Thanks for your response but I still didn't find any RX configurations n TX configurations which will not give any RX BUFFER ERROR

antonmeyer, 2020-08-23:
I struggling with a similar problem. that message is likely to come from deep in the espresif or freetos core lib.
even disabling the esp logging via esp_log_level_set("*", ESP_LOG_NONE); did not help.

sara-patel, 2020-08-24:
so what should be done for removing this error ? kindly suggest any solution for the problem.

antonmeyer, 2020-08-24:
it was more a #metoo comment. I'm still searching for the switch to disable the message. as esp_log does not help, I hope for the switch in a config file. the perfect solution would be, to have a logging-off option for the ringbuffer or a configurable logging rate (same message only once per minute and a counter for the number of same messages) but as this is deep in the framework, it would be a long way.

antonmeyer, 2020-08-26:
in general I got a bit frustrated about the RMT implementation. there seam to be now way to check if there are enough items in the buffer. vRingbufferGetInfo looks like to be updated after xRingbufferReceive call. the hal implementation rmtRead with the callback function misses a user defined parameter handling for non static data ...

sara-patel, 2020-08-27:
ohk so wht do you think we should do ?

antonmeyer, 2020-08-27:
as mentioned above uncomment the log message in esp-idf/components/driver/rmt.c and re-compile everything. A bit softer: change it to ESP_LOGE(RMT_TAG, ... ) and control it via esp_log_level_set.
If you have spare time convince the manufacture to improve it.
I can also not exclude, that there is a way to disable ESP_EARLY_LOGE messages. May be you can find one.
And as your question is very generic:
stay healthy and enjoy your life ;-)

sara-patel, 2020-08-28:
ok thanks . will try to comment it and see into more detail.

antonmeyer, 2020-08-30:
@sara-patel I just recognized that your issue is a different one.
For mine I just found a better solution (just testing)
 rmt_set_err_intr_en(rmt_rx.channel, pdFALSE);
I guess this suppresses the error interrupt and with that the message.
Your case sounds like a real error. I'm not familiar with esphome. May be you can not allocate 4x buffers? without esphome rmt_rx.mem_block_num = 8; works for me.

ogry, 2020-10-24:
Me too. It's anoying to cannot go further investigating this promising way to configure ESP32 for remotes.

BTW, after lots of RMT RX BUFFER FULL, they are also a bit of strange error logs:
```
[00:07:23]E (99394) rmt: RMT[0] ERR
[00:07:23]E (99394) rmt: status: 0x04000000
```

After first this happend, boot is delayed up to 3 minutes, even more. Then appears the message:
`[I][ota:046]: Boot seems successful, resetting boot loop counter.`

May be all this is connected with the last changes in the core
https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/rmt.h

After a full day breaking my head with parameter combinations, the defaults works the better...

```
remote_receiver:
  pin: 
    number: GPIO27
#    inverted: True
  dump: # all
    - rc_switch
#    - raw
  tolerance: 50%
#  filter: 250us
#  idle: 4ms
#  buffer_size: 2kb

#  tolerance: 60%
#  filter: 2ms
#  idle: 50ms
#  buffer_size: 256
##  memory_blocks: 8
```

closed 2021-03-08 by stale[bot]