esphome/issues #1144: SDS011 update_interval configuration ignored

blackskull182, 2020-04-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 3.12 on Raspberry Pi 3 Model B+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266(LiLon NodeMCU V3)

**Affected component:**

https://esphome.io/components/sensor/sds011.html

**Description of problem:**

ESPHome ignore completely the "update_interval" configuration so the sensor still powered ON and in continuous measurement.

According to the documentation this is a critical issue because it reduces the lifetime of the SDS011.
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_air
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "XXXXXXXXXXX"
  password: "XXXXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Air Fallback Hotspot"
    password: "XXXXXXXXXXX"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "XXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXX"

uart:
  rx_pin: D0
  tx_pin: D1
  baud_rate: 9600

sensor:
  - platform: sds011
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
    update_interval: 5min

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

swifty99, 2020-04-26:
true, I have the same issue on a wemos d1 mini. Even with nothing else attached.

blackskull182, 2020-04-26:
> true, I have the same issue on a wemos d1 mini. Even with nothing else attached.
> 
> I have been looking into the code and the available documentation:
> [https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwjT-rio8IXpAhVlxaYKHSIpC6gQFjAAegQIARAB&url=https%3A%2F%2Fnettigo.pl%2Fattachments%2F415&usg=AOvVaw2uQTA-3dYZul2uZvhYITRK](url)
> 
> I can not find an issue there. however I did not debug the serial yet. Maybe our SDS011 firmware is too old.
> There is another way to send the device to sleep manually and wake it up periodically. Maybe I find some time to try this.

Ok, let me know please :)

lenisko, 2020-08-11:
Hey there!
Deploying configuration with `0min` and after then with `5min` solved it to me.

```
esphome:
  name: balcony_particulate
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s
  fast_connect: true
  ap:
    ssid: "Balcony Particulate Fallback"
    password: !secret ap_password

captive_portal:

logger:
  baud_rate: 0

uart:
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600

api:
  password: !secret password
  reboot_timeout: 0s

ota:
  password: !secret password

sensor:
  - platform: uptime
    name: "Balcony Particulate Uptime"

  - platform: wifi_signal
    name: "Balcony Particulate WiFi Signal"
    update_interval: 60s

  - platform: sds011
    rx_only: false
    update_interval: 5min  # first deploy config with 0min
    pm_2_5:
      name: "Balcony Particulate <2.5µm"
    pm_10_0:
      name: "Balcony Particulate <10.0µm"

switch:
  - platform: gpio
    pin: 2
    name: "Balcony Particulate OnBoard LED"
```

prutseltje, 2020-08-17:
I have the same issue, temporarily setting the update_interval to 0min does not solve the issue for me.

closed 2020-12-29 by stale[bot]