esphome/issues #1145: Compile does not collect AsyncTCP-esphome - temporary fail in name resolution

g4challenge, 2020-04-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on RPI4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Cam

**Affected component:**

Core/OTA

**Description of problem:**

The problem build starts, yet the name resolution fails.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: espcam
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "+++"
  password: "+++"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espcam Fallback Hotspot"
    password: "+++"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "+++"

ota:
  password: "+++"

esp32_ble_tracker:
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/espcam.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/espcam
Processing espcam (board: esp32cam; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Looking for AsyncTCP-esphome library in registry
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf685aaf0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)

```

**Additional information and things you've tried:**

brandond, 2020-04-13:
Esphome uses platformio. The host you're building on needs to be able to connect to platformio services to pull down libraries such as AsyncTCP.

g4challenge, 2020-04-14:
Yes, curl works just fine, yet I somehow believe the integration of homeassistant (installed via docker) - has problems connecting outside.

the command `cat /etc/resolv.conf` results in:
```
search local.hass.io
nameserver 172.30.32.3
````

curl from inside the container is not possible - using busybox and nslookup results in:

```
Server:         1.0.0.1
Address:        1.0.0.1:53

Non-authoritative answer:
api.platformio.org      canonical name = nm1.platformio.org
Name:   nm1.platformio.org
Address: 31.28.1.238

*** Can't find api.platformio.org: No answer
```

nhendin, 2020-05-01:
I am seeing this exact same issue (pasted below).  It's as if name resolution inside the container is not working. This is a fresh install on a docker container running on Ubuntu 18.04 host. I migrated a BRUH-Multisensor to esphome from MQTT tonight for the first time and it's awesome.  Very clean setup.  I was able to use the docker container to flash over USB and get the node up, but I wanted to start the dashboard to be able to manage nodes moving forward. 

HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f1c93490208>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)

g4challenge, 2020-05-05:
I can confirm nhendin's settings and suspicion - tried to run bash within docker - but the esphome container is unable to lookup api.platformio.org

nhendin, 2020-05-05:
Updated (thanks g4challenge for the reminder).  

I got it working by adding --dns 8.8.8.8 --dns 8.8.4.4 to the docker run command, and then I switched to docker-compose to integrate it into my iot docker stack. Those are the Google public DNS servers, but I am reasonably confident any valid DNS servers that can be reached would work as well.  Interestingly I was not able to run a shell in the container, but was trying "sh" not "bash", that would have helped me find this faster.  Will try that again if needed. 

With this config I am able to update nodes over the air and esphome is able to reach platformio.org to download the bits it needs. 

Here is a snippet from  docker-compose.yaml:

```
version: '3'
services:
 esphome:
    container_name: esphome
    image: esphome/esphome
    volumes:
            - /opt/docker/esphome/:/config
    restart: always
    network_mode: "host"
    dns:
            - 8.8.8.8
            - 8.8.4.4 

```

glmnet, 2020-05-07:
So close this? just DNS issue?

g4challenge, 2020-05-07:
For me at least, with the standard configuration home-assistant and esphome running as docker container - the problem persists. Even after "forcing" dns.

Manually using the above docker-compose.yaml may work, but I am still wondering - if there is no simpler fix for everyone.

nhendin, 2020-05-07:
g4challenge:  May I ask what host OS are you using?  

Is it Ubuntu 18.04?  (that's what I am using), or other OS using systemd-resolved?  

Apparently there can be odd bugs with systemd-resolved (check your /etc/resolve.conf on the host OS for 127.0.0.53 as a nameserver and if so that's likely).  I've asked around and did some more digging and was referred to the following articles from a friend:

1: [https://moss.sh/name-resolution-issue-systemd-resolved/](https://moss.sh/name-resolution-issue-systemd-resolved/)
2: [https://www.unixgr.com/disabling-systemd-resolvd-service-on-ubuntu-systems/](https://www.unixgr.com/disabling-systemd-resolvd-service-on-ubuntu-systems/)

(1) goes into the odd complexities of this setup which is the default on Ubuntu 18.04 (and perhaps other versions), and (2) goes into how to turn it off.

I don't think it's as simple as the loopback address not being found inside the container, since I did have "host" mode networking activated.

closed 2020-09-18 by stale[bot]