esphome/issues #1148: Question: Security and reverse connection

ski7777, 2020-04-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
All platforms

**Affected component:**
native API

**Description of problem:**
As far as I know the communication between the ESP and my WiFi AP is encrypted but it is unencrypted to any other network participants. Is there a way to encrypt this data properly?
Secondly, is there a way to let esphome establish the connection to HomeAssistant?
I want to run HomeAssistant on my vServer. With encryption this would be secure and if esphome could start the connection I would not need to open ports or use dyndns every time I launch a new esphome module.
If this is not implemented yet I would offer my python knowledge to solve those problems. If someone tells me where to start in the source code I could try on my own.

**Problem-relevant YAML-configuration entries:**
Not applicable

**Logs (if applicable):**

Not applicable

**Additional information and things you've tried:**

Not applicable

Raphael

glmnet, 2020-04-18:
Please search before asking

See https://github.com/esphome/feature-requests/issues/46
See https://github.com/esphome/feature-requests/issues/133

closed 2020-04-18 by glmnet

ski7777, 2020-04-18:
I did some kind of searching but I could not find anything that seemed right. Thank for the links.

Raphael