esphome/issues #1149: Where is cover position kept? Flash/RTC Memory?

theone11, 2020-04-18:
**Describe the problem you have/What new integration you would like**
I have started using the timer based cover feature - Working great !!!

I have noticed that the position is kept (restored) upon restart and power failure.
This happens when esp8266_restore_from_flash is both TRUE and FALSE.

Does this mean that the position is kept on flash as well as in “RTC memory”?

**Please describe your use case for this integration and alternatives you've tried:**
If the position is actually saved in flash memory (reducing write count) the cover component also have a restore_mode attribute to decide what state/position (open/close) the cover should wake up in - if I don't want it to be constantly be written to flash.

theone11, 2020-04-18:
> > the cover component also have a restore_mode attribute to decide what state/position (open/close) the cover should wake up in
> 
> I can't find `restore_mode` in time based cover

Exactly

glmnet, 2020-04-27:
I just tested this, behavior is as coded:

state is preserved if `esp8266_restore_from_flash: true`

otherwise, it starts at 50% as [coded here](https://github.com/esphome/esphome/blob/dev/esphome/components/time_based/time_based_cover.cpp/#L21)

closed 2020-12-05 by stale[bot]