esphome/issues #1151: ESP32 more than one touch sensor crashes esphome

kmuret, 2020-04-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Proxmox
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

https://esphome.io/components/binary_sensor/esp32_touch.html

**Description of problem:**
I'm having the same issue as this guy: https://github.com/esphome/issues/issues/574
I was attempting to use all touch inputs, but realized any more than one caused esphome to crash every minute or so.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[13:51:15]Backtrace: 0x4008c8ac:0x3ffbe170 0x4008cadd:0x3ffbe190 0x400e8ebf:0x3ffbe1b0 0x400848fd:0x3ffbe1d0 0x40147e40:0x3ffbcff0 0x40147f25:0x3ffbd010 0x4008b566:0x3ffbd040 0x4008b5ff:0x3ffbd080 0x40088f99:0x3ffbd0b0
[13:51:15]
[13:51:15]Rebooting...
[13:51:15]ets Jun  8 2016 00:22:57
[13:51:15]
[13:51:15]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:51:15]configsip: 0, SPIWP:0xee
[13:51:15]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:51:15]mode:DIO, clock div:2
[13:51:15]load:0x3fff0018,len:4
[13:51:15]load:0x3fff001c,len:1044
[13:51:15]load:0x40078000,len:8896
[13:51:15]load:0x40080400,len:5828
[13:51:15]entry 0x400806ac
[13:51:15][I][logger:156]: Log initialized
[13:51:15][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[13:51:15][I][app:028]: Running through setup()...
[13:51:15][C][esp32_touch:012]: Setting up ESP32 Touch Hub...
[13:51:15][C][wifi:033]: Setting up WiFi...
[13:51:15][D][wifi:304]: Starting scan...
[13:51:15][D][binary_sensor:036]: 'ESP32 Touch Pad GPIO4': Sending state OFF
[13:51:15][D][binary_sensor:036]: 'ESP32 Touch Pad GPIO15': Sending state OFF
[13:51:15][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1571
[13:51:15][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1472
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1570
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1570
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1569
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1569
[13:51:16][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:17][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1570
[13:51:17][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:17][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1569
[13:51:17][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1469
[13:51:17][D][wifi:319]: Found networks:
[13:51:17][I][wifi:365]: - '533' (74:83:C2:03:30:60) ▂▄▆█
[13:51:17][D][wifi:366]:     Channel: 11
[13:51:17][D][wifi:367]:     RSSI: -51 dB
[13:51:17][I][wifi:365]: - '533' (74:83:C2:BA:35:0F) ▂▄▆█
[13:51:17][D][wifi:366]:     Channel: 6
[13:51:17][D][wifi:367]:     RSSI: -60 dB
[13:51:17][D][wifi:369]: - 'IoT' (86:83:C2:03:30:60) ▂▄▆█
[13:51:17][D][wifi:369]: - '' (96:83:C2:03:30:60) ▂▄▆█
[13:51:17][D][wifi:369]: - 'NoT' (76:83:C2:03:30:60) ▂▄▆█
[13:51:17][D][wifi:369]: - 'NoT' (76:83:C2:BA:35:0F) ▂▄▆█
[... 101 lines cut ...]
[13:51:22][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1608
[13:51:22][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1484
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1608
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1483
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1607
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1484
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1607
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1484
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO4' (T0): 1606
[13:51:23][D][esp32_touch:125]: Touch Pad 'ESP32 Touch Pad GPIO15' (T3): 1484
[13:51:23]E (8476) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[13:51:23]E (8476) task_wdt:  - IDLE0 (CPU 0)
[13:51:23]E (8476) task_wdt: Tasks currently running:
[13:51:23]E (8476) task_wdt: CPU 0: Tmr Svc
[13:51:23]E (8476) task_wdt: CPU 1: IDLE1
[13:51:23]E (8476) task_wdt: Aborting.
[13:51:23]abort() was called at PC 0x400e8ebf on core 0
[13:51:23]
[13:51:23]Backtrace: 0x4008c8ac:0x3ffbe170 0x4008cadd:0x3ffbe190 0x400e8ebf:0x3ffbe1b0 0x400848fd:0x3ffbe1d0 0x40147e3d:0x3ffbcff0 0x40147f25:0x3ffbd010 0x4008b566:0x3ffbd040 0x4008b5ff:0x3ffbd080 0x40088f99:0x3ffbd0b0
[13:51:23]
[13:51:23]Rebooting...
[13:51:23]ets Jun  8 2016 00:22:57
```

**Additional information and things you've tried:**

fake-name, 2020-04-23:
This is the watchdog timing out, some operation was blocking the RTOS for long enough that the watchdog timer resets the CPU.

I suspect there's a blocking operation in the touch sensors. There's certainly a few of them elsewhere (the UART interface can also cause watchdog timeouts).

glmnet, 2020-04-26:
I've been trying to fix this, the sensors are iterated in the loop inside the component, so I've added `App.feed_wdt()` and even other feed mechanisms and none seems to fix it.

I disabled the task watchdog in the setup and then it works ok, of course this is not a solution, I did this mostly to know if the thing was hanging.

It is only happening wen the `iir_filter` is used, so a workaround right now is disabling it.

glmnet, 2020-04-27:
I've run [this sample](https://github.com/espressif/esp-idf/blob/741960d5ce0394e8ecb99deef8a5ebf579cb45a2/examples/peripherals/touch_pad_read/main/esp32/tp_read_main.c) and it works ok, so I really don't know what is going on here.

glmnet, 2020-04-27:
btw to disable watchdog I did this at setup:

````
esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(0));
````

fake-name, 2020-04-27:
> I've been trying to fix this, the sensors are iterated in the loop inside the component, so I've added App.feed_wdt() and even other feed mechanisms and none seems to fix it.

I would assume its blocking in one of the IDF functions, then.

Is it possible esphome has a shorter WTD timeout then that demo?

glmnet, 2020-04-27:
I've tried increasing the timeout, not 100% sure my settings where taking effect, anyway these T WDT are measured in seconds, what I did was flip an output every time the the loop is called and every time the T WDT feed is called for the esp 32, they show the same square wave in the scope at 30hz (loop runs at 60hz)

so the `yield()` is called normally

I'm leaning my intuition to some race condition / priority starvation issue... there are some docs saying the iir_filter uses some timer stuff.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/touch_pad.html#_CPPv422touch_pad_filter_start8uint32_t

donkawechico, 2020-05-24:
FWIW, I'm having this problem too. I just implemented the "[$5 Smart Bed](https://medium.com/the-smarter-home/smart-bed-7de9ad55276e)" project at home today and everything works perfectly *until* I add in the `iir_filter: 10ms` option. Then I start seeing all the same issues described by @kmuret .

donkawechico, 2020-11-21:
Bumping the stalebot as this is still an issue.

adityapattiyeri, 2021-03-17:
Facing the same issue....works perfectly without `iir_filter: 10ms`

closed 2021-07-29 by stale[bot]