esphome/issues #1152: BME680 - Incorrect GAS Values

madchuska, 2020-04-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

homeassistant  - HassOS 3.12
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
**Affected component:**

https://esphome.io/components/sensor/bme680.html

**Description of problem:**
Gas sensor readings never update and are constantly at 48633.00000 this did not update for over 8 hours

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
`esphome:
  name: bme_test
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "my_ssid"
  password: "my_pw"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bme Test Fallback Hotspot"
    password: "xL5wbPSBnUYK"

captive_portal:

# Enable logging
logger:
 level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

i2c:
  sda: 5
  scl: 4
  scan: True
  id: bus_a
    
sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
      oversampling: 16x
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    gas_resistance:
      name: "BME680 Gas Resistance"
    address: 0x77
    update_interval: 60s`
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
`INFO Reading configuration /config/esphome/bme_test.yaml...
INFO Starting log output from bme_test.local using esphome API
INFO Connecting to bme_test.local:6053 (192.168.1.135)
INFO Successfully connected to bme_test.local
[... 368 lines cut ...]
[23:51:10]}
[23:51:10][V][sensor:013]: 'BME680 Humidity': Received new state 58.184673
[23:51:10][D][sensor:092]: 'BME680 Humidity': Sending state 58.18467 % with 1 decimals of accuracy
[23:51:10][VV][api.service:120]: send_sensor_state_response: SensorStateResponse {
[23:51:10]  key: 1277562889
[23:51:10]  state: 58.1847
[23:51:10]  missing_state: NO
[23:51:10]}
[23:51:10][V][sensor:013]: 'BME680 Gas Resistance': Received new state 48633.000000
[23:51:10][D][sensor:092]: 'BME680 Gas Resistance': Sending state 48633.00000 Ω with 1 decimals of accuracy
[23:51:10][VV][api.service:120]: send_sensor_state_response: SensorStateResponse {
[23:51:10]  key: 426116527
[23:51:10]  state: 48633
[23:51:10]  missing_state: NO
[23:51:10]}
[23:51:10][V][app:076]: A component took a long time in a loop() cycle (0.23 s).
[23:51:10][V][app:077]: Components should block for at most 20-30ms in loop().
[23:51:13][VV][api.service:250]: on_ping_request: PingRequest {}
[23:51:13][VV][api.service:035]: send_ping_response: PingResponse {}`
```

**Additional information and things you've tried:**

I have tried to set the heater temp, and also duration - neither of which helped.
I tested the same BME680 with a tasmota device and the sensor updated as expected, I then flashed the tasmota device with ESP Home (another D1 Mini) and end up getting the same result as the original D1 Mini (same value and no updates).

madchuska, 2020-04-22:
another log rom today as it is behaving slightly differently

`INFO Reading configuration /config/esphome/bme_test.yaml...
INFO Starting log output from bme_test.local using esphome API
INFO Connecting to bme_test.local:6053 (192.168.1.135)
INFO Successfully connected to bme_test.local
[23:37:20][I][app:100]: ESPHome version 1.14.3 compiled on Apr 20 2020, 22:38:16
[23:37:20][C][wifi:415]: WiFi:
[23:37:20][C][wifi:283]:   SSID: [redacted]
[23:37:20][C][wifi:284]:   IP Address: 192.168.1.135
[23:37:20][C][wifi:286]:   BSSID: [redacted]
[23:37:20][C][wifi:287]:   Hostname: 'bme_test'
[23:37:20][C][wifi:291]:   Signal strength: -51 dB ▂▄▆█
[23:37:20][V][wifi:293]:   Priority: -1.0
[23:37:20][C][wifi:295]:   Channel: 1
[23:37:20][C][wifi:296]:   Subnet: 255.255.255.0
[23:37:20][C][wifi:297]:   Gateway: 192.168.1.1
[23:37:20][C][wifi:298]:   DNS1: 192.168.1.1
[23:37:20][C][wifi:299]:   DNS2: (IP unset)
[23:37:20][C][i2c:028]: I2C Bus:
[23:37:20][C][i2c:029]:   SDA Pin: GPIO5
[23:37:20][C][i2c:030]:   SCL Pin: GPIO4
[23:37:20][C][i2c:031]:   Frequency: 50000 Hz
[23:37:20][I][i2c:033]: Scanning i2c bus for active devices...
[23:37:20][I][i2c:049]: Found no i2c devices!
[23:37:20][C][logger:175]: Logger:
[23:37:20][C][logger:176]:   Level: VERBOSE
[23:37:20][C][logger:177]:   Log Baud Rate: 115200
[23:37:20][C][logger:178]:   Hardware UART: UART0
[23:37:20][C][bme680.sensor:208]: BME680:
[23:37:20][C][bme680.sensor:209]:   Address: 0x77
[23:37:20][C][bme680.sensor:213]:   IIR Filter: OFF
[23:37:20][C][bme680.sensor:214]:   Update Interval: 60.0s
[23:37:20][C][bme680.sensor:216]:   Temperature 'BME680 Temperature'
[23:37:20][C][bme680.sensor:216]:     Unit of Measurement: '°C'
[23:37:20][C][bme680.sensor:216]:     Accuracy Decimals: 1
[23:37:20][C][bme680.sensor:216]:     Icon: 'mdi:thermometer'
[23:37:20][C][bme680.sensor:217]:     Oversampling: 16x
[23:37:20][C][bme680.sensor:218]:   Pressure 'BME680 Pressure'
[23:37:20][C][bme680.sensor:218]:     Unit of Measurement: 'hPa'
[23:37:20][C][bme680.sensor:218]:     Accuracy Decimals: 1
[23:37:20][C][bme680.sensor:218]:     Icon: 'mdi:gauge'
[23:37:20][C][bme680.sensor:219]:     Oversampling: 16x
[23:37:20][C][bme680.sensor:220]:   Humidity 'BME680 Humidity'
[23:37:20][C][bme680.sensor:220]:     Unit of Measurement: '%'
[23:37:20](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[23:37:20](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[23:37:20][C][bme680.sensor:221]:     Oversampling: 16x
[23:37:20][C][bme680.sensor:222]:   Gas Resistance 'BME680 Gas Resistance'
[23:37:20][C][bme680.sensor:222]:     Unit of Measurement: 'Ω'
[23:37:20][C][bme680.sensor:222]:     Accuracy Decimals: 1
[23:37:20][C][bme680.sensor:222]:     Icon: 'mdi:gas-cylinder'
[23:37:20][C][bme680.sensor:226]:   Heater temperature=320°C duration=150ms
[23:37:20][C][captive_portal:169]: Captive Portal:
[23:37:20][C][ota:029]: Over-The-Air Updates:
[23:37:20][C][ota:030]:   Address: bme_test.local:8266
[23:37:20][C][api:095]: API Server:
[23:37:20][C][api:096]:   Address: bme_test.local:6053
[23:37:33][W][i2c:070]: Received NACK on transmit of address 0x77
[23:37:55][V][api.connection:672]: Error: Disconnecting Home Assistant 0.108.6 (192.168.1.124)
[23:37:55][D][api:067]: Disconnecting Home Assistant 0.108.6 (192.168.1.124)
[23:37:57][V][api.connection:567]: Hello from client: 'Home Assistant 0.108.6 (192.168.1.124)'
[23:37:57][D][api.connection:583]: Client 'Home Assistant 0.108.6 (192.168.1.124)' connected successfully!
[23:38:17][V][api.connection:672]: Error: Disconnecting ESPHome v1.14.3 (192.168.1.124)
ERROR Error while reading incoming messages: Error while receiving data: [Errno 9] Bad file descriptor
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to bme_test.local:6053 (192.168.1.135)
INFO Successfully connected to bme_test.local
[23:38:33][W][i2c:070]: Received NACK on transmit of address 0x77
STOP`

h0jeZvgoxFepBQ2C, 2020-07-25:
I also have an bme680 and very often it jumps between 4000 and 4500 directly (which is not related to real gas changes)..

closed 2020-09-04 by stale[bot]

sethdeckard, 2020-09-06:
In my experience the BME680 is pretty much useless for VOCs without using Bosch's proprietary BSEC library. I have a few of these working with BSEC and the results are great, I'm now in the process of moving them over to ESPHome now just to keep everything as config-driven as possible. Looks like someone is doing the same: https://github.com/esphome/issues/issues/1209 (I had to add similar build flags when using the Arduino IDE to flash the ESP32)

h0jeZvgoxFepBQ2C, 2020-09-07:
Could you post somewhere how to do this? I'm trying to do the same, but can't figure it out :(

sethdeckard, 2020-09-08:
@h0jeZvgoxFepBQ2C It seems like the user in that linked issue got it working (https://github.com/esphome/issues/issues/1209). I was about to start on this myself and was just looking through issues to see if someone else attempted it and found these. I'll find out for myself pretty soon but in theory that custom sensor approach should work, the linker issues do not seem to be specific to ESPHome at all.

h0jeZvgoxFepBQ2C, 2020-09-08:
I tried to adapt my platform.io file, but it didn't change anything :(

```
[platformio]

[env:nodemcuv2]
build_flags = -I .piolibdeps/BSEC-Arduino-library/src/inc  -L .piolibdeps/BSEC-Arduino-library/src/esp8266  -lalgobsec

lib_deps = 
  BSEC Software Library
```

results in following:

```
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/bme680/lib20d/libBSEC Software Library.a(bsec.cpp.o):(.text._ZN4Bsec11beginCommonEv+0x4): undefined reference to `bsec_init'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/bme680/lib20d/libBSEC Software Library.a(bsec.cpp.o):(.text._ZN4Bsec11beginCommonEv+0x8): undefined reference to `bsec_get_version'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/bme680/lib20d/libBSEC Software Library.a(bsec.cpp.o): in function `Bsec::beginCommon()':
bsec.cpp:(.text._ZN4Bsec11beginCommonEv+0x8b): undefined reference to `bsec_init'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: bsec.cpp:(.text._ZN4Bsec11beginCommonEv+0x95): undefined reference to `bsec_get_version'
collect2: error: ld returned 1 exit status
*** [.pioenvs/bme680/firmware.elf] Error 1
```