esphome/issues #1156: Property "name" in uptime sensor and wifi_signal sensor not working in ESP32

melwinek, 2020-04-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini and Wemos D1 ESP32 Mini

**Affected component:**

Wifi sensor and uptime sensor

**Description of problem:**
Property "name" in uptime sensor and wifi_signal sensor not working in ESP32

**Problem-relevant YAML-configuration entries:**
```
ESP 8266
--------------------
esphome:
  name: dmalypokoj
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "yyy"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "zzz"

ota:
  password: "zzz"

sensor:
  - platform: uptime
    name: "Dol MPokoj Uptime"
    filters:
      - lambda: return x / 86400;
    unit_of_measurement: "days"
    accuracy_decimals: 2

  - platform: wifi_signal
    name: "Dol Mpokoj RSSI"
	
	
	
ESP 32
--------------------
esphome:
  name: kotlownia
  platform: ESP32
  board: wemos_d1_mini32

wifi:
  ssid: "xxx"
  password: "yyy"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "zzz"

ota:
  password: "zzz"
  
  
sensor:
  - platform: uptime
    name: "Kotlownia uptime"
    filters:
      - lambda: return x / 86400;
    unit_of_measurement: "days"
    accuracy_decimals: 2

  - platform: wifi_signal
    name: "Kotlownia RSSI"
    update_interval: 60s

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

randybb, 2020-04-20:
Never had any problem with Entity ID. Are you 100 % sure have it flashed with configured name?

melwinek, 2020-04-20:
I'm sure. I flashed as in the attached sources.

randybb, 2020-04-20:
Ok, then the problem is in HA how it handles entities.
The easiest way is just to remove your integration  `HA > Configuration > Integrations`, click on _kotlownia_, remove in (bin icon in Top Right corner) and then associate with HA again.
It looks like that renaming/deleting entities doesn't work...

closed 2020-09-01 by stale[bot]