esphome/issues #1157: SGP30 Baseline readings are misplaced

keniji, 2020-04-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266

**Affected component:**
SGP30

**Description of problem:**
According to [Issue 950](https://github.com/esphome/issues/issues/950), baseline readings are separated in an update: [SGP30](https://github.com/esphome/esphome/blob/dev/esphome/components/sgp30/sensor.py), and I am now using this update. However, after I set the baseline readings and get the new readings after 1 hour, I found the readings are "exchanged". For example, if I set:
````
eco2_baseline: 0xAAAA
tvoc_baseline: 0xBBBB
````
After 1 hour I will see in log:
`Current eCO2 baseline: 0xBBBB, TVOC baseline: 0xAAAA

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: sgp30
    baseline:
      eco2_baseline: 0x8BD9
      tvoc_baseline: 0x94F9
    eco2:
      name: aq_shb_bed_eco2
      id: sgp30_eco2
      accuracy_decimals: 1
    tvoc:
      id: sgp30_tvoc
      accuracy_decimals: 1
    address: 0x58
    update_interval: 30s
    compensation:
      temperature_source: aht10_temperature
      humidity_source: aht10_humidity
```

**Logs (if applicable):**
```
[03:39:42][D][sgp30:131]: External compensation data received: Humidity 53.75%
[03:39:42][D][sgp30:141]: External compensation data received: Temperature 26.14°C
[03:39:42][D][sgp30:150]: Calculated Absolute humidity: 13.159 g/m³ (0x0D28)
[03:39:42][I][sgp30:112]: Current eCO2 baseline: 0x94F9, TVOC baseline: 0x8BD9
```

**Additional information and things you've tried:**
In the environment above, I found the eCO2 baseline is getting smaller and TVOC's is larger, so I think the correct readings should be 0x8BD9 for eCO2 and 0x94F9 for TVOC?

keniji, 2020-04-21:
It's quite strange, I have now set:
```
eco2_baseline: 0x94F9
tvoc_baseline: 0x8BD9
```
Then the log stays on:
`[05:40:06][I][sgp30:112]: Current eCO2 baseline: 0x94F9, TVOC baseline: 0x8BD9`
And the readings are not changed to smaller or larger as just now... But I am sure when I got the calibrated readings, they are:
```Current eCO2 baseline: 0x8BD9, TVOC baseline: 0x94F9```
I will make another test environment and try to see if I can reproduce this issue. Will update later.

babblerz, 2020-04-29:
According to the datasheet the order for getting the baseline is first eco2 then the tvoc. But to set it the order is first tvoc and then the eco2. 
While getting does seem implemented correctly, setting does not.
In esphome/components/sgp30/sgp30.cpp the code should be changed
from this:
```
void SGP30Component::write_iaq_baseline_(uint16_t eco2_baseline, uint16_t tvoc_baseline) {
  uint8_t data[7];
  data[0] = SGP30_CMD_SET_IAQ_BASELINE & 0xFF;
  data[1] = eco2_baseline >> 8;
  data[2] = eco2_baseline & 0xFF;
  data[3] = sht_crc_(data[1], data[2]);
  data[4] = tvoc_baseline >> 8;
  data[5] = tvoc_baseline & 0xFF;
  data[6] = sht_crc_(data[4], data[5]);
```

to this:
```
void SGP30Component::write_iaq_baseline_(uint16_t eco2_baseline, uint16_t tvoc_baseline) {
  uint8_t data[7];
  data[0] = SGP30_CMD_SET_IAQ_BASELINE & 0xFF;
  data[1] = tvoc_baseline >> 8;
  data[2] = tvoc_baseline & 0xFF;
  data[3] = sht_crc_(data[1], data[2]);
  data[4] = eco2_baseline >> 8;
  data[5] = eco2_baseline & 0xFF;
  data[6] = sht_crc_(data[4], data[5]);
```

keniji, 2020-04-30:
Hi babblerz,
Thanks for the reply.

I just have a test environment set and will see if it works fine.

keniji, 2020-04-30:
Hi @babblerz ,
I have a quick check and seems the component should works fine now, thanks again for your help!

closed 2020-04-30 by keniji

babblerz, 2020-05-04:
Hi @keniji,

Have you made the suggested changes and tested that, or you tested the original code?
The fix I have suggested is not incorporated in the sources yet, it was just a comment.