esphome/issues #1159: WaveShare e-paper 2.7in display does not work

ukewea, 2020-04-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.15.0-dev

**ESP (ESP32/ESP8266, Board/Sonoff):**

[E-Paper ESP8266 Driver Board (by Waveshare)](https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board)

**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html (`2.70in` to be specific)

**Description of problem:**
The display is filled with random pixels, not what I put into the DisplayBuffer.

**Problem-relevant YAML-configuration entries:**
```yaml
font:
  - file: "Arial.ttf"
    id: my_font
    size: 16

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  # I defined the MISO pin (though not connected) so hardware SPI will be used
  miso_pin: GPIO12

display:
  - platform: waveshare_epaper
    update_interval: 10s
    cs_pin: GPIO15
    dc_pin: GPIO4
    busy_pin: GPIO5
    reset_pin: GPIO2
    model: 2.70in
    lambda: |-
    it.printf(20, 120, my_font, "Hello world %d!", millis());
```

**Logs (if applicable):**

```
```

**Additional information and things you've tried:**
I've noticed that on the wiki page the 2.70in model is marked with `(not tested)`, so I downloaded the demo code from WaveShare and tried to figure out how they send buffer to the display. With some modification in the `waveshare_epaper.cpp` generated by ESPHome I actually made the display work properly, but I am not sure if my yaml configuration is done right.

Thanks.

PaulAntonDeen, 2020-04-24:
I have the same problem. What did you modify to get it to work?

ukewea, 2020-04-24:
Hi, 

you should be able to find a file named `waveshare_epaper.cpp` in your disk (but not the one copied to your project directory), replace `WaveshareEPaper2P7In::display()` member function with the following:

```cpp
void HOT WaveshareEPaper2P7In::display() {
  uint32_t bufLen = this->get_buffer_length_();

  // COMMAND DATA START TRANSMISSION 1
  this->command(0x10);
  delay(2);
  for (uint32_t i = 0; i < bufLen; i++) {
    this->data(this->buffer_[i]);
  }
  delay(2);

  // COMMAND DATA START TRANSMISSION 2
  this->command(0x13);
  delay(2);
  for (uint32_t i = 0; i < bufLen; i++) {
    this->data(this->buffer_[i]);
  }

  // COMMAND DISPLAY REFRESH
  this->command(0x12);
}
```

What I modified is calling `this->data()` when sending every single byte.

I'm thinking of making a pull request for this modification, your feedback will help testing the new code, please report your test result, thanks.

ukewea, 2020-04-24:
By the way, Since 2.70in model can only do a FULL REFRESH when drawing the display, the flash while clearing the current content is very annoying, I will buy other models which is able to do a partial refreshing though...

ukewea, 2020-04-24:
By the way, you have to run `esphome you_project.yaml compile` after the modification so the new code can be compiled.

Awaiting your feedback 🤓

PaulAntonDeen, 2020-04-24:
So this is the original code that I found in /opt/esphome/esphome/components/waveshare_epaper/waveshare_epaper.cpp
Using ESPhome in Hassio using Portainer to access the shell in the container.

`
void HOT WaveshareEPaper2P7In::display() {
  // COMMAND DATA START TRANSMISSION 1
  this->command(0x10);
  delay(2);
  this->start_data_();
  this->write_array(this->buffer_, this->get_buffer_length_());
  this->end_data_();
  delay(2);

  // COMMAND DATA START TRANSMISSION 2
  this->command(0x13);
  delay(2);
  this->start_data_();
  this->write_array(this->buffer_, this->get_buffer_length_());
  this->end_data_();

  // COMMAND DISPLAY REFRESH
  this->command(0x12);
}
`
your modifications look a bit different and I don't understand enough of whats going on to know how to adapt this.

PaulAntonDeen, 2020-04-24:
AH I just overwrote the whole section with your code and now the display is working! Your fix is confirmed :)

ukewea, 2020-04-25:
@PaulAntonDeen glad to know that, we can then remove the `(not tested)` mark on Wiki because at least two people have tested it 😎

PaulAntonDeen, 2020-04-25:
Thank you!
I hope your commit makes it into the next version of ESPhome!

closed 2020-04-25 by glmnet