esphome/issues #1161: set value for template Sensor

Heiner1967, 2020-04-23:
the following sample code to set a value of a template sensor is givenas instruction and works fine:

`  - platform: template
    name: templatesensor_x
    id: x
    lambda: |-
      return 5.6;
    update_interval: 60s` 

If however I like now to write the 5.6 by way of a variable into the template Sensor
I would replace 

> return 5.6;
with
> return id(ch1s);
which is the name of a variable and defined here:
`globals:
  - id: ch1s
    type: float
    restore_value: false
    initial_value: "5.6"`

The code compiles but the template sensor does not have any value.
What am I doing wrong?
( At the end i even like to do some math to calculate a value, but I only get the code for a variable to work, but not transfered into the template Sensor
> 

DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
nodemcu-32s

**Affected component:**
https://esphome.io/components/sensor/template.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: watersystem
  platform: ESP32
  board: nodemcu-32s 

globals:
  - id: ch1s
    type: float
    restore_value: false
    initial_value: "5.6"

wifi:
  ssid: "xx"
  password: "xx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:

  - platform: template
    name: x
    id: x
    lambda: |-
      return id(ch1s);
    update_interval: 60s

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-04-26:
Your code looks good and works for me

Heiner1967, 2020-04-27:
for me x is reported as 0, however I found the solution with a friend (Thanks Cips):

`    lambda: |-
      return float(id(ch1s)); `

Is working

closed 2020-04-27 by Heiner1967