esphome/issues #1164: transmit_raw not transmitting

NeoMatrixJR, 2020-04-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Essentially same as this I believe: https://templates.blakadder.com/ytf_ir_bridge.html

**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
transmit_raw is not transmitting.
having same issue as this: https://github.com/esphome/issues/issues/1082 - according to closing comment I'm not sure this was fixed.  Please see additional comments below.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: restart
    id: naxa_reset
    internal: true
  - platform: template
    name: Vizio TV Power
    id: vizio_tv_power
    optimistic: true
    restore_state: true
    turn_on_action:
     - remote_transmitter.transmit_nec:
         address: 0x20DF
         command: 0x10EF
    turn_off_action:
     - remote_transmitter.transmit_raw:
         code: [4088, -1542, 1019, -510, 513, -1019, 510, -509, 511, -510, 1020,
                -1020, 1022, -1019, 510, -509, 511, -510, 511, -509, 511, -510,
                1020, -1019, 510, -511, 1020, -510, 512, -508, 510, -1020, 1022]
         carrier_frequency: 38

```

**Logs (if applicable):**

```
Relevant log portion:
[18:27:01][D][switch:021]: 'Vizio TV Power' Turning ON.
[18:27:01][D][remote_transmitter:066]: Sending remote code...
[18:27:01][D][switch:045]: 'Vizio TV Power': Sending state ON
[18:27:01][D][remote.raw:041]: Received Raw: 214
[18:27:03][D][switch:025]: 'Vizio TV Power' Turning OFF.
[18:27:03][D][remote_transmitter:066]: Sending remote code...
[18:27:03][D][switch:045]: 'Vizio TV Power': Sending state OFF
[18:27:05][D][switch:021]: 'Vizio TV Power' Turning ON.
[18:27:05][D][remote_transmitter:066]: Sending remote code...
[18:27:05][D][switch:045]: 'Vizio TV Power': Sending state ON
[18:27:05][D][remote.raw:041]: Received Raw: 214
[18:27:06][D][switch:025]: 'Vizio TV Power' Turning OFF.
[18:27:06][D][remote_transmitter:066]: Sending remote code...
[18:27:06][D][switch:045]: 'Vizio TV Power': Sending state OFF
```

**Additional information and things you've tried:**
having same issue as this: https://github.com/esphome/issues/issues/1082
He stated it's resolved, but it looks by the code he posted at the end that he'd switched to transmit_nec, so I think raw may still have an issue.

My switch yaml is included above. (note, yes, I know I have an NEC for the on, and raw for the off.  this is a test for another switch I have that I ONLY have RAW for that's also not working...this is an example contrast of working/not working.)

I'm having trouble getting good codes from dump: all, so I'm using an arduino with an IR receiver, and the following code I found on IRDB website: https://gist.github.com/probonopd/5793692#file-sendandreceive-ino

using the transmit_nec produces output on the arduino.  Using transmit_raw does not.

urbatecte, 2020-04-26:
Quite same here.
When record and resend ir with Arduino lib [https://github.com/z3t0/Arduino-IRremote](url) the IR transmit lead to the same received message as guenine RCswitch remote and is perfectly decoded by ESP Home.
When whatever the transmit protocol used RCswitch, raw, brand... nothing is decoded neither by ESP home nor by Arduino.
Transmit IR LED is working (I got an builtin LED) and check it as well with my phone camera wich detect IR (very handy ;o)

glmnet, 2020-04-26:
> ```yaml
> ```yaml
> carrier_frequency: 38
> ```
> ```

this is 38 hz, you might want 38kHz here

closed 2020-04-27 by glmnet

urbatecte, 2020-04-27:
Hi,

Solve the issue for transmit RAW, thanks a lot !
1/ Any chance to apply the carrier frequancy to other protocol ?
2/ If the transmit work great, it's not received by ESPHome (when emitting by itelf) whereas if I send the command by the guenuine remote, reception is ok (same data) by both ESPHome and Arduino

glmnet, 2020-04-27:
1 what? Should be supported check docs 

2 yes. Esphome can not attend other stuff while transmitting so he basically miss its own transmission