esphome/issues #1165: esp01s keep reconnecting to HA API

SnowViet, 2020-04-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Working with HA 0.107.7
The ESP is connected to a UniFi nanoHD, at about 2 meters.
esphome version is 1.14.3
pip3 version 9.0.1
python 3.6
On Ubuntu VM

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP01S

**Affected component:**

https://esphome.io/components/remote_receiver.html

**Description of problem:**
The ESP keep reconnecting to HA (see logs).

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp01sir
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SSID"
  password: "PASSWORD"
  manual_ip:
    static_ip: 192.168.XXX.XXX
    gateway: 192.168.XXX.XXX
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "PASSWORD"

ota:
  password: "PASSWORD"

sensor:
  - platform: wifi_signal
    name: "WiFi Signal ESP_IR"
    update_interval: 60s
  - platform: uptime
    unit_of_measurement: "days"
    filters:
      - lambda: return x / 86400;
    name: "Uptime ESP_IR"

remote_receiver:
  pin:
    number: GPIO2
    inverted: True
  dump: raw
  tolerance: 35

binary_sensor:
  - platform: status
    name: "Status ESP_IR"

  - platform: remote_receiver
    name: "XBOX remote - Main"
    raw:
      code: [2682, -881, 460, -443, 447, -445, 444, -879, 457, -878, 1348, -879, 458, -443, 446, -445, 445, -445, 471, -398, 467, -416, 474, -445, 469, -402, 465, -444, 445, -446, 445, -446, 890, -445, 444, -445, 447, -396, 518, -853, 904, -444, 445, -445, 446, -879, 903, -874, 460, -400, 514, -402, 911, -442, 446, -881, 455, -444, 891, -877, 459, -444, 446]
  - platform: remote_receiver
    name: "XBOX remote - Main 2"
    raw:
[... 369 lines cut ...]
[11:44:43][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:03][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:03][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:03][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:03][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:04][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:04][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:04][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:04][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:04][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:04][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:24][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:24][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:25][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:25][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:25][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:25][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!
[11:45:25][D][api:067]: Disconnecting Home Assistant 0.107.7 (192.168.XXX.XXX)
[11:45:25][D][api.connection:583]: Client 'Home Assistant 0.107.7 (192.168.XXX.XXX)' connected successfully!

```

**Additional information and things you've tried:**
It was working smoothly before I updated the ESP (changed the SSID), I think esphome updated, and now it's not working anymore.
If I remove few of the "remote_receiver", like if I keep only 5 of it, it's seems to work.

glmnet, 2020-04-24:
Wowww those are a lot of buttons!! 

Try the fix about sensor reconnecting #877

SnowViet, 2020-04-29:
I tried, but it didn't work. So I downgraded to 1.13.6 and it's working since few days.

closed 2020-04-29 by SnowViet