esphome/issues #1166: Unexpected error: Error changing ESP upload baud rate: Timed out waiting for packet header

sander816, 2020-04-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Esp01 and esp01s

**Description of problem:**

when i try to flash esphome on a esp01 i get this:
it used to work fine, no idea if this is from the last update:

Using 'COM6' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting...

Chip Info:

Chip Family: ESP8266
Chip Model: ESP8266EX
Chip ID: 00157A04
MAC Address: CC:50:E3:15:7A:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Unexpected error: Error changing ESP upload baud rate: Timed out waiting for packet header

when trying another flashtool:
esptool v0.4.12 - (c) 2014 Ch. Klippel ck@atelier-klippel.de
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from to COM7
setting address from 0x00000000 to 0x00000000
espcomm_upload_file
espcomm_upload_mem
setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
read 0, requested 1
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
Uploading 857776 bytes from .\ESP_Easy_mega-20200222_normal_ESP8266_4M1M.bin to flash at 0x00000000
erasing flash
size: 0d16b0 address: 000000
first_sector_index: 0
total_sector_count: 210
head_sector_count: 16
adjusted_sector_count: 194
erase_size: 0c2000
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
setting serial port timeouts to 15000 ms

```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->

**Additional information and things you've tried:**

I tried flsahing with esphome flsher and espeasy flasher

Are these esp01 dead?

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

glmnet, 2020-04-24:
Sometimes they don’t enter in program mode properly. How is it wired? Are you pulling gpio0 down?

sander816, 2020-04-25:
Yes, at bootup I connect it to ground

closed 2020-09-06 by stale[bot]