esphome/issues #1167: sonoff l1 ledstrip

Wriman, 2020-04-25:
i have sonoff l1 ledstrip, and i flash it for esphome
it is connect fine and add fine in homeassistant.
when i do something with ledstrip in homeassistant nothing happen my led

my config:
esphome:
name: corridor_led_light
platform: ESP8266
board: esp01_1m

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password

manual_ip:
static_ip: 192.168.1.52
gateway: 192.168.1.1
subnet: 255.255.255.0
dns1: 192.168.1.1
dns2: 8.8.8.8

ap:
ssid: "corridor_led_light_hotspot"
password: !secret esphome_fbpw

captive_portal:

logger:
level: DEBUG

api:
password: !secret esphome_pw
ota:
password: !secret esphome_pw

light:

    platform: rgbww
    name: "corridor_led_light"
    red: red_component
    green: green_component
    blue: blue_component
    cold_white: cold_white_component
    warm_white: warm_white_component
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

output:

    platform: esp8266_pwm
    id: red_component
    pin: GPIO5
    max_power: 80%
    platform: esp8266_pwm
    id: green_component
    pin: GPIO4
    max_power: 80%
    platform: esp8266_pwm
    id: blue_component
    pin: GPIO15
    max_power: 80%
    platform: esp8266_pwm
    id: cold_white_component
    pin: GPIO12
    max_power: 80%
    platform: esp8266_pwm
    id: warm_white_component
    pin: GPIO14
    max_power: 80%

in this node logs write when i change something in homeassistant, but the led is doesnt change.

and when i controlled this strip with ir remote, the esphome log doesnt write any

emorydunn, 2020-08-08:
I just bought one of these and after a few days of research and testing I have a basic implementation for the Sonoff L1. At the moment you can set the color and brightness from Home Assistant, but any changes made from the remote aren't pushed back. 

https://gist.github.com/emorydunn/db410db8bf68c8a335f3362d69624aaa

adamdobson, 2020-08-26:
> I just bought one of these and after a few days of research and testing I have a basic implementation for the Sonoff L1. At the moment you can set the color and brightness from Home Assistant, but [...]

Thanks @emorydunn I see it's a seperate chip on a serial connection thats doing the pwm. Your sollution works great for my needs.

closed 2021-01-07 by stale[bot]