esphome/issues #1170: NodeMCU ESP32 waking up immediately from deep sleep

gaggio, 2020-04-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (from AZ delivery, ESP-WROOM-32)

**Affected component:**

https://esphome.io/components/deep_sleep.html

**Description of problem:**

Expected behaviour is ESP entering deep sleep and waking up when the pin is touched, or after 10min of deep sleep. What is instead observed is that about 10-15 s after entering deep sleep, the ESP reconnects to wifi and home assistant, becoming available again, even if the pin is not touched.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: keypad_esp32
  platform: ESP32
  board: nodemcu-32s

esp32_touch:

spi:
  clk_pin: GPIO19
  miso_pin: GPIO25
  mosi_pin: GPIO5

pn532:
  cs_pin: GPIO17
  update_interval: 1s

binary_sensor:
  - platform: pn532
    uid: XX-XX-XX-XX
    name: "Tag Federico"
  - platform: pn532
    uid: YY-YY-YY-YY
    name: "Tag Erica"
  - platform: esp32_touch
    name: "keypad_esp32 Touch Pad GPIO32"
    pin: GPIO32
    threshold: 500
  - platform: gpio
    pin: 
      number: GPIO32
      mode: INPUT_PULLUP
      inverted: True
    name: "Wake-up pin"

deep_sleep:
  run_duration: 3min
  sleep_duration: 10min
  wakeup_pin: GPIO32

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Note that the touch on pin 32 is correctly reported, and threshold seems appropriate.
Also note that pn532 is correctly reporting tags. The module also seems to be powered during deep sleep, as its led stays on. This is ok for me because the final objective would be to keep the ESP in deep sleep until a tag is read. Alternatively, should this be not possible, a touch to wake the ESP up before proposing a tag to read would be ok.

Tried experimenting with `INPUT_PULLUP`, `INPUT_PULLDOWN`,`INVERT_WAKEUP`, `inverted: True`, with no success.

gaggio, 2020-05-05:
I've been experimenting more, in particular adding a sensor to report the reason for waking up:

```
esphome:
  name: keypad_esp32
  platform: ESP32
  board: nodemcu-32s
# test wakeup reason
  on_boot:
    priority: 900
    then:
      lambda: |-
        id(wake_up_reason) = esp_sleep_get_wakeup_cause();

globals:
  - id: wake_up_reason
    type: int
    restore_value: no
    initial_value: '0'
    
sensor:
  - platform: template
    name: "Wake-up Reason"
    accuracy_decimals: 0
    lambda: |-
      return (id(wake_up_reason));
# end test wakeup reason
```

After wake-up the sensor report value 2.
Now I don't know what this correspond to, and if it helps to find the reason for immediate wake-up.

jesserockz, 2021-01-20:
Following the docs, the wake up reason comes from the enum here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#_CPPv418esp_sleep_source_t

Where `2` corrosponds to:
> ESP_SLEEP_WAKEUP_EXT0
>    Wakeup caused by external signal using RTC_IO.

closed 2021-06-04 by stale[bot]