esphome/issues #1171: can't add esphome repo

iamkarlson, 2020-04-26:
Hi guys,
I can't add a repo to HACS because its structure is not compliant:

```
An error occurred while processing
Exception: HacsException
Message: Repostitory structure for v1.14.3 is not compliant
Could not add this repository, make sure it is compliant with HACS.
```

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HACS:
```
HACS (Home Assistant Community Store)
Manage (Install, track, upgrade) and discover custom elements for Home Assistant.
Authors:
ludeeus
GitHub stars: 916
Downloads: 7269
Last updated: 7 hours ago
Frontend version: 20200422201603
Installed: 0.24.2
```
Home Assistant 0.109.0b3

glmnet, 2020-04-26:
what should be able to find HACs within this repo?

iamkarlson, 2020-04-27:
@glmnet I don't know. I'm blindly follow documentation available at esphome website here: https://esphome.io/guides/getting_started_hassio.html

I thought that if you have only 1 way to install esphome to HomeAssistant then it should work - but it doesn't.

randybb, 2020-04-27:
HACS != Hass.io

glmnet, 2020-04-27:
Ok, forget about HACs, ESPHome does not need it and that thing is useful but unrelated to ESPHome

Now ESPHome docs are a bit old, what you see there "Hass.io" is now (as of version 0.108) called "Supervisor"

I've read they are going to move the supervisor thing somewhere else later.

closed 2020-04-27 by glmnet