esphome/issues #1172: Documentation for PID Controller in next wasn't merged.

carlos-sarmiento, 2020-04-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
N/A

**ESP (ESP32/ESP8266, Board/Sonoff):**
N/A

**Affected component:**

**Description of problem:**
Pull request esphome/esphome-docs#432 was never merged, even though the code for the component was merged esphome/esphome-docs#885. 

We should reopen and commit the PR so that the documentation is available on next.esphome.io 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-04-27:
I just did it, can you review the docs are complete? Anything to add/remove?

I see you did some work related to PIDs

Thanks

carlos-sarmiento, 2020-04-27:
Docs look good, but I still can't comment on the PR. Also there seems to be a conflict but I can't modify it either

carlos-sarmiento, 2020-05-01:
@glmnet any update on this?

carlos-sarmiento, 2020-06-09:
Is there any chance this documentation will be merged so it's at least available in next.esphome.io?

glmnet, 2020-06-13:
Done, thanks @carlos-sarmiento for pushing

closed 2020-06-13 by glmnet