esphome/issues #1173: Dual SSD1306 breaks wifi (Auth Expired)

ymoona, 2020-04-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCUv2

**Affected component:**

display/wifi

**Description of problem:**
When I add in the config a second SSD1308 display the it cant connect to wifi anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esphome_slaapkamer
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "*********"
  password: "*************"

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

mqtt:
  broker: 192.168.178.203
  username: ************
  password: **********

i2c:
    sda: D1
    scl: D2

font:
  - file: 'Arial.ttf'
    id: arial
    size: 50

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.printf(0, 20, id(arial), "1%s", id(total).state.c_str());
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3D
    lambda: |-
      it.printf(0, 20, id(arial), "2%s", id(current).state.c_str());
      
text_sensor:
  - platform: mqtt_subscribe
    name: "total"
    id: total
    topic: disp/energy/electricity/total
  - platform: mqtt_subscribe
    name: "current"
    id: current
    topic: disp/energy/electricity/current

```

**Logs (if applicable):**

When the second display is connected via hardware but not yet configured:
```
[20:32:50][C][i2c:028]: I2C Bus:
[20:32:50][C][i2c:029]:   SDA Pin: GPIO5
[20:32:50][C][i2c:030]:   SCL Pin: GPIO4
[20:32:50][C][i2c:031]:   Frequency: 50000 Hz
[20:32:50][I][i2c:033]: Scanning i2c bus for active devices...
[20:32:50][I][i2c:040]: Found i2c device at address 0x3C  <---- disp 1
[20:32:50][I][i2c:040]: Found i2c device at address 0x3D <---- disp 1
[20:32:50][C][logger:175]: Logger:
[20:32:50][C][logger:176]:   Level: DEBUG
[20:32:50][C][logger:177]:   Log Baud Rate: 115200
[20:32:50][C][logger:178]:   Hardware UART: UART0
[20:32:50][C][ssd1306_i2c:023]: I2C SSD1306
[20:32:50][C][ssd1306_i2c:023]:   Rotations: 0 °
[20:32:50][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[20:32:50][C][ssd1306_i2c:024]:   Address: 0x3C
[20:32:50][C][ssd1306_i2c:025]:   Model: SSD1306 128x64
[20:32:50][C][ssd1306_i2c:027]:   External VCC: NO
[20:32:50][C][ssd1306_i2c:028]:   Update Interval: 1.0s
```
But when I add the display to the config:
```
[20:43:54][I][logger:156]: Log initialized
[20:43:54][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[20:43:54][I][app:028]: Running through setup()...
[20:43:54][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[20:43:54][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[20:43:54][C][wifi:033]: Setting up WiFi...
[20:43:54][D][wifi:304]: Starting scan...
[20:44:01][D][wifi:319]: Found networks:
[20:44:01][I][wifi:365]: - 'Mywifi' (18:35:**:**:**:**) ▂▄▆█
[20:44:01][D][wifi:366]:     Channel: 11
[20:44:01][D][wifi:367]:     RSSI: -75 dB
[other wifis]
[20:44:01][I][wifi:193]: WiFi Connecting to 'Mywifi'...
[20:44:04][W][wifi_esp8266:337]: Event: Disconnected ssid='Mywifi' bssid=18:35:**:**:**:** reason='Auth Expired'
[20:44:04][W][wifi:451]: Error while connecting to network.
[20:44:04][W][wifi:488]: Restarting WiFi adapter...
[20:44:04][I][wifi:193]: WiFi Connecting to 'Mywifi'...
[20:44:06][W][wifi_esp8266:337]: Event: Disconnected ssid='Mywifi' bssid= 18:35:**:**:**:** reason='Auth Expired'
[20:44:06][W][wifi:451]: Error while connecting to network.
[20:44:06][W][wifi:488]: Restarting WiFi adapter...
[20:44:11][D][wifi:304]: Starting scan...
etc,
```

**Additional information and things you've tried:**
- When I add just one display, it works fine. This could be disp1 or disp2, independently configured then the wifi works. 
- I also tried disconnecting 1 display, then the other display works and wifi works. 
- When I configure both displays, they both work, but the wifi does not.

spattinson, 2020-04-30:
I think this may be related to framework, libs, or esphome and not specific to oled display. I am getting this without display on esp32, there was a guy on reddit asking too with nodemcu esp8266. I am using deep sleep and bme280 only, i have sleep for 60s for testing and frequently see it on wake up.

randybb, 2020-04-30:
I have two ESP32 with bunch of i2c and other sensors, both have the same two OLED displays and I never had a problem. ESP8266 needs a loot of power, so if you don't have stable PSU, then it can cause an issue...
Probably I will find some nodemcu's in my electronics trash bin (this is exactly where it belongs together with atmega) and will test it.

ymoona, 2020-05-02:
I just tested with D1 mini pro. The issue is the same.

randybb, 2020-05-03:
I believe it is related to https://github.com/esphome/issues/issues/455

thieaux, 2020-05-09:
> I just tested with D1 mini pro. The issue is the same.

same issue here with dual display, did u ever sorted it out?

madmic1314, 2020-05-12:
I've done a few more tests with the same issue - dual SSD1306 screens and only able to run on one.

[0;33m[W][wifi_esp8266:337]: Event: Disconnected ssid='SSIDHERE' bssid=[5m24:A4:3C:3F:CF:26[6m reason='Auth Expired'[0m
[0;33m[W][wifi:451]: Error while connecting to network.[0m
[0;32m[I][wifi:193]: WiFi Connecting to 'SSIDHERE'...[0m
[0;33m[W][wifi_esp8266:337]: Event: Disconnected ssid='SSIDHERE' bssid=[5m24:A4:3C:3F:CF:26[6m reason='Auth Expired'[0m
[0;33m[W][wifi:451]: Error while connecting to network.[0m
[0;36m[D][wifi:304]: Starting scan...[0m

Is a snippet from the Serial Logs.

Things I've tried:
Flashing to EasyESP - both screens work (this was to test that both screens were working)
Tried various settings for output_power & fast_connect: True settings. No luck.
Removed the SSID & password section, forcing the device to create it's own AP. Connected to this and was stable (monitored via ping). Set the SSID & PW, but back in the same disconnection loop when rebooting.
Turned off I2c scanning.
Created a hotspot on my phone with no password - same issue.

One thing I did notice, when I disconnect the screens it scans the i2c bus and stops at 0x77. Both screens are inside this address range (0x3D & 0x3C) but the boards are marked as 0x78 and 0x7A - not sure if this is relevant or not, but thought I'd point it out.

ymoona, 2020-05-12:
I never resolved the issues, now I'm back to a single display.

randybb, 2020-05-12:
Different addresses is just due conversion between 7-bit and 8-bit address https://www.letscontrolit.com/forum/viewtopic.php?p=22817#p22818 The standard are 7-bit addresses, that allows to have 128 unique addresses, but only 112 can be used.

Easy solution - switch to esp32, but it doesn't mean this doesn't need to be fixed.

Btw, this issue is related only to two displays, or do we have the same problem with multiple i2c devices?

thieaux, 2020-05-12:
about to post the same, apparently theres some issues with the i2c bus, i switched to esp32 and the problem went away

madmic1314, 2020-05-12:
I read about the 7, 8 and 10 bit encoding, very interesting.
I've ordered a pair of spi screens and will see if I can use two of those at the same time (learning) or one i2c and one spi.

I've used two i2c devices before - have 3 running at once (temp, light and a screen) all works ok. I can't test with two of the temp sensors as they have the same i2c address and I can't change it. Bit would be an interest test for sure.

By esp32, do you mean to switch to a full esp32 board or a different firmware on the D1 mini? I tried LetsControl it and dual screens worked but this is limited to 6 lines of text and is nowhere near as flexible as esphome.

randybb, 2020-05-12:
I have bunch of the SSD1306 with SPI - the disadvantage is that it needs a loot of GPIOs - MOSI, CLK, DC, CS (RS). If you have other SPI devices, then it is not a big problem...

esp32 is a completely new SoC, that means you need new esp32 board. Fortunately, you can buy on Ali D1 mini with ESP32, but the price is ~4 € comparing to 2,20 € for one ESP8266.

madmic1314, 2020-05-16:
I went ahead and got me an ESP32 - ran the same code as I used on the D1 mini and it's working perfectly with two SSD1306's on the same i2c bus.

Oddly enough I did some playing with the D1, had a lot of troubles with the i2c bus with even a single OLED on, monitoring the serial port was showing the wifi looping. Sometimes even with the bus up but nothing configured the wifi wouldn't come up. Strange!

One thing I've wondered - I wasn't using the ESPHome flasher - so I'll go back and double check with this in case this flashes differently.

pinderh2, 2020-08-31:
My guess is this is a RAM issue. The ESP8266 has 80K of RAM. A simple script without any display buffer might use about 35K bytes of RAM or so. The RAM needed for the display buffer on a SSD1306 is 128x64x2 bytes which is 16K, so for two of them that's 32K. Adding these together makes 67K and that's seems pretty tight, as other runtime uses (stack, malloc, etc) haven't been accounted for.
As evidence, I tried integrating a ST7735R with 128x160 TFT display (40K bytes needed for the buffer) and it crashed during the display buffer initialization. 
Also, the ESP32 has a lot more RAM (520 KB) so perhaps that's why its working on that platform.

closed 2021-01-12 by stale[bot]